PaulWieland/ratgdo #2: Door Status Noise When Opening/Closing

maniacaris, 2022-07-24:
Thanks for working on this and getting the boards shipped out so quickly. Overall I'm really please with how this is working. One thing I've noticed on my opener is the status reported on the MQTT client consistently displays the following statuses when I use the open or close commands:

topic/Main Door/**command -> close**
topic/Main Door/**status -> closed** (door just started moving)
topic/Main Door/**status -> closing**
topic/Main Door/**status -> closed** (door is actually closed at this point)

The same thing happens with open but with the open/opening statuses.

I don't know if this is necessarily an issue with the firmware on the ESP because I recall seeing similar issues with MyQ integration, so it could very well be an issue with Chamberlain's hardware/interface. Just wanted to post this in case I can offer any help debugging or if other people observe the same thing. Thanks again!

PaulWieland, 2022-07-24:
What model garage door opener do you have?

Can you connect to the esp to get the serial log? 

There is an option within ESP web tools to view the log. Once connected trigger the door and then copy/paste the whole log.

maniacaris, 2022-07-25:
It's a Chamberlain B550

**Open via MQTT:**

> open
> MQTT: open the door
> door_code[0]
> door_code[1]
> door_code[2]
> door_code[3]
> door_code[4]
> door_code[5]
> door_code[6]
> Obstruction Detected
> Open
> Door Position: 2224
> Door Position: 2225
> Door Position: 2226
> Door Position: 2227
> Door Position: 2228
> Door Position: 2229
> Door Position: 2230
> Door Position: 2231
> Door Position: 2232
> Door Opening...
> Door Position: 2233
> Door Position: 2234
> Door Position: 2235
> Door Position: 2236
> Door Position: 2237
> Obstruction Cleared
> Door Position: 2238
> Door Position: 2239
> Door Position: 2240
> Door Position: 2241
> Door Position: 2242
> Door Position: 2243
> Door Position: 2244
> Door Position: 2245
> Door Position: 2246
> Door Position: 2247
> Door Position: 2248
> Door Position: 2249
> Door Position: 2250
> Door Position: 2251
> Door Position: 2252
> Door Position: 2253
> Door Position: 2254
> Door Position: 2255
> Door Position: 2256
> Door Position: 2257
> Door Position: 2258
> Door Position: 2259
> Door Position: 2260
> Door Position: 2261
> Door Position: 2262
> Door Position: 2263
> Door Position: 2264
> Door Position: 2265
> Door Position: 2266
> Door Position: 2267
> Door Position: 2268
> Door Position: 2269
> Door Position: 2270
> Door Position: 2271
> Door Position: 2272
> Door Position: 2273
> Door Position: 2274
> Door Position: 2275
> Door Position: 2276
[... 1103 lines cut ...]
> Door Position: 2828
> Door Position: 2827
> Door Position: 2828
> Door Position: 2827
> Door Position: 2828
> Door Position: 2827
> Door Position: 2828
> Door Position: 2827
> Door Position: 2828
> Door Position: 2827
> Door Position: 2826
> Door Position: 2825
> Door Position: 2826
> Door Position: 2825
> Door Position: 2824
> Door Position: 2825
> Door Position: 2826
> Door Position: 2825
> Door Position: 2826
> Door Position: 2825
> Door Position: 2826
> Door Position: 2827
> Door Position: 2828
> Door Position: 2827
> Closed

PaulWieland, 2022-07-25:
At first I thought maybe you had the RPM1 & 2 signal wires reversed, but after studying the log I'm not so sure...

Here's what a "normal" rotary encoder plot looks like:
[image]

The door started out closed. As the door opens, the counter increments in a nice linear fashion. Once its fully open, I closed it, and the counter decreases in a nice linear fashion. 

Now here's what your GDO's rotary encoder plot looks like:
[image]

The opening/climbing part looks normal. But the count down after you close the door is not working correctly at all.

You also mentioned:

> I recall seeing similar issues with MyQ integration
> I've been having issues with ‘excessive closing force’ hence why it opened again

These three things have me wondering if your rotary encoder is defective.

The GDO's logic board uses the rotary encoder in lieu of limit switches to detect the door's physical location. It needs this to know which direction the motor should spin and when to stop the motor. The motor should stop when the door is fully open, fully closed or when it stops moving unexpectedly (due to an undetected obstruction or track jamb).

When we look at the plot of counters coming from your rotary encoder, we can see that the count down as the door closes is abnormal looking. Lots of duplicated numbers and the counter does not progress in a linear fashion. The end position/counter of the graph line is really a long way away from the start position/counter. You can see in the test graph that the start/end counters are much closer together.

I suspect that your GDO logic board occasionally interprets the slope of this line this as an obstruction/track binding event and so it stops and reverses the motor. Certainly the ratgdo firmware is confused by it which is why you are getting the erratic status messages.

PaulWieland, 2022-07-25:
One more thing, take a close up, well lit photo of your soldering on the RPM1 & 2 signals. Maybe you have a solder bridge that is messing up the signal. If you had problems with the myQ app and with the door bouncing open before you soldered the wires for ratgdo, then I really think you have a faulty rotary encoder module.

Here is the rotary encoder module (Chamberlain calls it AC Travel Module): 
https://www.chamberlain.com/ac-travel-module-kit/p/041D7742-7

maniacaris, 2022-07-25:
Really interesting insight Paul. Thanks!

The issues with the false closes existed before I soldered the connections so the rotary encoder is certainly something I'll look into. I'll admit my soldering skills and equipment aren't the greatest so that's entirely a possibility and I'll check it out.

maniacaris, 2022-08-12:
I replacd the rotary encoder and it doesn't seem to have made a difference. I noticed it seemed to work perfectly when it opens with minimal repeats and a linear progression, but the closing is all over the place if it moves at all. I disconnected RPM1 and RPM2 and it seems to work slightly better now, but I still get the reopens sporadically. I'll try resoldering next.

PaulWieland, 2022-08-12:
Unhook the obstruction sensor wire from the shield and test the door again just to rule out some sort of issue being caused by the shield itself. 

If you can, get another log from it so we can plot the encoder values.

Dulaithol, 2022-10-04:
I ran into the issue with my Chamberlain B550 as well and was able to fix it with a software change. From what I can tell the encoder is noisy on this model and causing random values when reading RPM1 and RPM2. My inexpensive logic analyzer didn't show any issues while the ESP8266 was counting more level transitions compared. I am unable to hook a scope up to confirm so this is just a theory.

The fix works by using interrupts instead of reading values on the loop. It tracks the values and counts when both values go low with a 2ms debounce. The order of the transitions determine if the door is opening or closing. You can find the code stashed in a branch on my fork[1].

If there is interest happy to open a pull request but such change would require significant testing using other models. Leaving this here for others to save them the effort if they wish to journey done the path I did.

[1] https://github.com/Dulaithol/ratgdo/blob/3946e9665a9d2bd81d2f7081b7534970b130778d/src/ratgdo.cpp#L155

PaulWieland, 2022-10-04:
> My inexpensive logic analyzer didn't show any issues while the ESP8266 was counting more level transitions compared.

How did you have the logic analyzer wired? I wonder if the grounding of the rotary encoder sensor is different between models.

Dulaithol, 2022-10-04:
I had it wired to RATGDO directly so they shared a common ground through the obstruction sensor. Like you said it's possible that internally there are different grounds.

Hoping to barrow my brothers scope this weekend.

PaulWieland, 2022-10-04:
I'm going to try implementing a 2ms debounce per your suggestion and see if that fixes it. will need some people to test...

Dulaithol, 2022-10-04:
Happy to test.

PaulWieland, 2022-10-05:
@Dulaithol can you share your counter plot?
I changed over to using an ISR with 2ms debounce for RPM1/2 and tested, but the plot is not looking so good. As a result I'm getting glitched statuses. 
[image]

Here it is with the ISR enabled, but with debounce turned off:
[image]

chrispatin, 2022-10-09:
@Dulaithol  - I applied your patch and it fixed the issue I was seeing with the status on closing.

PaulWieland, 2022-10-13:
I just published version 1.5a (alpha) which uses an ISR + a rudimentary filter to read the rotary encoder. I'm flying blind here because my test openers don't have an issue mis-counting pulses, so I need those affected to test this out and see if it fixes their issue.

Dulaithol, 2022-10-13:
I'll give it a test soon.

When I tried using a single interrupt reading of the second pin I was getting unpredictable values. Based on the signal in theory your solution should work.

chrispatin, 2022-10-14:
Quick test results:

- The device was reset with the garage door closed. The device incorrectly reports the door position changing, and alternating states between closing / closed. 
- Door activated to open (@Obstruction Cleared), the door opens, and after open the log output shows no activity.
- Door activated to close, and after closed the log shows no activity.

Regarding the 'Obstruction Cleared', I have noticed with all firmware versions, when the door is closed an obstruction will be falsely reported and never clears. If the door is open, the obstruction sensor appear to report actual events when the beam is broken, and clears shortly after.

[esp-web-tools-logs-1.5a.txt](https://github.com/PaulWieland/ratgdo/files/9782140/esp-web-tools-logs-1.5a.txt)

PaulWieland, 2022-10-14:
@chrispatin  Thanks for the test results. Its good news... the counter is perfectly linear in both directions. Now I just have to figure out why the door status is interpreted incorrectly from the values. 
[image]

chrispatin, 2022-10-14:
To ensure we are on the same page, after resetting the device, the door status / door positions are being falsely reported while the door remained closed during the period. Should the device be able to detect the door state accurately after a reset?

After I opened the door, the status appeared correct. What I am not clear on is that after clearing the obstruction, the 'phantom' logs continued for more time until the garage door button was pressed a second time. 

```
Obstruction Cleared
Door Position: -78
Door Closing...
Door Position: -79
Door Position: -75  <- did the open actually start here?
Door Position: -72
Door Opening...
```

obstruction, 2022-10-14:
Hello,

As the obstruction ***@***.***), I believe I am mistakenly recieving these
emails. Perhaps the intended obstruction is @github?

Thank you!

PaulWieland, 2022-10-15:
My theory is that the ISR is falsely being triggered if the rotary encoder is parked on a high pulse. 

This is the only explanation I can come up with given the log file that shows door closing, closed happening repeatedly. 

Now to test the theory and come up with a solution…

chrispatin, 2022-10-15:
Today, I hooked up the laptap while the door was closed and I did not see the previous behavior. The door was opened / closed, and reset ratgdo a few times.

[esp-web-tools-logs (4).txt](https://github.com/PaulWieland/ratgdo/files/9793246/esp-web-tools-logs.4.txt)

I have not seen the issue with the garage keypad flashing (obstruction) for a few days. It happened 3x time today, while the laptop was attached and I reset the device.

PaulWieland, 2022-10-17:
Everyone with this issue, please upgrade to the just released version 1.5 which implements completely new logic for handling the rotary encoder.

If a new problem related to door position detection / door counter exists, please open a new issue.

closed 2022-10-17 by PaulWieland