PaulWieland/ratgdo #3: Proper MQTT Status, but unable to send Commands

switchtower, 2022-07-29:
Hello,

I received my ratgdo this week and today I attempted to install it.  The Liftmaster I have is a bit older, but I was able to find a 3.3 pin on it to power the device.  It appears the area of the board for the rotary encoder is similar to your picture so I used your soldering suggestion for that.  The obstacle detection works as well.

I'm able to receive opened, opening, closing, closed via mqtt.

Unfortunately, I haven't been successful in sending commands to the ratgdo via mqtt.

Let me know if there's any information I can provide you.

Thanks!

PaulWieland, 2022-07-30:
What’s the model number of the opener?

If you plug it back into the USB port of your computer and visit the esp tools page, you can view the serial log. Send an open or close mqtt command to the opener while the log viewer is open then paste the result here. 

I want to verify that it outputs `MQTT: open the door` or `MQTT: close the door`

switchtower, 2022-07-30:
send open command

```
[10:12:27]door_code[1]
[10:12:27]door_code[2]
[10:12:27]door_code[3]
[10:12:27]door_code[4]
[10:12:27]door_code[5]
[10:12:27]door_code[6]
[10:12:27]Obstruction Detected
[10:12:27]Open
```

send another open command

```
[10:12:44]open
[10:12:44]MQTT: open the door
[10:12:44]The door is already open
```

send close command

```
[10:12:58]close
[10:12:58]MQTT: close the door
[10:12:58]door_code[0]
[10:12:58]door_code[1]
[10:12:58]door_code[2]
[10:12:58]door_code[3]
[10:12:58]door_code[4]
[10:12:58]door_code[5]
[10:12:58]door_code[6]
[10:12:58]Obstruction Detected
[10:12:58]Closed
```

send another close command

```
[10:13:01]close
[10:13:01]MQTT: close the door
[10:13:01]The door is already closed
```
[image]
[image]
[image]

switchtower, 2022-07-30:
Perhaps related, when i hit the physical button, the door goes down, but then immediately comes back up, here is the output of that:

```
[10:46:18]Door Closing...
[10:46:18]Door Position: 251
[10:46:18]Door Position: 250
[10:46:18]Door Position: 249
[10:46:18]Door Position: 250
[10:46:18]Door Position: 251
[10:46:18]Door Position: 252
[10:46:18]Door Position: 253
[10:46:18]Door Position: 254
[10:46:18]Door Position: 255
[10:46:18]Door Position: 256
[10:46:18]Door Position: 255
[10:46:18]Door Position: 256
[10:46:18]Door Position: 255
[10:46:18]Door Position: 256
[10:46:19]Door Position: 257
[10:46:19]Door Position: 258
[10:46:19]Door Opening...
[10:46:19]Door Position: 257
[10:46:19]Door Position: 258
[10:46:19]Door Position: 259
[10:46:19]Door Position: 260
[10:46:19]Door Position: 261
[10:46:19]Door Position: 262
[10:46:19]Door Position: 263
[10:46:19]Door Position: 264
[10:46:19]Door Position: 265
[10:46:19]Door Position: 266
[10:46:19]Door Position: 267
[10:46:19]Door Position: 268
[10:46:19]Door Position: 269
[10:46:19]Door Position: 270
[10:46:19]Door Position: 271
[10:46:19]Door Position: 272
[10:46:19]Door Position: 273
[10:46:20]Door Position: 274
[10:46:20]Door Position: 275
[10:46:20]Door Position: 276
[10:46:20]Door Position: 277
[10:46:20]Door Position: 278
[10:46:20]Door Position: 279
[10:46:20]Door Position: 278
[10:46:20]Door Position: 279
[10:46:20]Door Position: 280
[10:46:20]Door Position: 279
[10:46:20]Door Position: 280
[10:46:20]Door Position: 279
[10:46:20]Door Position: 280
[10:46:20]Door Position: 281
[10:46:20]Open
```

PaulWieland, 2022-07-30:
I suspect that the 3.3v power source you used may not be able to provide enough current. Try and disconnect the yellow power wire and power the esp from usb instead. 

Repeat the oem button test to see if the door opens/closes correctly, and repeat the mqtt test. 

Also I don’t see your model number. I can try and look that up by part number later, but I think the model number is on the opposite end of the opener.

switchtower, 2022-07-30:
I've tried a usb power supply as well, a higher amp one, with the yellow cable disconnected, but with same results.

Attached is the model number, sorry about not including that to being with.

[image]

Scott--R, 2022-07-30:
You can see in the feedback that it detected an obstruction. I know when a garage door opener detects an obstruction it will reverse and go back up. It detects obstructions when either the travel distance or force setting are triggering it. So maybe just re-learn the closing force / travel distance and it will behave?

PaulWieland, 2022-07-30:
> I've tried a usb power supply as well, a higher amp one, with the yellow cable disconnected, but with same results.
> 
> Attached is the model number, sorry about not including that to being with.
> 
> [image]

What’s the model number of your wall mounted button?

switchtower, 2022-07-30:
@Scott--R, would connecting to the rotary encoding cause that issue? I tried disconnecting each wire, individually, and only when I disconnected the RPM2 cable did the door finally go down.

@PaulWieland Model 882LM

PaulWieland, 2022-07-30:
This is a bit of a mystery.  I did my reverse engineering with the 882LMW. The LM is an older version from what I can tell, but the LMW is supposed to be compatible with your opener. 

You wouldn’t happen to have a logic analyzer would you?

switchtower, 2022-07-30:
Unfortunately, I don't have a logic analyzer. :( 

 Does the opener have to be connected? Can I just use the ratgdo and then use a buttons and the IO ports to control the up/down status? 

Can I send the wall mount button to you?

PaulWieland, 2022-07-30:
If you’re willing to lend your LM to to assist with the r&d, then I would gladly accept your help. I’ll send it back when finished. 

Before you bother shipping it though, send me a close up pic of the LM pcb so I can see what microcontroller it is. Hopefully it’s an stm8

switchtower, 2022-07-30:
[image]

PaulWieland, 2022-07-30:
Looks good. It looks basically the same as the LMW. Send me an email if you don’t have my mailing address from the return address when I shipped it to you. 

p@ul.sickdimension.com

Scott--R, 2022-08-01:
> @Scott--R, would connecting to the rotary encoding cause that issue? I tried disconnecting each wire, individually, and only when I disconnected the RPM2 cable did the door finally go down.
> 
> @PaulWieland Model 882LM

@switchtower  I'm too many beers in tonight to be sure, but the absolute first thing that comes to mind is loading. Any time hardware behaves differently when you connect something to it vs not connecting something to it, it means you're messing with the signals by connecting to it. RPM2 appears to go into the ESP32 module GPIO pin. If that pin has pull up / pull down enabled, it could mess with it. Perhaps try putting a 10K or 100K resistor in series?

PaulWieland, 2022-08-01:
> RPM2 appears to go into the ESP32 module GPIO pin. If that pin has pull up / pull down enabled, it could mess with it. Perhaps try putting a 10K or 100K resistor in series?

Yes, RPM2 is connected to D2 / GPIO 4. (RPM1 is D1 / GPIO 5)

The impedance of the ESP8266 GPIO is really high. It's only drawing a couple of µA to read the signal from the rotary encoder. Granted, it's a couple µA more than the encoder would otherwise supply, but I would be really surprised if its enough to cause an issue.

PaulWieland, 2022-08-01:
I just published 1.4alpha for you to test (it's available in the ESP Tools installer). It uses soft serial instead of the hardware uart. I am hoping that the issue you are having was being caused by the dynamic switching of the D4 pin from serial to GPIO and back.

This was creating a false voltage when the pin changed state. My test rig doesn't seem to care, but maybe the older GDO hardware does and so it ignores the entire message.

switchtower, 2022-08-01:
@PaulWieland thanks for the hard work.  Once I'm done with work, I'll give this a try.  I already have the code on the ESP, I just need to plug it in and test.

BTW, the PCB is already on its way, if this ends up being the problem, just sent that back, unless you wanted to do some work on it regardless.

PaulWieland, 2022-08-01:
> BTW, the PCB is already on its way, if this ends up being the problem, just sent that back, unless you wanted to do some work on it regardless.

Even if this fix works, having your LM button is going to be valuable in ensuring compatibility with all openers. I will be able to tell if they have patched or changed the firmware since 2015.

switchtower, 2022-08-01:
@PaulWieland 
The changes you added seemed to fix the rogue/ghost obstruction issue and with proper power, I'm able to get the door to go up and down, reliably using the wall mounted button and the myQ home assistant integration.   If I use the 3.3v power I discovered on the board, the issues come back.

I am not, however, able to use the MQTT commands, here is the output from the console:
```
[14:52:13]MQTT: close the door
[14:52:13]door_code[0]
[14:52:13]door_code[1]
[14:52:13]door_code[2]
[14:52:13]door_code[3]
[14:52:13]door_code[4]
[14:52:13]door_code[5]
[14:52:13]door_code[6]
[14:52:13]Closed
[14:52:25]open
[14:52:25]MQTT: open the door
[14:52:25]door_code[0]
[14:52:25]door_code[1]
[14:52:25]door_code[2]
[14:52:25]door_code[3]
[14:52:25]door_code[4]
[14:52:25]door_code[5]
[14:52:25]door_code[6]
[14:52:26]Open
[14:52:37]close
[14:52:37]MQTT: close the door
[14:52:37]door_code[0]
[14:52:37]door_code[1]
[14:52:37]door_code[2]
[14:52:37]door_code[3]
[14:52:38]door_code[4]
[14:52:38]door_code[5]
[14:52:38]door_code[6]
[14:52:38]Closed
```
Use MQTT results in the the logs above but with no movement from the door itself.  I also tried using the GPIO pins to open/close/toggle the light, with no luck

PaulWieland, 2022-08-01:
The change I made wont affect the door bouncing or erratic messages. It would have only caused the door to respond to the open/close commands. 

The door having issues when you power it with the 3.3v pin indicate that the power source you are using is browning out when powering the ESP8266. This is causing the readings for the rotary encoder to be wrong as the sensor is probably being deprived of power.

I await the 882LM and hopefully I can figure out whats wrong after studying it.

switchtower, 2022-08-02:
it was mailed out yesterday.

PaulWieland, 2022-08-08:
I got the LM button hooked up to the debugger and found out that it cannot control my door opener. So first lesson learned is that the security + 2.0 protocol changed somewhere along the line. I'm going to keep prodding, but I may have to source a 2015 era door opener.

switchtower, 2022-08-08:
interesting.

I was just going to suggest if I could send you just the logic board from my garage door opener.

Let me know if there's anything else you need.

PaulWieland, 2022-08-08:
@switchtower before we go down that road, I have a couple of ideas to try. Assuming you don't mind anyway...

version 1.42a (available from the ESP Tools installer) adds a 500ms delay between each code that is sent to the opener. One hypothesis I have is that the older GDO might have a slower processor. It might be that the original 45ms delay is simply too fast for it to process, but works fine on newer openers which presumably would have faster hardware.

switchtower, 2022-08-09:
@PaulWieland I don't see 1.42a, i do see 1.41a though.

PaulWieland, 2022-08-09:
[It’s there](https://github.com/PaulWieland/ratgdo/commit/46c1c951ffe4c94038136aaed34ec2b8075c9aea)… maybe your browser cached the old page? 

None the less @mterry63 [tested it](https://github.com/PaulWieland/ratgdo/issues/4#issuecomment-1208652879) and didn’t have any luck. 

I’m scouring the classifieds for a 2015 opener. I know you offered to send me your logic board, but I would hate for  you to not have a functioning GDO while I figure this out. 😅

PaulWieland, 2022-08-09:
I just bought a 2015 logic board off of ebay to develop against

PaulWieland, 2022-08-15:
Well there's good news and there's bad news: The 050DCTB logic board from model year 2015 accepts the commands from ratgdo.

switchtower, 2022-08-15:
OK, my board has "45ACBL1" "Rev C" on it.  I never personally texted 1.42a, I might go test that right now.

Paul also, can you elaborate on what you mean by "[it] accepts the commands"? Via MQTT or are you doing something else?  Just wanting to make sure we're testing the same.

PaulWieland, 2022-08-15:
@switchtower I'm sending the light / door open/close commands to the opener from ratgdo and the opener is responding (relays are clicking). It dosen't matter if this is done via mqtt or dry contact. the output to 12v line is the same.

Now I'm trying to locate a `45ACBL1` to test with.

switchtower, 2022-08-15:
I just tested 1.4a, and here is the output I received from the console:

```
[14:07:15]door_code[1]
[14:07:15]door_code[2]
[14:07:16]door_code[3]
[14:07:16]door_code[4]
[14:07:17]door_code[5]
[14:07:18]door_code[6]
[14:07:18]Obstruction Detected
[14:07:18]Open
[14:07:31]close
[14:07:31]MQTT: close the door
[14:07:31]door_code[0]
[14:07:32]door_code[1]
[14:07:33]door_code[2]
[14:07:33]door_code[3]
[14:07:34]door_code[4]
[14:07:34]door_code[5]
[14:07:35]door_code[6]
[14:07:35]Obstruction Detected
[14:07:35]Closed
[14:08:02]open
[14:08:02]MQTT: open the door
[14:08:02]door_code[0]
[14:08:02]door_code[1]
[14:08:03]door_code[2]
[14:08:03]door_code[3]
[14:08:04]door_code[4]
[14:08:04]door_code[5]
[14:08:05]door_code[6]
[14:08:06]Obstruction Detected
[14:08:06]Open
```

PaulWieland, 2022-08-15:
@switchtower Test the light function. If it works, the relay will click (and bulb will turn on/off if you have it installed)

switchtower, 2022-08-15:
@PaulWieland I tested the light through MQTT, I get a flicker on the LED but that's all, no change with the lights.

Let me know if it would be easier to test on my board then try to source one.

PaulWieland, 2022-08-15:
@switchtower yes it would be easier. I can’t find that exact board anywhere at the moment.

PaulWieland, 2022-08-31:
Update on the situation - I acquired an 45 part number series logic board. It seems that 45 series boards do not have onboard wifi and 50 series boards do. I believe that a bug existed on the 45 series that was corrected in the newer 50 series design.

The 45 series openers will not accept messages generated with the same counter. That means if your existing wall mounted opener started at counter 0 and you have since pressed it 250 times, and then the button fails and you replace it - you have to press the new button 251 times before the door opener will accept commands from it. This scenario probably doesn't happen often, but when it does there is no practical way for a consumer to fix it.

The newer 50 series openers accept a sync counter command (what I'm calling it anyway). When a new button is connected to the opener, the button sends a "please sync and accept my counter" command. The GDO's logic board will then accept commands based on the synchronized counter value. ratgdo exploits this by syncing the counter every time a command is sent.

This strategy will not work with 45 series openers. Instead I will have to fully implement the rolling code generator and transmit new codes based on the correct counter. 

This is good and bad. Its good because it means the hardware doesn't need to change, but bad because it's going to take me some time to implement the rolling code generator in C++.

switchtower, 2022-08-31:
@PaulWieland thanks for the update, and I apologize for not getting you the board from mine.  I have it ready to go, but due to some life circumstances, I have not had a chance to get it sent out to you.

I'll go ahead and reconnect my board and wait for any updates from you so I can test.

Will newer boards work where a 45 board was originally installed; can I just swap the board for a newer one?

PaulWieland, 2022-08-31:
@switchtower  maybe... The `45ACCL1` board I bought to test with has the same connectors as the `50ACCL1` board in my opener. I plugged it in and it seems to work.

However I have another opener board `50DCBL5` that has totally different connectors on it. This board probably came from an opener that had a DC backup battery. I'm theorizing that the 45AC & 50AC boards are 120v AC only, while the boards with DC in the number are 120v AC + battery backup.

The next problem is the numbering... when I search for replacement parts on chamberlains website, they list `50ACTWFMC` as the replacement board for mine, even though mine is labeled `50ACCL1`. They physically look the same, so I think it's right, I just don't understand the numbering scheme.

I found quite a few on ebay, so check there. FWIW it seems the 45 series boards are worth more than the 50 series boards, so you could flip your old board if it works.

mterry63, 2022-09-02:
> Update on the situation - I acquired a 45 series part number logic board. It seems that 45 series boards do not have onboard wifi and 50 series boards do. I believe that a bug existed on the 45 series that was corrected in the newer 50 series design.
> 
> The newer 50 series openers accept a sync counter command (what I'm calling it anyway). When a new button is connected to the opener, the button sends a "please sync and accept my counter" command. The GDO's logic board will then accept commands based on the synchronized counter value. ratgdo exploits this by syncing the counter every time a command is sent.
> 
> The 45 series openers will not accept any messages generated with the same counter, so the strategy that I implemented for ratgdo will not work. It also means that if your existing wall mounted opener started at counter 0 and you have since pressed it 250 times, and then the button fails and you replace it - you have to press the new button 251 times before the door opener will accept commands from it. This scenario probably doesn't happen often, but when it does there is no practical way for a consumer to fix it.
> 
> I believe this is why the newer 50 series openers behave differently.
> 
> This is good and bad. Its good because it means the hardware doesn't need to change, but bad because it's going to take me some time to implement the rolling code generator in C++.

It's interesting because the manual for my opener says it may be necessary to press the wall button "up to 3 times" for it to sync with the logic board for the 45ACT logic board. It seems to imply it has some kind of sync command, but maybe not the same?

PaulWieland, 2022-09-03:
> It's interesting because the manual for my opener says it may be necessary to press the wall button "up to 3 times" for it to sync with the logic board for the 45ACT logic board. It seems to imply it has some kind of sync command, but maybe not the same?

yeah that’s what is supposed to happen - and it does on the 50 series logic board. But if you modify the memory of the wall mounted button and set the counter back to zero, you have to click past the last user counter before the button starts to work again with a 45 series logic board.

mterry63, 2022-09-04:
Any timeline on the new update to support 45 series logic cards? I'm trying to decide if I need to deploy an alternate solution using a Sonoff 4CHPRO. I can wait if it's only a couple of weeks.

PaulWieland, 2022-09-04:
It’s going to be a while. I have to port 1500 lines of JavaScript to C++

antsar, 2022-09-26:
Also interested in 45ACCL1 support. Any way to contribute to the effort? I've got ratgdo shields, a logic analyzer, and JS experience. Glad to spend some time testing/coding if that would help, though I could use a cue where to get started.

PaulWieland, 2022-09-26:
> Also interested in 45ACCL1 support. Any way to contribute to the effort? I've got ratgdo shields, a logic analyzer, and JS experience. Glad to spend some time testing/coding if that would help, though I could use a cue where to get started.

I would love some help. Please contact me on iMessage or Signal 518-378-5420

mterry63, 2022-10-15:
Any update on 45xxx series board support? I bought 3 of the ratgdo cards but no way to use them at this point.

PaulWieland, 2022-10-17:
@mterry63 The current plan is to release a version which has thousands of pre-calculated rolling codes stored in memory which can be used in sequence for these 45 series boards.

Zebble, 2022-10-17:
@PaulWieland it appears I have a similar vintage GDO with a slightly newer 45-series board (45DCBR1 - 10/17/2016 - Rev C).  Looks pretty similar to the 45A pictures here but i'll have to do a bit of searching to match the pins.

I have some electronics skills and a decent oscilloscope if you need any analysis done!

Zebble, 2022-10-17:
Photo of 45DCBR1 included for reference.
[image]

PaulWieland, 2022-10-20:
A short update for those 45 series owners patiently waiting for a fix (also from [Issue 3](https://github.com/PaulWieland/ratgdo/issues/3))

I have a proof of concept working with my test board. What I'm doing is pre-generating 10s of thousands of codes and saving them to a file. ratgdo will then keep track of which codes have been used and advance to the next code.

The trick here is that codes are generated from a code type and a counter. There are 6 code types. So I have to pre-generate every combination since there is no way to predict which code type will be needed next.

The other challenges are:

1. figuring out how to get ESP Tools to flash the code list which is stored in a ~2MB+ binary file to the ESP8266's flash storage.
2. keeping track of the counter your device is on when flashing new firmware, since ESP Tools wipes the flash memory clean 
3. coming up with a way for users to easily change their counter in case something stops working. I can add an input to the web interface, but if you pick a value thats either out of bounds or close to the end of the code range, then it wont work at all, or it wont work for long enough :D  
4. making sure this approach isn't less reliable than the original approach for those with 50 series boards

Zebble, 2022-10-20:
Thanks for the update @PaulWieland !  I'm eager to test when it's ready.
As a possible fallback, I also have a 50-series board coming (should be here Tuesday) to try and retrofit in place of the 45DCBR1.

switchtower, 2022-11-16:
> Thanks for the update @PaulWieland ! I'm eager to test when it's ready. As a possible fallback, I also have a 50-series board coming (should be here Tuesday) to try and retrofit in place of the 45DCBR1.

@Zebble were you able to retrofit the 50 series board in place of the 45? I'm contemplating doing the same; for some reason now, i can't the remote garage door openers to sync with the opener I was testing on and with all the problems they have, it might be better to just replace them.

Zebble, 2022-11-17:
> > Thanks for the update @PaulWieland ! I'm eager to test when it's ready. As a possible fallback, I also have a 50-series board coming (should be here Tuesday) to try and retrofit in place of the 45DCBR1.
> 
> @Zebble were you able to retrofit the 50 series board in place of the 45? I'm contemplating doing the same; for some reason now, i can't get the remote garage door openers to sync with the opener I was testing on, and with all the problems they have, it might be better to just replace them.

Hey @switchtower.  I haven't done the swap yet, but have gone through some preliminary tracing based on the new board I now have in my hands, and photos of the board that's installed in my opener.  I now have to find some time to remove that board and start bench testing.  I believe things will work as the connectors appear to be the same wire colours, they're just in different locations and possibly reversed.

I will make a point of doing that hopefully over the weekend, now that someone else is relying on my results.  ;)

mterry63, 2022-11-30:
> A short update for those 45 series owners patiently waiting for a fix (also from [Issue 4](https://github.com/PaulWieland/ratgdo/issues/4))
> 
> I have a proof of concept working with my test board. What I'm doing is pre-generating 10s of thousands of codes and saving them to a file. ratgdo will then keep track of which codes have been used and advance to the next code.
> 
> The trick here is that codes are generated from a code type and a counter. There are 6 code types. So I have to pre-generate every combination since there is no way to predict which code type will be needed next.
> 
> The other challenges are:
> 
> 1. figuring out how to get ESP Tools to flash the code list which is stored in a ~2MB+ binary file to the ESP8266's flash storage.
> 2. keeping track of the counter your device is on when flashing new firmware, since ESP Tools wipes the flash memory clean
> 3. coming up with a way for users to easily change their counter in case something stops working. I can add an input to the web interface, but if you pick a value thats either out of bounds or close to the end of the code range, then it wont work at all, or it wont work for long enough :D
> 4. making sure this approach isn't less reliable than the original approach for those with 50 series boards

Any update for people with the 45 series boards?

PaulWieland, 2022-12-01:
Im testing an alpha version right now. Having a lot of problems with the arduino bootstrapper library and littlefs, but im making progress.

PaulWieland, 2022-12-22:
**ratgdo 2.0 alpha which includes support for rolling codes is ready for testing.** This version requires a ~3mb file which contains 64000 counters worth of codes. Each time the ESP reboots a sync message is sent which consumes 5 counters worth of codes.

I'm [having a problem](https://github.com/esphome/esp-web-tools/issues/297) getting Improv/ESP Web Tools to flash the filesystem, so for the time being you will have to use the ESP Flasher python tool/VS Code/etc. Incidentally, if anyone knows JavaScript and wants to try and figure out how to work around the ESP Web Tools timeout problem, let me know!

To manually flash 2.0a:

1. make sure you have esptool.py (requires python3). [Instructions](https://tasmota.github.io/docs/Esptool/)
1. download both the [firmware](https://github.com/PaulWieland/ratgdo/blob/main/docs/flasher/ratgdo_esp8266_v2.0a.bin) and the [filesystem](https://github.com/PaulWieland/ratgdo/blob/main/docs/flasher/ratgdo_esp8266_v2.0a_littlefs.bin) image.
2. Erase your ESP: `esptool.py erase_flash`; On my system the port that the ESP8266 is connected to is automatically found. Make sure no other programs (vs code serial monitor for example) are using the ESP's serial port
3. Flash the firmware at position 0x0: `esptool.py write_flash 0x0 /path/to/ratgdo_esp8266_v2.0a.bin`
4. Flash the file system image at position 0x100000: `esptool.py write_flash 0x100000 /path/to/ratgdo_esp8266_v2.0a_littlefs.bin`
5. After successfully flashing, the ESP8266 will reboot into hotspot mode. Connect to the `ArduinoBootstrapper` wifi network and then open the configuration page in your web browser. In MY case the ESP's IP and webconfig address is `http://192.168.4.1`. You will then be presented with the ratgdo configuration page.
6. Enter all of the configuration settings you need. Enable rolling codes by checking the box at the bottom and save the configuration. 
7. After saving, the ESP8266 will reboot and it will begin using rolling codes **starting at counter 128** (they will be output to the serial monitor).

Send an MQTT command to try and toggle the light on/off. If your door opener does not respond, its possible that your factory button used up the counters at value 128. My test rig did not. 

It is possible to manually change the counter to a different value via MQTT, but BE CAREFUL as counters cannot be re-used* I would suggest only advancing by 50 or 100 at a time until the opener begins responding to light toggle commands. 

Each time you advance the counter you are fragmenting the available counter space. If you end up ahead of your wall mounted door opener's counter, the button will not function until you click it enough times to advance the counter into fresh territory.

*The counter value is a 32 bit number, so there are 4,294,967,295 possible values. ratgdo is programmed to use the first 64,127 of them. That is as many as I can fit on a 4MB D1 Mini. It is entirely probably that spent counters will _eventually_ be forgotten by the opener and can be re-used, but I have no way of knowing at this time.

Let me know your findings!

Once I figure out this ESP Web Tools time out problem, installing will be done the normal way which is much more convenient.

mterry63, 2022-12-23:
@PaulWieland, I'm willing to try the 2.0alpha on one of my 45 series boards, but I want to be cautious of the gotchas. My ratgdo is currently connected to the garage door for my wife's car so I don't want to make any relationship-breaking mistakes. Can you elaborate a bit on how to proceed if the inital counter value doesn't work? Any idea what the GDO starts with? Mine is 4-5 years old. Do they increase sequentially by 1? It appears from the description above that once you get ahead of the current GDO counter things start working? But you may have to press the door button enough to "catch up"?

PaulWieland, 2022-12-23:
> Can you elaborate a bit on how to proceed if the inital counter value doesn't work?

ratgdo starts at counter 128. If it works, everything should be good. If it doesn't work, you can try increasing the counter value by 50, syncing, and trying again.

> Any idea what the GDO starts with? Mine is 4-5 years old. 

I wish I knew, but for what its worth the used 45 board I got off ebay which is probably 8+ years old worked just fine starting at 128.

> Do they increase sequentially by 1?

Yes. Each code sent consumes one counter. It's important to note though, that not every event uses just a single counter. When rebooting, 5 codes are sent. The factory button sends two messages when opening the door (though only one is required - I have no idea why it sends 2). The factory buttons periodically send a code while idle too. With ratgdo, I do not send periodic ping commands and I also do not send the second door code when opening/closing. It doesn't seem to be required and I don't want to burn through codes unnecessarily.

> It appears from the description above that once you get ahead of the current GDO counter things start working? But you may have to press the door button enough to "catch up"?

No not quite. Each counter is used to generate a code. Once sent, that counter is used up and the door opener will not accept any other command for the same counter. So if your wall mounted button is at 900, ratgdo probably wont work at 128 since your button probably used up counter 128 - 900 already. If your button is at counter 25,999 and you set ratgdo to 26,000, ratgdo will work, but your wall button will stop working because it's going to be sending commands based off of counters that were spent. It will start working again once the internal counter value reaches an un-used value.

The safest place to start counting with ratgdo was 0 (I used the first 128 because I spent the early counters with version 1.0). If it works, then it will not interfere with your wall mounted button unless you somehow manage to trigger your door/light a million times.

So start there, if it doesn’t work we can try advancing the counter manually until it does work, but its also important to verify that your wall mounted button continues to work as well. If one works and not the other, it means the ranges are overlapping.

Hope this helps explain it...

mterry63, 2022-12-23:
Thanks for the details, I'll give it a shot.

mterry63, 2022-12-27:
@PaulWieland, I haven't had a chance to test yet but was re-reading all this today as I have an opportunity today to do so. 

Given your explanation above, how does ratgdo and the wall button not always have conflicts if they are competing for counters? I guess the same applies to transmitters in cars as well. Do they communicate the usage of counters to each other? It would seem there would always be a conflict, so I must be missing some critical detail.

mterry63, 2022-12-27:
Flashed the 2.0a alpha and connected but no love. How do you advance the code via MQTT? I'd have to get a ladder to connect to the serial monitor with my PC.

PaulWieland, 2022-12-27:
Two devices can’t occupy the same counter space. ratgdo starts at 128. Your wall opener probably starts at something much much higher. 

When you flashed 2.0 did you enable rolling codes in the setup? You can unplug the esp and check the serial monitor at your desk. It will output the counter number it’s on as well as the code it sent.

[22 more replies]

closed 2023-01-23 by PaulWieland