PaulWieland/ratgdo #4: Status of GDO working, Commands not working

mterry63, 2022-07-31:
GDO model Chamberlain HD520EVP mfg 08/26/15

MQTT Status following the general physical status of the door, closed, closing, open, opening, obstructed, cleared. Here are the logs begun from an open door position and going through close, then open from the arrow buttons on the GDO logic board and a second set of close then open from the pushbutton on the wall. Also walked through beam after exercising door. Wall button is a 882LMW. Door functions completely normally from buttons. ratgdo is powered from USB 5V supply with RPM1, RPM2, Control, Ground, and Obstruction cables connected. 

Here is the log during the above testing.
[esp-web-tools-logs.txt](https://github.com/PaulWieland/ratgdo/files/9229955/esp-web-tools-logs.txt)

PaulWieland, 2022-08-01:
As you pointed out, this is the same vintage as the GDO in this [issue](https://github.com/PaulWieland/ratgdo/issues/3), and I think they are related. Whats interesting is that you are using a 882LMW which is the same control panel I use. Presumably yours is older though, and I wonder if the firmware changed.

I'm also going to have a closer look at the signal timing on the older 882LM with my scope to make sure they are lining up.

PaulWieland, 2022-08-01:
Please test version 1.4a. ([see here](https://github.com/PaulWieland/ratgdo/issues/3#issuecomment-1201338076))

mterry63, 2022-08-01:
I've flashed version 1.4a and will test soon. I was wrong about my wall controller and GDO model, the wall controller is a Chamberlain 041A7185-1 and the GDO is a HD520EVP. The FCC dataplate is the same as the other user and the logic board is exactly the same. I should be able to post testing results within an hour or two.

mterry63, 2022-08-01:
No change when sending commands via 1.4a. Status changes to the command but the GDO does nothing.

PaulWieland, 2022-08-01:
Ok. I'm awaiting an older wall control so I can analyze it. Will get back to you as soon as I have had a chance to analyze it.

mterry63, 2022-08-08:
@PaulWieland I saw your note to the other user to test so I tried version 1.42a and no change. Commands do not work, status does work.

PaulWieland, 2022-08-08:
Thanks for the feedback. I think I’m going to have to get ahold of a 2015 logic board

mterry63, 2022-08-15:
My logic board is 45ACBR5.

PaulWieland, 2022-10-20:
Duplicate of #3 
I will close this one. Please subscribe to Issue #3 as I post updates there.

closed 2022-10-20 by PaulWieland