PaulWieland/ratgdo #8: How to find out RPM1 and rmp2 on other Boards

Bjoern3003, 2022-09-04:
Hi,

interesting Project.

Did you know, if there is a way to find out rpm1 and rpm2 on other boards than yours?

I have door opener ML1040EV from CHamberlain with yellow button and want to come back from cloud service.

PaulWieland, 2022-09-04:
Have a look at the module in your opener and check to see if it has a 6 wire harness connecting it to the logic board. 

This is what the module looks like: https://www.chamberlain.com/ac-travel-module-kit/p/041D7742-7

Bjoern3003, 2022-09-04:
I have took 2 pictures. Not looking like that. 😔

[image]
[image]

Scott--R, 2022-09-04:
Please note that if you're not in North America your opener most likely responds to simple contact closure commands. Your ML1040EV would be one of them. If you short the control terminals together you will most likely find that it works just fine. The encrypted technology as far as I know is only found in North American openers.

Bjoern3003, 2022-09-04:
Very sad ☹️ the project sounds interesting. But the "without soldering method" is not that, what I am looking for.

Scott--R, 2022-09-04:
I think perhaps you misunderstood @Bjoern3003 I meant that you should be able to control your garage door opener without requiring the commands from RATGDO. You may still be able to tap the RPM signals somewhere, but I just wanted you to know that in my experience, openers like yours can be operated by simple contact closure, making them much easier to work with.

Bjoern3003, 2022-09-05:
Sure, this is possible, but then I did not get the actual position back as response.

Maybe we can find more peaple to extend this project for other models.

jbbjarnason, 2022-09-17:
I have a similar case, with Chamberlain ML810EV. 
[image]

@Bjoern3003 Have you made any progress?

Bjoern3003, 2022-09-17:
Not really. But I am still interested in any solution. Your model looks same like my.

jbbjarnason, 2022-09-17:
@Bjoern3003 have you been able to connect it in a soldreless setup? I can't find any information on the pinout on this device.

Bjoern3003, 2022-09-17:
I have to disconnect the complete pcb for that. Sure I can. But I think I am not able to find out the correct pins for rpm1 and 2. 🙁

PaulWieland, 2022-09-17:
The pins of the rotary encoder (travel module) for my opener were labeled. Check the pcb and see. If not post a picture of it

Bjoern3003, 2022-09-18:
On the rotary side or pcb side? On pcb is it not labeled and or travel module does only have 3 pins.

PaulWieland, 2022-09-18:
On the rotary encoder side. If it only has 3 pins I don’t think it can be a rotary encoder. Do you have a multimeter to check the pins for voltage?

Bjoern3003, 2022-09-18:
Sure I have. Do I have to check during up and down running?

PaulWieland, 2022-09-19:
Can anyone find a description of what all the push terminals are for on the back of the opener? This model is nothing like the openers sold in the USA.

Scott--R, 2022-09-19:
So these are basically the Merlin openers. If you look up Merlin MT100EVO you'll find more details, but I've attached a screenshot of the terminals and put my own descriptions below:

0 serial communications port

1 and 2 are the wall control (similar to USA), but the strange thing is that even though it is a MyQ opener my experience has been that these types of openers (with push terminals that look like this) are directly compatible with contact closure on 1 and 2 and don't require anything special to control them. They respond to both types of inputs and the Merlin openers even ship with a green pushbutton that inserts right into these terminals so you can press that button on the opener directly to operate it.

2 and 3 are the obstruction sensors (same as USA)

4 and 5 have a jumper. You remove that jumper and insert wires for something called a "door connection"  or "door in door" feature.

6 and 7 are for a "flashing light" feature. Generally just leave empty.

[image]

PaulWieland, 2023-03-24:
Version 2.0 of the board eliminates the need for soldering.

Version 1.0 boards are incompatible with 2.0 firmware unless you solder a transistor to the board. I will send anyone with a 1.0 board a free transistor if they email me their order number. Instructions for the mod are forthcoming.

That being said, i don't think the door openers mentioned in this thread are security + 2.0 openers. I see no mention of it in the manual. They seem to support traditional dry contacts so theres no real need for ratgdo.

closed 2023-03-24 by PaulWieland

Bjoern3003, 2023-11-02:
New Try... its working with Version 2.0. I have it tried with the latest version of https://github.com/PaulWieland/ratgdo/tree/2.0-secplus1.0 and Secure 2.0

Red to Green (Pin 0)
White to White (Pin2)
Grey to Grey (Pin 3)