PaulWieland/ratgdo #11: mqttTopicPrefix note defined

joshpearce, 2022-10-13:
I'm probably missing something simple here, but this project will not build with `mqttTopicPrefix` undefined.

PaulWieland, 2022-10-13:
https://github.com/PaulWieland/ratgdo/blob/b338c5499409c234139959f25a29aaf788256a0e/src/ratgdo.h#L14

joshpearce, 2022-10-13:
Ah, ratgdo branch. Got it! Thanks.

closed 2022-10-13 by joshpearce

joshpearce, 2022-10-13:
Hey, couple other dumb questions here:
1) What is the best way to upload the firmware and then access the setting page? Do I just use "Upload and Monitor" from Platformio?
2) When I was using the released firmware, even if I didn't check the option to "Disable OTA & Web Server Config Access" I could still never again directly hit the IP with browser and bring up the config page. Is that a bug?