PaulWieland/ratgdo #15: Manual Home Assistant YAML

tony-fav, 2022-10-18:
I just wanted to share my HA yaml configuration for anyone else interested.

It's a bit more complicated than it could be. It would be significantly if the status topic was split into door_state, obstruction_state, and LWT. Additionally, if state publications were retained, it would know status on HA restart.

This created 4 entities: the cover itself, the light button, the obstruction binary sensor, and a button for the `query` command which is useful for getting the other entities back to an available state if HA restarts.

[image]

```yaml
mqtt:
  cover:
    - name: "RATGDO"
      unique_id: ratgdo
      command_topic: "ratgdo/command"
      state_topic: "ratgdo/status"
      availability:
        - topic: "ratgdo/status"
          value_template: |-
            {% if value == "offline" %}
              offline
            {% else %}
              online
            {% endif %}
      payload_open: "open"
      payload_close: "close"
      payload_stop: 
      state_open: "open"
      state_opening: "opening"
      state_closed: "closed"
      state_closing: "closing"
      payload_available: "online"
      payload_not_available: "offline"
      optimistic: false
      value_template: |-
        {% if value == "open" %}
          open
        {% elif value == "opening" %}
          opening
        {% elif value == "closed" %}
          closed
        {% elif value == "closing" %}
          closing
        {% else %}
          {{ states("cover.ratgdo") }}
        {% endif %}
      device_class: garage
        
  button:
    - name: "RATGDO Get Status"
      unique_id: ratgdo_status
      command_topic: "ratgdo/command"
      payload_press: "query"
      device_class: restart
      
    - name: "RATGDO Light"
      unique_id: ratgdo_light
      availability:
        - topic: "ratgdo/status"
          value_template: |-
            {% if value == "offline" %}
              offline
            {% else %}
              online
            {% endif %}
      unique_id: ratgdo_light
      command_topic: "ratgdo/command"
      payload_press: "light"
      icon: mdi:light-recessed
  
  binary_sensor:
    - name: "RATGDO Obstruction"
      unique_id: ratgdo_obstruction
      availability:
        - topic: "ratgdo/status"
          value_template: |-
            {% if value == "offline" %}
              offline
            {% else %}
              online
            {% endif %}
      unique_id: ratgdo_light
      state_topic: "ratgdo/status"
      payload_on: obstructed
      payload_off: clear
      value_template: |-
        {% if value == "obstructed" %}
          obstructed
        {% elif value == "clear" %}
          clear
        {% else %}
          {{ states("binary_sensor.ratgdo_obstruction") }}
        {% endif %}
      device_class: safety
```

PaulWieland, 2022-10-18:
I think splitting the topics up and adding retention are easy to do at this stage. It will disrupt NodeRED users, but then again NodeRED seems to be far more flexible than HA, so it's not a big deal.

My priority at the moment though is to get a solution for everyone who has a 45 series logic board. Right now nothing works for them.

tony-fav, 2022-10-18:
Sounds good. I'll take a look at publication. There's probably no need for a breaking change, could publish to .../status as normal but then .../status/cover .../status/obstruction .../status/LWT as well. Though may take some thinking with the LWT.

PaulWieland, 2022-10-18:
https://github.com/PaulWieland/ratgdo/wiki/Example-YAML-file-for-Home-Assistant

joshpearce, 2022-10-18:
Is this stuff kosher? There are a couple entries with two unique_ids.

tony-fav, 2022-10-19:
> Is this stuff kosher? There are a couple entries with two unique_ids.

A flub of the copy paste. Check now.

PaulWieland, 2022-10-19:
@tony-fav I guess to be consistent, it would be better to include examples in the github io pages instead of in the wiki.
[I had reserved a section for examples](https://paulwieland.github.io/ratgdo/05_homeassistant_example.html).

You can issue a pull request to update the docs if you want.

PaulWieland, 2022-12-22:
2.0 Alpha addes discrete MQTT topics for the various states in an effort to make home assistant configurations cleaner. [See here](https://github.com/PaulWieland/ratgdo/issues/3#issuecomment-1363192500) for instructions.

PaulWieland, 2023-03-24:
Home Assistant auto discovery is part of firmware 1.7 and 2.0.

closed 2023-03-24 by PaulWieland