PaulWieland/ratgdo #17: 50DCCl1MC Revision D board, where to solder?

wwsteele, 2022-10-20:
[image]
I purchased the kit and was excited to get it installed, and I just had a new garage lift installed, manufactured August of this year. The motherboard pictured is what is inside. I cannot locate the 3.3v power, and the rotary encoder connecter has 5 pins instead of 6. Hoping someone can tell me where to solder my 3 wires, I did turn the board over and I did not see a 3.3v label on the other side either. I can take a picture of the other side if needed. Thanks in advance if anyone can help me out!

PaulWieland, 2022-10-21:
A photo of the other side might help. It would also help to take the rotary encoder/travel module out and see if the pins are labeled on that pcb. The module has a plastic clip that holds it to the motor shaft.

wwsteele, 2022-10-21:
Thanks for the reply Paul. They have really changed the internals of this model.
[image]

Here is a close up of the 5 pins. I see RPM. But not RPM1/2
[image]

Back of the board
[image]

PaulWieland, 2022-10-21:
I'd like to know what the unlabeled 5th pin is:
[image]

Unless you have a scope or a logic analyzer its going to be tough to figure out. Rotary encoders require two pins for two hall effect sensors in order to determine direction. This might be something else.

wwsteele, 2022-10-21:
I do not. Well, I guess I'll do the no solder method then.

PaulWieland, 2022-10-21:
What's the door opener model? I'll see if I can order the module and reverse engineer it.

wwsteele, 2022-10-21:
Cool guy! This is an 8160W.

Zebble, 2022-10-25:
I have a similar board coming today (050DCTBMC).  It has a 5-pin encoder connector, as does my original 45DCBR1 which is why I thought I'd try the 050DCTBMC.  I'll throw a scope/analyzer on the new board when it comes, but @wwsteele would you be able to take a photo of where the wires go into the encoder module and where they go into the logic board?  Just so I can see the colours/wiring order and compare to what I have.  I'll post my findings.

wwsteele, 2022-10-25:
[image]
[image]
Cool! Here you go.  @Zebble

Zebble, 2022-10-25:
Thanks @wwsteele !  The board I currently have has exactly the same colours/pinout as yours so sounds like we might have the same setup.  I'm hoping to spend some time pinning things out this week/weekend and will post my findings.

It's interesting to note that the 050DCTBMC I just got in has what looks like an identical pin configuration to yours, but the board design is very sparse on the back.  Looks like they have quite a few different designs for the same board.

[image]
[image]

acvigue, 2022-10-26:
I also have a logic board with a 5 pin encoder input. All I've figured out is that RPM1 pulses when the door is moving and it seems to be identical no matter which way the encoder is turning. PP is normally high but gets pulled low three times, once near the top, middle, and bottom of the door travel. You can get enough 3.3V to power the ESP8266 from the center pin of the LM317 voltage regulator (circled in last picture below)

I was thinking if there was a way to monitor the motor direction by tracing the motor output connector all the way to the MCU's pins that control the relay if that makes sense.

Logic board is the 50DCBR5 it mostly looks identical to the one above.

[image]
[image]
[image]
[image]

(ignore the soldered wires, especially the red one as that was before I found the LM317 3.3v output)

acvigue, 2022-11-17:
@Zebble did you ever have the chance to work on experimenting with your board?

Zebble, 2022-11-18:
Hey @acvigue.  I haven't touched it yet, but do plan to pull out the old board this weekend to bench test and compare it with the new board.

I do suspect the lack of an RPM2 pin was an attempt to simplify their design, albeit by losing the fully closed loop of the other encoder.  They don't really need to know what direction the door is going in if they're already telling it the direction to go in, they just need to know if it's moving and some key endpoints help through the PP pin.  We may need to find a different method to determine what direction the door is moving in.

dorkage, 2023-01-06:
My 8500C looks like it has the same encoder.

[image]

I'm willing to help debug this. 

I have the pinout RPM, 63, 64, +5(VCC?), GND

[image]

It's a jackshaft opener from around 2018. Unsure of the revision of the board since I did not see any stickers.

dorkage, 2023-01-06:
Here's the 6Pin Encoder:

[image]

And the 5Pin Encoder:

[image]

Looks like the tests points for RPM2 are even missing. I might see if I can find a couple cheap ones and debug them on the bench and see if the 6pin can be swapped into the 5pin openers.

Zebble, 2023-01-06:
Interesting!  Just by looking at the back of the boards, you can see the 4-pin vs 8-pin configuration of what is very likely 1 or 2 phototransistor/LED pairs like this (https://www.mouser.ca/ProductDetail/Omron-Electronics/EE-SX1081).  The 6-pin board ties RPM1 and RPM2 to each of the two pairs where the 5-pin board only goes to the output of 1.  With 2 pairs, the 6-pin board is able to detect direction.

I suspect whatever is passing through the detectors is identical so this board or module should be a drop-in replacement.

Notice there is also another phototransistor/LED pair on the bottom right of the board.  That is tied directly to the PP pin.

I haven't had a chance to swap in the 5-pin 50-series board into my 45-series opener yet.  That still won't solve the RPM1 vs RPM1/RPM2 problem.

Has anybody taken a look at the MOT pin on the connector?  I wonder if it would indicate the current direction of the motor so we would only need RPM1?

dorkage, 2023-01-07:
The replacement encoder in my opener is "041A6408"

Chamberlain / Liftmaster refer to it as an "Absolute Encoder" and from looking into it, it seems it's a single channel rotary encoder with different sized pulses. So you'd basically have to compare one pulse to the last to determine direction.

My encoder looks easy enough to remove, so I might attempt that this weekend to see if the PCB matches the above photo I posted yesterday. I also ordered a logic module and have access to Dewesoft equipment which has "Super Counters" that would be able to extremely accurately capture the waveform coming out.
https://training.dewesoft.com/online/course/digital-counters#dewesoft-super-counter

Automayted, 2023-02-19:
@acvigue or @PaulWieland, has anyone been able to get ratgdo working with a 50DCBR5 board? Would love to order two kits ASAP, but need assurance they'll work with my Chamberlain B970 openers. Thanks!
[image]
[image]

acvigue, 2023-02-19:
Nope! not yet. I'm probing some points around the relay area to see if it's possible to get the direction of the door. The sensor only pulses RPM, at least for mine, it's the same waveform whether it's going up or down.

PaulWieland, 2023-02-19:
I have a fix for this coming soon, but unfortunately it requires a hardware change. It will be possible for those with 1.0 boards to retrofit a mosfet. 2.0 test boards are on the way and once I have them and have finished testing the 2.0 firmware I’ll have them up for sale. I’ll also mail anyone with a 1.0 board a mosfet + soldering instructions so that they can use to upgrade their shield if they wish.

Automayted, 2023-02-19:
> I have a fix for this coming soon, but unfortunately it requires a hardware change. It will be possible for those with 1.0 boards to retrofit a mosfet. 2.0 test boards are on the way and once I have [...]

Sounds great! Thanks, Paul!

Automayted, 2023-03-20:
Paul, any update on availability of v2.0 boards? Thanks!

PaulWieland, 2023-03-21:
There was an issue with the first batch. Waiting for the arrival of the second.

PaulWieland, 2023-03-24:
Version 2.0 is ready. I'm leaving them on back order since I'm going to be away for 2 weeks, but I will be shipping any orders received over the weekend on Monday.

1.0 boards can be retrofitted with a transistor to make them 2.0 compatible. This eliminates the need for soldering to the rotary encoder. Anyone who has a 1.0 board can email me their order number and I will send them a free transistor. Instructions for the mod will be forthcoming.

closed 2023-03-24 by PaulWieland

wwsteele, 2023-03-24:
Exciting news! Thanks for seeing this through. I cannot find my order number would you mind giving the part number of the transistor to purchase so I can go ahead and order one?

PaulWieland, 2023-03-24:
Whats the house number where I shipped the order to. I think I found your order, but just want to confirm.

wwsteele, 2023-03-24:
That would be 61

wwsteele, 2023-04-22:
Hi Paul, thanks I received the transistor. I checked the github and I do not see the instructions for the mod to version 2.0, unless I a missing it?

PaulWieland, 2023-04-23:
Hi Wes, sorry I haven’t had time to make instructions for it. Im returning home a week from today and should hopefully be able to get to it then. The gist of it is that the esp needs to read the 12v line. To do that we connect a mosfet in such a way that the signal from the 12v gdo signal line drives the mosfet, which then pulls esp’s input pin low (rpm2 iirc).Connect Source to groundDrain to rpm2Gate to red/signal inI will get a proper diagram and photo uploaded soon. Sent from my iPhone

wwsteele, 2023-04-23:
Thanks for the update. Appreciate the explanation and if I were just a
little smarter I'd get right on that ;). Looking forward to the diagram