PaulWieland/ratgdo #18: Obstruction detected when door is closed

chrispatin, 2022-10-26:
When the garage door is closed, and the light is turned off, the safety sensor led light goes off. This is detected as an obstruction. If the garage door is open or (closed with the light on) the safety sensors continue to work.

From the manual: 

```
NOTE: For energy efficiency the garage door opener will enter sleep mode when the door
is fully closed. The sleep mode shuts the garage door opener down until activated. The
sleep mode is sequenced with the garage door opener lights; as the lights turns off, the
sensor LEDs will turn off and whenever the garage door opener lights turn on, the sensor
LEDs will light. The garage door opener will not go into the sleep mode until the garage
door opener has completed 5 cycles upon power up
```

Three different doors exhibit this behavior. Liftmaster 8355 - (Manufactured: 2019)

tony-fav, 2022-10-26:
I have noticed some odd behavior of the obstruction sensor when I was using the light command. This makes it make a bit of sense now. Basically only trust the obstruction sensor is the garage is open?

chrispatin, 2022-10-27:
Configurable option for garage door openers with power saving mode? I modified the firmware to ignore obstructions detected when the door is closed. Seems to work well, and keeps the HA history from having false positives.

PaulWieland, 2022-10-27:
A couple issues...

- with reed switch installs since `reed_open` and `reed_closed` are up to the installer to decide where to mount the switch.
- with my personal install, I keep the gdo awake by sending it light commands when there is occupancy in the garage (I don't have the bulbs installed). This lets me break the beam 3x with my foot to open the door from the inside.

chrispatin, 2022-10-28:
I found some generic information which seemed to state what you mentioned. I tried below and it works.  Just experimenting, and removing one of the last cloud integrations I have left. 

`    if ((interruptTime - lastInterruptTime > 250) && (digitalRead(INPUT_OBST) == HIGH)) {`

mterry63, 2023-01-06:
Will this fix be implemented in future code? I'm seeing the same issue.

PaulWieland, 2023-01-06:
Yes it will be fixed. My focus has been on getting 45 series openers working.

mterry63, 2023-01-06:
Great, looks like you have cracked the code on the 45 series openers, great work. Looking forward to future updates.

PaulWieland, 2023-01-17:
This is fixed in version 1.6a. I will be releasing 1.6 soon (should be identical to 1.6a unless I missed a bug somewhere). 
Anyway feel free to install 1.6a and test.

mterry63, 2023-01-19:
@PaulWieland I flashed 1.6a with the web installer on the device I had manually flashed 2.0a on. It is working and controlling fine. Can I also flash 1.6a on the device that has the 2.01a code on (higher code count)?

mterry63, 2023-01-19:
There appears to be a bug (I saw it also on the 2.x code) where the status says obstruction = obsobstructed (obs is repeated twice). Unless it's a bug in MQTT Explorer.

PaulWieland, 2023-01-19:
> @PaulWieland I flashed 1.6a with the web installer on the device I had manually flashed 2.0a on. It is working and controlling fine. Can I also flash 1.6a on the device that has the 2.01a code on (higher code count)?

Yes you can. 1.6 supports the entire counter range.

PaulWieland, 2023-01-20:
> There appears to be a bug (I saw it also on the 2.x code) where the status says obstruction = obsobstructed (obs is repeated twice). Unless it's a bug in MQTT Explorer.

Found it and fixed it. Thanks 👍

mterry63, 2023-01-20:
@PaulWieland I flashed the device that previously had 2.01a with the web flasher v 1.6a and it is also working fine.

PaulWieland, 2023-01-23:
Fixed in 1.6

closed 2023-01-23 by PaulWieland