PaulWieland/ratgdo #19: Rotary Encoder Option - Detection of door state on startup

chrispatin, 2022-10-30:
After a power cycle, the value for the doorState is defaulted to 'reed_closed' (and published on MQTT), and will not reflect the physical state until the door is opened / closed. Any options to detect the physical door state on startup via the hardware?

Curious how the GDO achieves this, is this just an incremental rotary encoder?

PaulWieland, 2022-10-31:
The GDO just saves the last state of the door. While possible to do, it didn't really make much sense to do the same thing with ratgdo since the door could move with ratgdo powered off, and it could be wrong anyway.

And no, there isn't any way to know what state the door is in until it starts moving and pulses from the rotary encoder can be counted.

chrispatin, 2022-10-31:
The above ask was more geared towards testing changes with HA using message retention, and reducing the probability of invalid states. (stale MQTT data experienced during MQTT connectivity issues, ratgdo restarts, etc)

Still an automation rule based on door state probably should verify the target door state after the rule executes, but reducing the chance of incorrectly firing the rule is desired.

Thanks!