PaulWieland/ratgdo #20: Board 49DCBC3 Status

Cossid, 2022-11-10:
I see there are notes of boards starting with 45 or 045 not working whereas 50 or 050 work.  Any idea where a board labeled 49DCBC3 might stand?  It resides inside a Craftsman branded rebadge, mode 049DCTWF.

Unfortunately, I was just doing a preliminary check on an already installed unit to see about compatibility, so I don't have great pictures and was not able to remove the board or see the other side, but here is what I was able to snap with my phone:
[image]
[image]

PaulWieland, 2022-11-10:
That's a new number series, probably because its in a craftsman branded opener.
Does it have onboard wifi?

Cossid, 2022-11-10:
Yes, it has onboard wifi

[image]

Interestingly, the fcc label for the board appears shared with a LifeMaster version, and that one has a red PCB with a 50DCxxxxx label: https://fccid.io/HBW1D8169-1/Internal-Photos/Internal-photos-3438872

PaulWieland, 2022-11-10:
So from what I can tell based on a limited sample size, the distinction between the two is that 45 boards do not have wifi while 50 boards do. The sync commands are not accepted by the non wifi boards. Regardless, I do have a fix in the works that will make ratgdo work with both.

Cossid, 2023-01-23:
Sorry, I didn't mean to leave this open, and thought I had closed it.  Currently winter and have no ambition to try this in the cold.  I bought a couple boards and will give it a try in the spring/summer and follow up then.  I have one 49 and one 50.  I suspect 49 is more akin to 50 than 45 anyway.