PaulWieland/ratgdo #22: Garage Door Lockout Ability?

abishur, 2022-11-24:
I was about to install a couple of your board on the garage door openers at my mom's place so I don't need a new special garage door opener when she told me she wanted to swap out the garage door openers with ones that had a "lock out feature"

Apparently the idea here is that when she's home she can turn on a "lock out" mode such that if someone broke into her car and pushed the garage door opener button it would ignore the command and stay shut.

Do you think that's something that could be accomplished with your device?  It's a long shot here but I was thinking on a software level if the serial stream was flooded with garbage any door opener would fail?  Lacking a proper understanding of how these doors handle security I have no idea if I'm on the right track or not with that line of thinking.

A more practical solution if that's not viable was I was planning on just adding a smart outlet to turn cut power to the openers.

What are your thoughts on this?  Is a software solution something practical or am I grossly misunderstanding how it works?

Scott--R, 2022-11-24:
This may be possible if Paul can figure out the “vacation mode” or “lock” command. The Chamberlain wall controls are able to put the openers into a mode where they will ignore wireless remotes.

abishur, 2022-11-24:
Lol, I talked to my mom today and when she said "I'm getting a new garage door opener" she meant the button with the lock command.

Once she gets it, if there is a way I can have the program watch the commands sent let me know so I can help out

PaulWieland, 2023-03-24:
Lock/unlock has been added to version 2.0. 

Version 1.0 boards are incompatible with 2.0 firmware unless you solder a transistor to the board. I will send anyone with a 1.0 board a free transistor if they email me their order number. Instructions for the mod are forthcoming.

closed 2023-03-24 by PaulWieland

abishur, 2023-05-17:
I did the most to get my version 1.0 board to 2.0 and installed the 2.0 firmware.  The garage door is controlling fine, but I don't see the lockout option in home assistant, how do I enable that?

PaulWieland, 2023-05-17:
Setup a button that sends the mqtt commands for lock and unlock. There is not auto discovery for the lock as I believe it to be misleading since it doesn't actually "lock" the door. It just tells the GDO to ignore the wireless remotes.

abishur, 2023-05-18:
Awesome!  I added an MQTT switch that wrote lock/unlock to the command/lock topic and it works like a charm.

And yeah, like you said, it's not actually "locking" anything.  I can still open it from the local door control button or from Home Assistant, but it disables remote entry though any wireless door opener.  And that gives my mom peace of mind that no one can get into the house by breaking into her car and opening her garage door so it's well worth it to me!

Thanks for all the help!

calisro, 2023-11-08:
> Awesome! I added an MQTT switch that wrote lock/unlock to the command/lock topic and it works like a charm.
>
> And yeah, like you said, it's not actually "locking" anything. I can still open it from [...]

What did you use for your configuration for hte mqtt lock?

hacsjalano, 2023-11-09:
@calisro *Thank you*  I have been searching all over for how to do this using the official ratgdo firmware!  Your example should be part of the documentation!!

leptians, 2023-11-11:
Thank you @calisro , with your code, I'm able to add mine. I wish it's supported by default since it's also available in the ESPHome version (among other attributes). I labeled mine as `garage_remote_lock`, and I think people should be aware of its scope. It's a very useful feature to have
@PaulWieland , thank you for the ratgdo and I'm happy to finally able to move away from MyQ. But, please re-consider adding this officially in the MQTT version :)