PaulWieland/ratgdo #23: Documentation: Overview of reverse engineering steps

chris1h1, 2022-11-30:
Is it possible to document a summary of the steps involved in reverse engineering the data that must be transmitted to the garage door opener's logic board in order to open/close the door, and to toggle the light, and implementing this for Arduino (I think line 79 onwards of [ratgdo.h](src/ratgdo.h))?

I have an older Craftsman 139.18405D garage door opener featuring a 41DB102-2 logic board, with a 41A6317 wall mounted control panel. The documentation for this mentions "Security+" rather than "Security+ 2.0". The 41A6317 wall mounted control panel is discontinued and replaced by the LiftMaster 889LMC, which includes the note *"Not Compatible with Newer Security+ 2.0 garage door openers [...]"*.

So far I have found [https://github.com/argilo/secplus](https://github.com/argilo/secplus), but don't know where to begin for an Arduino implementation.

PaulWieland, 2022-11-30:
Are you sure that its not a simple dry contact switch?
Unhook the wires from your wall mounted button and short them out together. If the door opens, then you have a "good" garage door opener that works the way it should. You only need a simple relay board to control the (like a shelly one).

chris1h1, 2022-12-01:
Thanks for your reply. The wall mounted control panel has three buttons (open/close, light toggle, and lock), but the wire from it to the garage door opener only has two conductors. So while I haven't yet tested simply shorting those wires, I'm inclined to suspect that the communication between the two devices has something more interesting going on. It's not just a simple momentary button.

Scott--R, 2022-12-01:
I confirmed your model is Security+, which can be opened / closed using simple contact closure. Light and lock are different, but open and close simply touch the wires together.

chris1h1, 2022-12-01:
Ah ok, thanks. I should have actually tested this before making assumptions and creating this issue. Thank you both for your time.

PaulWieland, 2022-12-01:
If you want to reverse engineer how it works, buy a usb logic analyzer ($15) and use it with sigrok pulseview. 

This button probably works by sending a long continuous low pulse to open the door and a shorter series of low pulses to turn the light on or lock the opener. 

Those pulses can be recreated with ratgdo hardware using modified firmware.

Scott--R, 2022-12-01:
I agree with Paul. That's most likely how it works and his recommended approach is bang on.

I use this one https://www.amazon.com/Comidox-Analyzer-Device-Channel-Arduino/dp/B07KW445DJ

chris1h1, 2022-12-01:
Thank you both for your help with this; I'll close this issue.

closed 2022-12-01 by chris1h1