PaulWieland/ratgdo #28: Unreliable operation with V1.6a and 45 series control board

mterry63, 2023-01-23:
Hello all,

Since I've installed version 1.6a on a second garage door opener I've noted unreliable operation of the opener. When opening or closing, mostly via wall button or car remotes, the door will not fully open or close and frequently reverses. The interesting thing is that when reversing, the lights do not flash indicating an obstruction, the door simply changes directions. I've had to physically hold down the button on some instances to actually get the door to fully close or open on some occasions.

This is with ratgdo soldered for all connections - power, control, ground, obstruction, rpm1, and rpm2. Initally I though the 3.3v connection may have been drawing too much current and confusing the logic board, so I disconnected 3.3v and installed a USB power connector directly to the 8266. There was no change in the operation after that. I then disconnected power all together but left the powered-down board physically connected. This resolved the issue, the opener returned to reliable operation. It has worked reliably over the weekend without any of the symptoms when ratgdo was powered up.

I installed this board on a second opener 1 day before installing version 1.6a, so I can't really say how reliabily it worked with 2.0a. Not enough time. This is my most frequently used garage door, a 2 door opener with 2 active cars kept in the garage that generally move daily. I have a second opener that has had ratgdo installed longer, but it is rarely used. It was upgraded to 1.6a on the same day having previously had 2.01a installed.

Am I the only one experiencing this issue?

PaulWieland, 2023-01-26:
@mterry63 My 45 series test rig is just a logic board on my bench without any door attached - so the scenario you are describing was occurring without my knowledge. I improved my test rig to be able to better determine if the door fully opened/closed.

At this point I *think* that the issue is being caused by older door openers not liking transmitters which have an ID of 0. I added a non zero transmitter ID to ratgdo and published it as version 1.61

Sorry that you are the guinea pig for this, but I really do appreciate your patience and feedback!

mterry63, 2023-01-27:
@PaulWieland I can appreciate the difficulty of supporting a wide variety of opener models. I'll try version 1.61 and see if it helps my situation. Thanks for looking into this. I'll report back in a day or so.

mterry63, 2023-01-27:
I went ahead and flashed version 1.61, but no difference. Closing commands by MQTT resulted in the door going halfway down and back up twice in a row.

PaulWieland, 2023-01-27:
What’s the behavior of the wall mounted button? Anything weird? does the led flash? Do you have to click the button more than once to get the door to work correctly?

and are you certain it’s not an issue with the obstruction sensor?

PaulWieland, 2023-01-27:
One more thing I thought of - open the serial monitor (called logs & console on the ESP Web Tools page), then reboot ratgdo. Verify the version number that ratgdo spits out when it boots up and make sure it says 1.61

It should output version 1.61 followed by 6 reboot/sync codes.

```Setup Complete
 _____ _____ _____ _____ ____  _____ 
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 1.61
Syncing rolling code counter after reboot...
get rolling code for 189 : reboot1
55010040269B7D34D36DB6A035FA59B1A9E9A0
get rolling code for 190 : reboot2
550100961FEFB6DBEDF7FA4A29B4B36A1ECF7D
get rolling code for 191 : reboot3
550100952DA4DB6D369A694032D1052371FDB7
get rolling code for 192 : reboot4
550100141F7FF7FFFFBEFD51164B7E948D1493
get rolling code for 193 : reboot5
550100120D349EC924D269420D2FDA4E8CD279
get rolling code for 194 : reboot6
5501006900120244120804A111688094A98483
Write successful
```

mterry63, 2023-01-27:
@PaulWieland I pulled it down, powered it up and connected to the serial port. Pressed reset and got the same output as you:
[14:44:12] _____ _____ _____ _____ ____  _____ 
[14:44:12]| __  |  _  |_   _|   __|    \|     |
[14:44:12]|    -|     | | | |  |  |  |  |  |  |
[14:44:12]|__|__|__|__| |_| |_____|____/|_____|
[14:44:12]version 1.61
[14:44:12]Syncing rolling code counter after reboot...
[14:44:12]get rolling code for 211 : reboot1
[14:44:12]550100820B2D82CB649259421D66FACEA81A7C
[14:44:12]get rolling code for 212 : reboot2
[14:44:12]550100511A4D36934D3491211048109489E493
[14:44:12]get rolling code for 213 : reboot3
[14:44:12]5501005037DB7FB7DF7DB6120834414E45726B
[14:44:12]get rolling code for 214 : reboot4
[14:44:12]550100A61B69B6D34D369781140B4C17A93490
[14:44:12]get rolling code for 215 : reboot5
[14:44:12]550100A525B6D265965B2C622B25F74E56B279
[14:44:12]get rolling code for 216 : reboot6
[14:44:12]55010040369B4F76D36DB61117CBBE9129749E
[14:44:12]Write successful

I am also using an 882LMW wall switch. I didn't look at the switch while the opener moved, I'll check it now. I believe after the 2 attempts last night where the door went back up after halfway down I had to press the button twice. I'm fairly confident there is no issue with the obstruction sensor as the lights don't flash, no relay clicking, and if I pull power to ratgdo, everything works normally and consistently. 

I'll plug it back in and try some more tests since I'm the only one home at the moment.

mterry63, 2023-01-27:
OK, powered it back up and pressed the button, opened fully.
Pressed again, went about 2 feet closed and opened again. Wasn't watching the button.
Tried about 6-7 time open/close, and it worked fine. Wall button flashes 3 times during an open/close cycle.
One of those cycles was from the Home Assistant app, open and close.
Decided my neighbors were going to think I'm nuts and stopped there. Will let it go for the weekend unless it happens again and get intolerable. :)

PaulWieland, 2023-01-27:
> Pressed again, went about 2 feet closed and opened again. Wasn't watching the button.

If the door started closing, stopped, and then reversed open - that indicates there is an obstruction being detected.
Try unhooking the obstruction wire from ratgdo to rule out it interfering with the obstruction signal.

mterry63, 2023-01-28:
@PaulWieland I removed the obstruction connection from the ratgdo board and taped it off. Closing the door from the wall button resulted in the same results I've seen, door closed for 2 feet and reversed back to open. Removing power from the ratgdo board returns everything to normal without these symptoms.

abishur, 2023-01-29:
I have the same 45A board and am experiencing the same issue. I am also using the ratgdo board.  I left the breakout board in place and only removed the esp from the ratgdo breakout and the garage door closed like usual.

I was watching the mqtt topic while the esp was cpnnected and was getting an obstruction notification every few seconds when the board was connected to the garage door

PaulWieland, 2023-01-29:
What model openers do you guys have? I think I need to buy a complete used opener in order to recreate this problem. Sent from my iPhone

abishur, 2023-01-29:
I have the LiftMaster Security+ 2.0 Model number 41D7675 with the 45ACCL1 board

PaulWieland, 2023-01-29:
I have a hunch that having ratgdo wired to the rpm sensor is interfering with the openers interpretation of the signal. This is causing it to detect a pinch, which stops and opens the door. 

Try unhooking rpm1/2.

abishur, 2023-01-29:
I'll give that a go in a couple of days when I next get a chance.  What does RPM1/2 do?

PaulWieland, 2023-01-29:
> I'll give that a go in a couple of days when I next get a chance. What does RPM1/2 do?

They are the signal lines of the rotary encoder which is used to tell the position and direction of travel of the door. 

ratgdo can control the door without it, it just won’t know the door position. 

The goal here is to verify that this is the issue and then I’ll come up with a solution. 

My 45 series board doesn’t have this problem, but I am using a rotary encoder module from a much newer opener. It’s possible that other encoders can’t produce enough current on their signal pins to drive the esp8266 digital pins in addition to the i/o pins on the gdo logic board.

PaulWieland, 2023-01-29:
This is a admittedly a long shot, but if my theory about the rotary encoder is correct, I think I can reduce the required current by disabling the ESP8266's internal pullup resistors for RPM1/2. It might be just enough to make things work.

I just published 1.62a which disables the pullup resistors. This version will not work with a reed switch.

If 1.62a doesn't fix (or improve) the problem, then go back to 1.61, unhook RPM1/2 from the screw terminal (tape or isolate them up so they don't short on anything) and test again to help me verify my theory.

If you use 1.62a without RPM1/2 being connected to anything, you might get garbage door status messages because those pins are floating.

mterry63, 2023-01-29:
@PaulWieland My opener model is HD520EVP built on 8/26/15. The logic board is 45ACBR5. I remember getting RPM values from early versions of the firmware way back, they may be posted here somewhere. I haven't noticed any issues getting the correct values from the rotary encoder. I have a second ratgdo that doesn't seem to have this problem and it's the same model, build date, and logic board. It's running 1.60 right now. The only real difference for me is that all the wires are soldered on in the problematic unit and are bunched together exiting the opener to the ratgdo mounted above the opener. One thing I will try is separating the wires more. Not sure if close proximity can be causing digital noise on other leads.

I can try your suggestions and firmware today and report back. Like I mentioned, the last time I tried and had the issue I had the obstruction wire removed completely.

PaulWieland, 2023-01-29:
Twist the wires together to reduce noise.

abishur, 2023-01-29:
My wires are one of those thin rainbow wires all stuck together, I've only pulled apart the endings to land the individual points when I soldered them on.  I'll try pulling them apart and twisting them when I get a chance later this week.

mterry63, 2023-01-29:
@PaulWieland I took my opener apart and see I nicked the white wire with a likely short to the case. I put heat shrink on it and rerouted. So far, so good. Apologies if this winds up being my fault with a poor install. I'll let you know after a few exercises of the door, it's frequently used.

abishur, 2023-01-30:
> @PaulWieland I took my opener apart and see I nicked the white wire with a likely short to the case. I put heat shrink on it and rerouted. So far, so good. Apologies if this winds up being my fault with a poor install. I'll let you know after a few exercises of the door, it's frequently used.

Did you also twist the wires together when you put on the heatsink?  I'm fairly confident I didn't nick my wires when I installed it and used the provided soldering points picture for the 45A board when I soldered everything up, so I'm... mostly sure I got everything on the right spot.

PaulWieland, 2023-01-30:
I haven’t verified the 3.3v source on the 45 board - I will do that later today. The esp could also be causing a brown out to the rotary encoder. 

try powering the esp with usb for now.

mterry63, 2023-01-30:
I didn't twist the wires after I saw the exposed copper. I rerouted all the wires to the corner of the opener instead of the side, which eliminated the pinching that was occuring on the side. There is a natural gap in the steel chassis where they fold the metal to form short sides where the side panels and cover are screwed on. So far it appears I no longer have an issue.

abishur, 2023-01-30:
Interesting, I'll have to double check my setup, maybe the wire I used was too thin a gauge?

I placed the ratgo board next to the 45A controller board and used some circuit board tape to tape the wires down to the board (traveling left to right) over to the ratgdo.  Maybe with them going across the board I'm getting some signal interference?

PaulWieland, 2023-01-30:
> Interesting, I'll have to double check my setup, maybe the wire I used was too thin a gauge?
>
> I placed the ratgo board next to the 45A controller board and used some circuit board tape to tape the [...]

Definitely possible to pick up interference especially if they are near the relays or wires that drive the motor. 

You also have to be careful not to pinch the wires between the metal frame and plastic cover.

PaulWieland, 2023-02-02:
Going to close this. I'm 99% certain the issues described were being caused by interference on the RPM1/2 wires either due to routing or short circuiting. This interference caused the GDO itself to think the door had stopped prematurely, which it interpreted as a pinch/crush and thus stopped the door.

closed 2023-02-02 by PaulWieland