PaulWieland/ratgdo #31: I can't seem to be able too connect to wifi on the ESP8266 D1 Clone

Cookie-Monster-Coder, 2023-02-27:
Logs just keep trying to connect.
```
.Wifi attemps= 70
.Wifi attemps= 71
.Wifi attemps= 72
...
```

PaulWieland, 2023-02-27:
The credentials are invalid or the ESP8266 is outside of wifi coverage. Make sure you used the right case when entering both SSID and Password. unplug the ESP from the shield and power it on nearer to your wifi access point to try and rule out a coverage issue.

Cookie-Monster-Coder, 2023-02-27:
I changed the width from 40 MHz to 20 MHz of my 2.4 GHz and that seems to connect. Before this I used to get this error in my AP's logs.
```
daemon.notice hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
```

When I changed it the ratgdo connects but the signal is  ```-91/-93 dBm``` with ```RX Rate of 1.0 Mbit/s, 20 MHz``` and ```TX Rate of
9.0 Mbit/s, 20 MHz``` and the webpage just hangs for a bit and then goes to "The connection was reset". Router is about 15ft away from the ratgdo.

Cookie-Monster-Coder, 2023-03-04:
Would I be able to return just the ESP8266 D1 Clone back to you?

The signal on this is very bad. I need to be like 2 ft from my router. I have a Nighthawk R7800 and able to pick up a signal half way down my block on my phone, so It can't be the router. The antenna on the ESP8266 is very weak. I have a ESP-WROOM-32 ESP32 in the basement (with thick concrete walls) connected to a sump pump monitor and still get -68 dBM.

PaulWieland, 2023-03-06:
This isn't the right forum for this as github is for tracking software bugs.
But go ahead and ship it back and I will refund your order. Please write your order number on the package so I know which one to refund.

closed 2023-03-06 by PaulWieland

Cookie-Monster-Coder, 2023-03-10:
I'm sorry, I found this GitHub repo from google so I thought it was the only place for communication.

I ordered a new d1 mini clone from amazon couple of days ago, flashed it today and the wifi is way stronger then the one you shipped me with the shield. I'm able to place it in the garage and get around -63 dBm.

I ordered the shield with the d1 mini combo since I don't have a solder iron and thought it would be a piece of mind everything working out of the box. I have a soldering kit coming tmrw. I'll see how that goes.

I'm now 100% sure it was a bad d1 mini that was sent. I threw away the package that the shield & d1 mini came in, so I won't be able to pull your address from from that. Would you mind sending me a DM on [home-assistant.io](https://community.home-assistant.io/u/CookieMonster) with your address?

nicholashead, 2023-10-17:
@Cookie-Monster-Coder having the same issue here. I haven't been able to get the ESP8266 to connect to my wifi. I'm pretty technically savvy, so I don't think it's anything I'm doing. Did a replacement D1 fix it for you?

Cookie-Monster-Coder, 2023-10-17:
> @Cookie-Monster-Coder having the same issue here. I haven't been able to get the ESP8266 to connect to my wifi. I'm pretty technically savvy, so I don't think it's anything I'm doing. Did a replacement D1 fix it for you?

Yeah, I initially I bought the combo D1 and shield, since I didn't want to solder my own. It wouldn't connect. I even connected it to my phone hotspot to see how far it would connect, It was like 2 feet before it lost the signal. I ordered one from Amazon for like $9, soldered it, flashed it, and it instantly connected.