PaulWieland/ratgdo #32: Covert v1 board to v2

rafaelmathieu, 2023-03-30:
I recently bought a v1 board and never got around to installing it. I saw that a v2 was now available. Anyway way to convert it to v2?

abishur, 2023-05-17:
Here's the instructions I got for doing this and it worked for me.
You will need a mosfet with a gate threshold of >2, I used the 2n7000

Then connect:

source to ground
Gate to Red/gdo signal
Drain to RMP2

[image]

That's how I had it wired to the back and I affixed the mosfet under my ESP with some tape I use on circuit boards 

[image]

Then I placed it inside the garage door housing affixing it with some 3M tape, but in hindsight this means when I want to do a firmware update I need to open up my garage door unless OTA works?

[image]

Regardless I followed the soldering method for my garage door revision and it works just fine.  I am have some odd disconnect issues, but I think those are resolved in v2.1

PaulWieland, 2023-05-17:
Here's my install of a retrofitted 1.0 board. It's critical to use a mosfet with a gate threshold slightly more than 2v.

[image]
[image]

closed 2023-05-17 by PaulWieland

rafaelmathieu, 2023-05-17:
Perfect thanks for the update

PaulWieland, 2023-05-19:
I realized I made a mistake in my previous post... **drain goes to RPM2, not RPM1**. My board is running an unreleased firmware and it stopped working when I flashed it to 2.1 ;)