PaulWieland/ratgdo #34: v2.0 - Controlling the light via MQTT causes erroneous obstruction detected events most of the time

Automayted, 2023-04-02:
When publishing `on` or `off` to `/command/light`, I get an immediate sequence of `obstructed` and `clear` payloads received from `/status/obstruction`. When not toggling the light, the are no erroneous obstruction events published. Oddly, this doesn't occur at all times but is rather consistent (90%+) on both my openers. 

Maybe worth noting, using the LCD configuration panel on my openers, I've disabled light activation via the PIR motion detectors built into the control panel as well as disabled light activation from the obstruction sensors.

I'm running the latest code and my boards are USB powered, connected to two Chamberlain B970 openers (50DCBR5 mainboards).

Let me know what other info I can provide to assist in troubleshooting.

Thanks!

PaulWieland, 2023-04-14:
Does it do this every single time or only when waking the GDO from sleep/power save mode?

PaulWieland, 2023-04-14:
What about when you toggle the light using the wall mounted button?

iceman73, 2023-05-03:
I have the same issue and I can confirm when I toggle the Light from the wall mounted switch I also see the Obstruction detected and cleared message just like using the MQTT command for the light.

Automayted, 2023-05-06:
v2.1a seems to have resolved the issue on both my openers. Thanks!

Will future releases be available to download, so I can flash OTA vs directly-connected serial?

PaulWieland, 2023-05-08:
Considering this fixed in 2.1a. Still working on an issue with availability before releasing 2.1.

Automayted, 2023-05-08:
v2.1a resolved this for me, but the issue has returned with slightly different behavior in v2.1b. In v2.1b, every single light on command for both of my openers results in false Obstruction Detected and Obstruction Cleared status messages. Light off commands do not result in any erroneous messages, only light on.

[image]

PaulWieland, 2023-05-08:
this shouldn't be possible since the motion detection code is identical between the two. but just in case i put 2.1a & b back and published 2.1c (same as 2.1b + support for motion detection hardware) if you want to try and compare.

Automayted, 2023-05-19:
Checking back in on this. Is there any other testing or log captures that I can provide to assist with resolution?

PaulWieland, 2023-05-19:
It could be some sort of noise or interference with the obstruction signal on your model opener that occurs when the light relay is tripped. It’s hard to know for sure without a scope or signal analyzer

Automayted, 2023-05-19:
Thanks, Paul. That makes sense, feel free to close this. 

If I find time to drag my scope out to the garage to collect data, I'll reopen the issue.

PaulWieland, 2023-05-19:
We can leave open, i dont think you are the only one. I'm just unable to reproduce it. The fact you have a scope means we have the means to solve this... so when you have time, capture some data and share it here. 

Hopefully you have a two channel scope.

Connect ch1 to red ctrl line. set a trigger on that channel so it trips when you toggle the light.
Connect ch2 to black obs line.
Toggle the light and see what comes out on ch2.

iceman73, 2023-05-23:
With2.1 I still get the triggered Obstruction when changing the light as well.

Automayted, 2023-05-24:
@iceman73, is this also somewhat sporadic for you, or is it consistent for all light on/off commands?

iceman73, 2023-05-25:
Consistent
[image]

PaulWieland, 2023-05-25:
There are two possibilities I can think of:

* there is noise being picked up by the shield.
* the GDO itself is interrupting the obs signal when the light bulb relay is toggled.

Both issues can be solved in software but I need some data logging from a scope or signal analyzer to know exactly how to deal with it.

On my opener, the light is turned on as soon as you break the beam (so if the door was left open and the light has shut off, then you walk through the door way the light turns on). That means I cant just skip sending obstruction messages when the light is toggled because it would interfere with a common real world scenario where the obstruction state changes with the light state.

Automayted, 2023-05-25:
I'll should have time this weekend to bring my scope out to the garage and capture data as you requested [here](https://github.com/PaulWieland/ratgdo/issues/34#issuecomment-1554997031).

Thanks for the awesome support, Paul!

iceman73, 2023-05-25:
> There are two possibilities I can think of:
>
> * there is noise being picked up by the shield.
> * the GDO itself is interrupting the obs signal when the light bulb relay is toggled.
>
> Both issues can be [...]

This is how my system works as well

Automayted, 2023-06-11:
I migrated to https://github.com/ESPHome-RATGDO/esphome-ratgdo and I'm unable to duplicate this issue since. Closing.

closed 2023-06-11 by Automayted

DuelCoreF18, 2023-10-16:
@Automayted Are you running the Ratgdo board just fine via EspHome? I also have 2x Chamberlain B970 openers manufactured 10/02/20. Do I need to check the mainboard to ensure compatibility and if so any notes on how/what/where I'm looking?

PaulWieland, 2023-10-16:
If you have an overhead mounted security + 2.0 opener (yellow learn button), then it's supported. All chamberlain openers made since somewhere around 2015 are security + 2.0.

There was an issue with some jackshaft mounted openers using the older wireline protocol that is being addressed.

Automayted, 2023-10-16:
@DuelCoreF18, Ratgdo V2 boards with an ESP8266 running ESPHome has been absolutely flawless with my two B970 openers. I can't recommend this solution enough.

DuelCoreF18, 2023-10-16:
> @DuelCoreF18, Ratgdo V2 boards with an ESP8266 running ESPHome has been absolutely flawless with my two B970 openers. I can't recommend this solution enough.

Thanks for the feedback!