PaulWieland/ratgdo #35: Feature Request: Is it possible to get motion events from the PIR motion sensor built into my wall control panel?

Automayted, 2023-04-02:
My wall control panels have integrated PIR motion sensors used to trigger the GDO lights when motion is detected. Is it possible to get events published over MQTT when the integrated sensor changes state?

Wall control panel for reference:
https://www.chamberlain.com/wall-control-panel/p/041A7305-1

Thanks!

PaulWieland, 2023-04-14:
@Automayted it should be possible to do this with version 2.0, but I need you to capture the messages being sent when the motion sensor is tripped and send them to me, and then be willing to do so testing 😉

If you open the ratgdo serial log and then trip the motion sensor you should see some traffic being sent on the serial log. Copy and paste the log and send it over and I'll have a look.

Automayted, 2023-04-14:
> @Automayted it should be possible to do this with version 2.0, but I need you to capture the messages being sent when the motion sensor is tripped and send them to me, and then be willing to do so [...]

Awesome! I should have this data to you over the weekend.

Thanks, Paul!

Automayted, 2023-05-06:
> @Automayted it should be possible to do this with version 2.0, but I need you to capture the messages being sent when the motion sensor is tripped and send them to me, and then be willing to do so [...]

Hi Paul,

Sorry for the delay. The only thing seen in the log when the PIR activates is copied below. This exact same message is logged for each PIR activation.

`55010096114091DCEDF7FB6111CB211E3FB414`

This same exact message appears also about 1s after the PIR is activated, leading me to believe the same message for motion detected and motion cleared is used.

Thanks for considering this feature request and please reach out if I can assist with any other data capture or testing.

PaulWieland, 2023-05-06:
Are you sure it’s exactly the same both times? Sometimes only one byte is different. Paste both if you can.

Automayted, 2023-05-06:
```
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
55010096114091DCEDF7FB6111CB211E3FB414
```

Here's the exact log

PaulWieland, 2023-05-08:
So indeed its recycling the same counter and there is no "clear" message coming from the motion sensor. ratgdo can easily send a motion detected mqtt message, but the question becomes if/how/when to clear the motion detected status.

If we setup a motion device in home assistant and we only send it messages when motion is detected, then its state will always read as if motion is detected.  I could send a "clear" message after an arbitrary amount of time has passed since the last motion detected state, but I'm not sure if this is really the right thing to do.

There is also the problem of auto discovery since not everyone has a motion sensor, ratgdo can only send the auto discovery message when the first motion detected message is seen.

Or I can leave HA out of this all together and just publish an mqtt message when motion is detected and let the user do what they want with it...

Automayted, 2023-05-08:
Paul, thanks for taking a look at this.

I agree on keeping this out of auto discovery.

Also, I agree that arbitrarily sending a motion cleared message isn't the best solution. Instead, because the messages are sent in pairs, could ratgdo somehow track each motion message and then send motion cleared on the second message? A timer could be used to restore a cleared status in the event the second message isn't received.

Example:
Time 0 - motion payload received -> start timer, send motion detected
Time 1 - 2nd motion payload received -> stop timer,  send motion cleared 
If timer is >= 3s, send motion cleared

Happy to help test if you choose to implement this one way or the other.

PaulWieland, 2023-05-08:
2.1c adds support for motion detection. it toggles the status between `clear` and `detected` with each motion event. Every status update (which usually is issued by the GDO when idle) clears the motion detection in case things get out of sync. I hope it works, but I have no way of testing at the moment since I don't own one of these fancy wall panels with a PIR sensor :)

PaulWieland, 2023-05-08:
I think it’s best if I just broadcast every motion message and let the user do what they want with it. It’s not certain that the second event even means clear

Automayted, 2023-05-08:
> I think it’s best if I just broadcast every motion message and let the user do what they want with it. It’s not certain that the second event even means clear

This works for me! I'm currently only using the detected message anyways.

Reference for those wanting to add to HA manually, here's my MQTT binary sensor setup:

```
binary_sensor:
  - name: "Garage Door Motion Sensor"
    unique_id: "garageMotion"
    state_topic: "ratgdo/garage_opener/status/motion"
    payload_on: "detected"
    device_class: motion
    off_delay: 10    
```