PaulWieland/ratgdo #37: v1.7 - Disconnects from MQTT but device/UI still online

kantlivelong, 2023-04-13:
After some period of time ratgdo will disconnect from MQTT but still is online and the configuration UI is still accessible.

Simply clicking "STORE CONFIG" in the WebUI prompts the device to restart and connect.

kantlivelong, 2023-04-14:
I'm also using mosquito and not using DNS for defining the MQTT host. I'll turn the logging up but it might take some time since I'm not sure how to reproduce it.

iceman73, 2023-04-28:
@PaulWieland do you have webserial enabled?

Actually i just reviewed the code, might be a good add for T/S things like this

robgridley, 2023-05-03:
I installed my v2.0 board yesterday and I'm having the same problem. The device is online, but it's offline in MQTT. I can trigger the problem by rebooting whichever AP the ESP is connected to. The device remains offline until I click "Store Config", forcing it to reboot.

Here are the relevant lines from the Mosquitto debug log. After these messages it goes silent.

```
2023-05-03 08:00:31: New connection from 172.16.0.171:54226 on port 1883.
2023-05-03 08:00:31: Client Garage_Door already connected, closing old connection.
2023-05-03 08:00:31: Sending PUBLISH to 0FZ71SlYLAzMOUIMdf928M (d0, q0, r0, m0, '/ratgdo/Garage_Door/status/availability', ... (7 bytes))
2023-05-03 08:00:31: New client connected from 172.16.0.171:54226 as Garage_Door (p2, c1, k15, u'ratgdo').
2023-05-03 08:00:31: Will message specified (7 bytes) (r0, q1).
2023-05-03 08:00:31: 	/ratgdo/Garage_Door/status/availability
2023-05-03 08:00:31: Sending CONNACK to Garage_Door (0, 0)
2023-05-03 08:00:31: Received SUBSCRIBE from Garage_Door
2023-05-03 08:00:31: 	/ratgdo/Garage_Door/command/# (QoS 0)
2023-05-03 08:00:31: Garage_Door 0 /ratgdo/Garage_Door/command/#
2023-05-03 08:00:31: Sending SUBACK to Garage_Door
2023-05-03 08:00:31: Received SUBSCRIBE from Garage_Door
2023-05-03 08:00:31: 	/ratgdo/Garage_Door/set_code_counter (QoS 0)
2023-05-03 08:00:31: Garage_Door 0 /ratgdo/Garage_Door/set_code_counter
2023-05-03 08:00:31: Sending SUBACK to Garage_Door
```

PaulWieland, 2023-05-03:
Thanks @robgridley I was just now able to reproduce the problem by following your suggestion. It seems that the reconnect routine is not publishing the `online` message upon reconnect. Will issue a fix soon.

robgridley, 2023-05-03:
Awesome. Thanks, @PaulWieland. Here's the ESP side anyway.

```
......
      WIFI CONNECTED
                    IP Address: 172.16.0.171
nb of attempts: 6
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: /ratgdo/Garage_Door/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [/ratgdo/Garage_Door/command/#] 
TOPIC SUBSCRIBED [/ratgdo/Garage_Door/set_code_counter] 
```

kirsch33, 2023-05-03:
it may be unrelated, but after enabling 'birth message retain' and 'will message retain' in the mosquitto settings i havent had a single disconnect since. its been 2 days now, vs before changing these settings the max id get would be ~12 hours before having to issue a reboot:

[image]

kantlivelong, 2023-05-05:
@PaulWieland a little off topic but are you going to publish the source for 2.x?

robgridley, 2023-05-05:
@PaulWieland, I just flashed 2.1a using esptool and the problem persists for me. I rebooted the AP that the ESP was connected to, the ESP immediately connected to another AP, but MQTT is still offline.

PaulWieland, 2023-05-06:
@robgridley  Can you open the ratgdo serial log, disconnect the AP and then paste the entire log?

Mine looks like this (the reconnect message occurs after where your log stops):

```
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: home/garage/GlassDoor/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1
MQTT attempts=
0
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: home/garage/GlassDoor/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1
MQTT attempts=
1
....
WIFI CONNECTED
IP Address: 10.0.1.165
nb of attempts: 4
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: home/garage/GlassDoor/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [home/garage/GlassDoor/command/#] 
TOPIC SUBSCRIBED [home/garage/GlassDoor/set_code_counter] 
QUEUE MSG SENT [homeassistant/cover/GlassDoor/config] 
{
  "~": "home/garage/GlassDoor",
  "name": "GlassDoor",
  "unique_id": "GlassDoor_2C:F4:32:63:98:38",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "GlassDoor",
    "identifiers": "GlassDoor_2C:F4:32:63:98:38",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1a",
    "configuration_url": "http://10.0.1.165/"
  }
}
QUEUE MSG SENT [homeassistant/light/GlassDoor/config] 
{
  "~": "home/garage/GlassDoor",
  "name": "GlassDoor Light",
  "unique_id": "GlassDoor_2C:F4:32:63:98:38_light",
  "availability_topic": "~/status/availability",
  "command_topic": "~/command/light",
  "payload_on": "on",
  "payload_off": "off",
  "state_topic": "~/status/light",
  "device": {
    "name": "GlassDoor",
    "identifiers": "GlassDoor_2C:F4:32:63:98:38",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1a",
    "configuration_url": "http://10.0.1.165/"
  }
}
QUEUE MSG SENT [homeassistant/binary_sensor/GlassDoor/config] 
{
  "~": "home/garage/GlassDoor",
  "name": "GlassDoor Obstruction",
  "unique_id": "GlassDoor_2C:F4:32:63:98:38_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "GlassDoor",
    "identifiers": "GlassDoor_2C:F4:32:63:98:38",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1a",
    "configuration_url": "http://10.0.1.165/"
  }
}
QUEUE MSG SENT [home/garage/GlassDoor/status/availability] 
online
```

Automayted, 2023-05-06:
I was also seeing this on v2.0 and now again on v2.1a. Since install about a month ago, both openers would go offline sporadically, even though their web-UI was available when reported offline. A reboot of the devices or reboot of Home Assistant resolved the issue. Also using Mosquitto.

My current workaround uses Node-RED to subscribe to the availability topic, and if an offline message is received, then immediately publish an online message to the availability topic. Because the openers continue to work after manually updating their availability over MQTT, this confirms the devices are in fact online and reachable, but are publishing erroneous offline availability.

robgridley, 2023-05-08:
> Can you open the ratgdo serial log, disconnect the AP and then paste the entire log?
> 
> Mine looks like this (the reconnect message occurs after where your log stops):

There's nothing in the log after the the two topic subscription lines.

PaulWieland, 2023-05-08:
released 2.1

kirsch33, 2023-05-08:
when do you plan to release a 1.x version?

PaulWieland, 2023-05-08:
> when do you plan to release a 1.x version?

1.8 has been released