PaulWieland/ratgdo #38: importing into HA if not auto-added

dovy6, 2023-04-16:
I got my RATGDO over the weekend, hooked it up today, took just a few minutes, and seems to be working well as far as I can tell. However, I have it running on my IOT network, and Home assistant is on a different LAN, and (perhaps for this or another reason) HA is not auto-importing it. How do I manually add it to HA?
I can see in MQTT Explorer that it seems to be working fine, but I don't see it in home assistant. 
[image]

Thanks!

dovy6, 2023-04-20:
It seems I was looking for it in the wrong place. I was looking to install or configure an Integration, and it just shows up as an Entity. I'm still kinda new to HA and learning the different terminology. Anyone have any suggestions for a good card that can visualize if the garage door is open or closed from the dashboard?
Also, does the ratgdo website have a status page directly? Or is the only webpage exposed the config page?

Thanks so much for creating this
Dovy

PaulWieland, 2023-04-20:
Within Home Assistant you need to install and configure the MQTT integration (you have to have an MQTT broker running somewhere).

After that, each ratgdo shield shows up as a single HA device with 3 entities. The entities are the Door (cover), Obstruction sensor and light switch.

dovy6, 2023-04-20:
I'm asking about a status page for ratgdo because I had this happen to me, where the device was showing up in mqtt explorer and I was even able to open and close the door from sending mqtt messages, but it wasn't available in HA
[image]

PaulWieland, 2023-04-20:
Availability in Home Assistant is controlled by the last will and testament in ratgdo.

When ratgdo boots up, it sends [`prefix/status/availability` = `online`](https://github.com/PaulWieland/ratgdo/blob/9eab2fedf65cb47e5209c6b9b7210e805f9bb818/src/ratgdo.cpp#L103) and a last will and testament of [`prefix/status/availability` = `offline`](https://github.com/PaulWieland/ratgdo/blob/9eab2fedf65cb47e5209c6b9b7210e805f9bb818/src/ratgdo.cpp#L36).

MQTT last will messages are broadcast by the mqtt broker (not the client device) if the broker loses communication with the device. So if ratgdo loses its wifi connection or power the `offline` status message is sent and Home Assistant will show it as Unavailable.

bradmck, 2023-04-21:
On your question about a card, I know a lot of people use the Mushroom Cover card, but that's only going to show the door itself I think (I don't use it myself).  Also, Paul has the RATGDO showing up as a Devicein HA.  So you can go to Settings, Devices and you should see the device.  That device page will show the door, light, obstruction, etc.  Doesn't answer your card question at all, but just pointing out it's not just an entity/entities.

If you want one card to display everything about RATGDO, you're probably going to need one of the "Stack" cards and add multiple cards into it (like the Mushroom Cover, Mushroom Entity (the light isn't actually a light since you can't get status, it's just toggle) , etc.).  Here's a _super quick_ mock up.  And sorry everything is Unavailable, I don't even have a garage door opener.  :)

[image]

dovy6, 2023-04-23:
I'm looking for a card that can clearly show if the door is open or closed. For example this Light card shows that the Kitchen light is on. But it's very hard to tell if the door is open or closed
[image]

Here's another where the light is off (very easy to tell) and the door is open (barely noticeable.)
[image]

Would be cool if the garage icon turned yellow when the door is open the way the lights do when they are on

PaulWieland, 2023-04-23:
[image]

fwiw here’s the HomeKit Home.app view

bradmck, 2023-04-23:
Just looking at this quickly, you could use the custom-button card from HACS to do somthing like this pretty easily:
[image]
[image]
Here's the YAML for the garage door card - these three cards are just in a horizontal stack:
```
type: custom:button-card
entity: cover.ratgdo
icon: mdi:garage
state:
  - value: open
    icon: mdi:garage-open
    color: red
  - value: closed
    icon: mdi:garage
styles:
  card:
    - height: 140px
    - width: 140px
```

dovy6, 2023-04-23:
This is my entities card: `type: entities
entities:
  - entity: light.living_room_dimmer
  - entity: light.nook_dimmer
  - entity: light.dining_room_dimmer
  - entity: light.eat_in_kitchen
  - entity: light.kitchen
  - entity: cover.ratgdo`

With this each of the lights turn yellow when on, and actually the brightness of the yellow correlates to the brightness of the light (dimmers.) But for whatever reason the garage icon doesn't turn yellow when the door is open. It does switch to mdi:garage-open when it's opened, but not yellow or better red.
Adding `icon: mdi:garage
state:
  - value: open
    icon: mdi:garage-open
    color: red
  - value: closed
    icon: mdi:garage` doesn't seem to work, at least for the entities card

dovy6, 2023-05-07:
My RatGDO jumped to unavailable again this past week, apparently for several days before I noticed. Can you explain further why this is happening and or what logging I can do to help determine the cause?
[image]

Thanks

closed 2023-05-19 by PaulWieland