PaulWieland/ratgdo #39: reed switch constantly flipping

kirsch33, 2023-04-17:
So i have a v1 blue board (didnt know 2.0 was an option when ordering somehow) and I believe I have everything configured/wired correctly, including the reed switch on my garage door. however i noticed that in Home Assistant the door always has a status of "unknown", and when I connect the board to my pc to investigate the logs I notice that it is constantly sending MQTT messages of 'closed' following by 'open', repeating, nonstop. this is with or without the magnet engaged. 

I have confirmed with my multimeter that the reed switch is working, so im confused what would cause this problem? bad wiring is all that I can think of, but i followed the schematic to a T.

kirsch33, 2023-04-17:
@PaulWieland the setup docs dont show anything connected to RPM2, so just curious where that recommendation comes from? This is the wiring diagram I used:
[https://user-images.githubusercontent.com/4663918/217019299-372ad9b1-4a2d-43e1-bc8b-3a1607741f33.png](url)

To clarify, I am seeing reed_open and reed_closed both, but its never correct (as in im not doing anything to the magnet or reed switch after powering the board) and it just repeats constantly back and forth between open and closed. Here is a screenshot of my last 10 MQTT messages:
[image]

PaulWieland, 2023-04-17:
It looks like RPM1 is bouncing high/low.

Are you sure the switch is close enough to the magnet?
Try putting a jumper wire from RPM1 to ground (or to 3.3vin) temporarily to see if the bouncing stops.

kirsch33, 2023-04-17:
Ok I will try the jumper this evening and follow up.

I tested the positioning of the switch/magnet with my multimeter and I think its fine? Set to ohms, circuit reads as open loop with magnet removed, and shows 0 with magnet placed in normal position.

kirsch33, 2023-04-18:
So, with the jumper to gnd:

- obstruction sensors stop operating (lights on them go off as well)
- reed switch shows as closed and then wont update past that point even if completely disconnected

with jumper to 3.3v:

- reed switch shows as open and then wont update past that point even if completely disconnected

However with both options, obviously, the rapid bounce is gone. Not sure what to make of any of this.

PaulWieland, 2023-04-18:
Can you post a photo of your wiring?

PaulWieland, 2023-04-18:
You reed switch should not be wired to the black/obst pin. It has to be wired to gnd.

kirsch33, 2023-04-18:
Ok i will try this evening and follow up, but am I crazy or is that not how the v1 solder-free wiring diagram shows it? See below

[image]

PaulWieland, 2023-04-18:
> Ok i will try this evening and follow up, but am I crazy or is that not how the v1 solder-free wiring diagram shows it? See below
> 

you are not crazy, there’s a mistake in the diagram that went undetected until you just pointed it out! 😅

PaulWieland, 2023-04-19:
I've updated the docs. Sorry about the mistake in the diagram!