PaulWieland/ratgdo #41: Confused on wiring for v2

daredoes, 2023-04-26:
It would be great to have some photos of existing/successful setups. I'm a little lost on if I need to wire up the triggers. I think I've wired in the white, black, and red on the other side, but the controller seems to lack control.

shbatm, 2023-04-29:
A bit confused on this as well between the diagrams on the README of the repo and the diagrams on the Github pages site... so there is no connection between the top left (formerly RPM1 and RPM2) pins and the D1 mini anymore? Doesn't appear that these pins are connected at all internal on the board, so there's no option to still use the solder method to read the rotary encoder?

Would it be possible to post the schematics / Eagle files for the board? I'm interested in adding this back on mine if I can.

daredoes, 2023-04-30:
@PaulWieland thanks so much for the quick response! I've gotten it to work. My biggest issue was that the connection between the ratgdo board and the d1 mini clone I got wasn't secure enough using just some pins, so I soldered them in. The board now opens and closes the garage... sometimes when it turns on! 

Next issue though is that my garage doesn't seem to respond to my commands. I can confirm the commands are going through when I use the MQTT integration via home assistant. I tried enabling and disabling the static rolling codes feature in the config as well, but no change.

Is there some secret to figuring out the rolling codes? Do I need to press the sync button on my garage as well?
I'd love any advice. Here's a log dump

[log.zip](https://github.com/PaulWieland/ratgdo/files/11362197/log.zip)

dovy6, 2023-05-07:
That log dump appears to be a list of media tracks

daredoes, 2023-05-09:
@dovy6 that's what I get for having too many log.zip files :(

closed 2023-06-23 by PaulWieland