PaulWieland/ratgdo #43: Sharing Board Schematics

shbatm, 2023-04-30:
Thanks for the project. I just received my V2.0 board, but I was looking at possibly including this in an update to my existing multisensor board I have in my garage, would it be possible for you to share the board schematics / BOM / EAGLE files?

I recognize you are selling these, but since the code is open source, I thought I'd ask.

Marus, 2023-05-11:
Would also be interested in this as I wouldn’t mind doing a custom board with some additional parts.

dorkage, 2023-05-16:
I also would be very interesting in schematics.

tylermeredyk, 2023-06-08:
just received mine works great. board is a bit big. id like to make a pcb with jst connectors. +1 for schematics

closed 2023-10-03 by PaulWieland

fgonza2, 2023-10-12:
did they shared the schematics somewhere?

tylermeredyk, 2023-10-12:
Doesn’t seem like he wants to give them out… As far as I got. 
[image]

brad07x, 2023-10-12:
Was just going to post to see if there was any info out there on how to hook up the obst/ctrl lines as I'd like to hack something together until my backordered unit arrives; my setup broke again this week with the breaking API changes and related issues with pymyq.

ripnet, 2023-10-12:
Hello. Were the schematics ever released? Everything is on backorder or else I'd order a few. I am also interested in creating a board with an onboard 3.3v regulator.