PaulWieland/ratgdo #48: How to Perform OTA Update?

abishur, 2023-05-17:
In the web gui I saw that I needed to set a password for the gui/OTA feature, and I see the bin firmware files here on github.  Once I have the .bin I want to flash how do I perform an OTA update?

Automayted, 2023-05-22:
I too would love to be able to flash the recent ratgdo v2.x updates OTA.

cpfair, 2023-05-24:
You can use the [espota.py tool](https://github.com/esp8266/Arduino/blob/master/tools/espota.py):

```bash
python espota.py -i <DEVICE ADDRESS> -p 8199 -a <OTA PASSWORD> -f <PATH TO FIRMWARE BIN>
```

Automayted, 2023-05-24:
Good to know! While not difficult, this method may be a little over many users' heads. Also, the latest bins are not available online, only via the web flasher.

PaulWieland, 2023-05-24:
All of the bin files are in the repo and are the same ones referenced by the web flasher. And yeah, ota requires you to know how to use the cli or compile by source. There is no ota possible from esp web tools.

Automayted, 2023-05-24:
I'm only seeing the v1.7 source files [on your release page](https://github.com/PaulWieland/ratgdo/releases), unlike say, Tasmota, where anyone [can easily grab the latest bins](https://github.com/arendst/Tasmota/releases) for their hardware. Pardon the silly question, but where do I find the v2.1 bin?

PaulWieland, 2023-10-06:
Version 2.23a adds OTA from the Web UI and will be incorporated into version 2.2 as soon as it's ready.

PaulWieland, 2023-11-01:
v2.2 & v2.5 now have an option to upload a firmware.bin file through the web ui.

closed 2023-11-01 by PaulWieland