PaulWieland/ratgdo #49: Feature Request: Ability to use a temp/ humidity sensor with the V2

shosta3, 2023-05-17:
Since the v2 no longer uses the rotary encoder as the the v1 did, can we get a bin to provide support for a temperature/ humidity sensor. Adding 2 bins one for the bme280 and one for the dht22 as these are what seems to be the most common sensors should cover most people. I would think we would want ratgdo provide the data via mqtt and leave the sensor up to the end user.

Automayted, 2023-05-19:
Agreed this would be a great addition to the awesomeness that is ratgdo!

torqu3e, 2023-09-07:
@PaulWieland would this request be too much to accommodate? Expose a configurable one wire interface to add a sensor of choice that can report data back over MQTT. Would help folks run heaters/fans to maintain temperature based on what the sensor is reading. Thanks!