PaulWieland/ratgdo #50: Ratgdo worked for several weeks, now appears stuck in boot loop. Replacement unit online and reports commands, but cannot control the opener

Automayted, 2023-05-18:
One of two of my ratgdos suddenly stopped working. When directly connected, there appears to be a loop of MQTT connected/disconnected. Power-cycling the device or reseting from the web flasher results the same pattern returning. Please see attached "Original device logs. Note the logs were taken while the device was NOT connected to the opener. When connected, it is unreachable over WiFi and therefore cannot communicate with my MQTT broker. This unit is flashed with v2.1c

I flashed a brand new ESP8266 (D1 clone) with v2.1 and it is correctly reporting events, but is unable to control the light or door. Please see the attached "Replacement Device Logs. 

I've power cycled the opener itself as well as the ratgdo several times with no change.

Please let me know what other info I can provide.

Thanks!

[Original Device Logs.txt](https://github.com/PaulWieland/ratgdo/files/11512085/Original.Device.Logs.txt)
[Replacement Device Logs.txt](https://github.com/PaulWieland/ratgdo/files/11512086/Replacement.Device.Logs.txt)

PaulWieland, 2023-05-18:
> I flashed a brand new ESP8266 (D1 clone) with v2.1 and it is correctly reporting events, but is unable to control the light or door. Please see the attached "Replacement Device Logs.

It sounds like you are trying to re-use counters and the GDO is ignoring them. Did you erase the flash when you installed the firmware? Or was this flashing to a new D1 Mini and being connected to an opener that already had another ratgdo connected to it?

Either way, you need to advance the counter with `set_code_counter` and then `sync`. It's best if you know what the last used counter was from the previous device, otherwise you have to guess. [See here.](https://paulwieland.github.io/ratgdo/01_features.html)

As far as it connecting and then disconnecting, that sounds like a network issue but its hard to pinpoint the exact cause. If it does it again try powering the ESP on nearer your wifi access point.

Automayted, 2023-05-19:
> It sounds like you are trying to re-use counters and the GDO is ignoring them. Did you erase the flash when you installed the firmware? Or was this flashing to a new D1 Mini and being connected to [...]

I'm flashing a brand new ESP and still have the old unit, however it will not stay connected to WiFi, even though it's less than 2m from the AP in my garage and less than 3m from an AP outside my office. All other devices on both APs have no connectivity issues, so I assumed it was a hardware issue. The new ESP is in the exact same spot and has no connectivity issues.

> Either way, you need to advance the counter with `set_code_counter` and then `sync`. It's best if you know what the last used counter was from the previous device, otherwise you have to guess. [See here.](https://paulwieland.github.io/ratgdo/01_features.html)

Because I have the original ESP, I'm able to read the logs before it goes into it's MQTT connect/disconnect loop:

```
version 2.1c
Syncing rolling code counter after reboot...
rolling code for 577|reboot1 : 5501000A3DF79F6DB6DF7D5529B0832A3BC30A
rolling code for 578|reboot2 : 550100650DB69A69A4DA68043CBF4D7A5FBF7F
rolling code for 579|reboot3 : 550100643F7FBFFBEDFEFFA527B7ED6A551B28
rolling code for 580|reboot4 : 550100420B6492C92C9258941DEE9AFC1FAEFF
rolling code for 581|reboot5 : 55010041124126524124908539F4B76A569BCF
rolling code for 582|reboot6 : 5501009804804B44120124243CFECBB93F5FAF
Write successful
IMPROVhttp://10.10.86.232�
### MQTT DISCONNECTED ###
```

With only the new ESP online and connected to the ratgdo board, I've tried publishing `ratgdo/garage_west_door_opener/command/set_code_counter` with a payload of `583` followed by `ratgdo/garage_west_door_opener/command/sync` with no payload, but the unit still does not respond to any incoming commands. Is there something I'm not understanding about this process?

> As far as it connecting and then disconnecting, that sounds like a network issue but its hard to pinpoint the exact cause. If it does it again try powering the ESP on nearer your wifi access point.

See above, certainly not the issue in my case.

Thanks!

PaulWieland, 2023-05-19:
Verify that your mqtt credentials are correct

Automayted, 2023-05-19:
> Verify that your mqtt credentials are correct

They are indeed correct. The new unit is accurately reporting events (motion, obstruction, door open/closed, and light on/off.

Automayted, 2023-05-19:
I grabbed the logs over serial, including me publishing the messages noted above. I'm not seeing the rolling code counter change to the value sent as the `set_code_counter` payload:

```
{$ l��| �l�|�l�c|����{�c�b��og�lgn���cp��dsdrd8�o��d��cg�|���c��gn� lćd`�gn�lorǛ�'{;�`x�o�{������co�<�c��gn� �l`�g'�lo;���gsl s��'{l`���l{d`��o�l
Mounting LittleFS...
[setup.json]
JSON parsed
Storage OK, restoring WiFi and MQTT config.
DPsoftware domotics
Connecting to: 
SPARSE...
Using static IP address
Set WiFi output power to: 20.50
.
WIFI CONNECTED
IP Address: 10.10.86.40
nb of attempts: 1
IMPROVhttp://10.10.86.40�

WiFi connected
Local IP: 10.10.86.40
SoftAP IP: (IP unset)
Server started
Launching webserver for improv
Starting ArduinoOTA service
doorCommandTopic: ratgdo/garage_west_door_opener/command/door
lightCommandTopic: ratgdo/garage_west_door_opener/command/light
lockCommandTopic: ratgdo/garage_west_door_opener/command/lock
Setup Complete
 _____ _____ _____ _____ ____  _____ 
| __  |  _  |_   _|   __|    \|     |
|    -|     | | | |  |  |  |  |  |  |
|__|__|__|__| |_| |_____|____/|_____|
version 2.1
Syncing rolling code counter after reboot...
rolling code for 219|reboot1 : 550100943FFDE7DF7FBEFF520C26DA4EA98A67
rolling code for 220|reboot2 : 550100620D26924D36926841144A6897E13490
rolling code for 221|reboot3 : 5501006132CB24965B2496220A21344EE75245
rolling code for 222|reboot4 : 55010008349249B4924D20922BA7BE4AC4C269
rolling code for 223|reboot5 : 550100061B2DBF5B6DB64B843BF5B7EA1FBF5F
rolling code for 224|reboot6 : 55010081024122D2092490861F67DE5E8DF66B
Write successful
IMPROVhttp://10.10.86.40�
### MQTT DISCONNECTED ###
Connecting to
MQTT Broker...
MQTT Last Will Params: 
willTopic: ratgdo/garage_west_door_opener/status/availability
willPayload: offline
qos: 1
retain: 0
clean session: 1

MQTT CONNECTED

Reading data from
the network...
TOPIC SUBSCRIBED [ratgdo/garage_west_door_opener/command/#] 
TOPIC SUBSCRIBED [ratgdo/garage_west_door_opener/set_code_counter] 
QUEUE MSG SENT [homeassistant/cover/garage_west_door_opener/config] 
{
  "~": "ratgdo/garage_west_door_opener",
  "name": "garage_west_door_opener",
  "unique_id": "garage_west_door_opener_48:3F:DA:6C:7D:6A",
  "availability_topic": "~/status/availability",
  "device_class": "garage",
  "command_topic": "~/command/door",
  "payload_open": "open",
  "payload_close": "close",
  "payload_stop": "stop",
  "state_topic": "~/status/door",
  "device": {
    "name": "garage_west_door_opener",
    "identifiers": "garage_west_door_opener_48:3F:DA:6C:7D:6A",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1",
    "configuration_url": "http://10.10.86.40/"
  }
}
QUEUE MSG SENT [homeassistant/light/garage_west_door_opener/config] 
{
  "~": "ratgdo/garage_west_door_opener",
  "name": "garage_west_door_opener Light",
  "unique_id": "garage_west_door_opener_48:3F:DA:6C:7D:6A_light",
  "availability_topic": "~/status/availability",
  "command_topic": "~/command/light",
  "payload_on": "on",
  "payload_off": "off",
  "state_topic": "~/status/light",
  "device": {
    "name": "garage_west_door_opener",
    "identifiers": "garage_west_door_opener_48:3F:DA:6C:7D:6A",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1",
    "configuration_url": "http://10.10.86.40/"
  }
}
QUEUE MSG SENT [homeassistant/binary_sensor/garage_west_door_opener/config] 
{
  "~": "ratgdo/garage_west_door_opener",
  "name": "garage_west_door_opener Obstruction",
  "unique_id": "garage_west_door_opener_48:3F:DA:6C:7D:6A_obs",
  "availability_topic": "~/status/availability",
  "device_class": "motion",
  "state_topic": "~/status/obstruction",
  "payload_on": "obstructed",
  "payload_off": "clear",
  "device": {
    "name": "garage_west_door_opener",
    "identifiers": "garage_west_door_opener_48:3F:DA:6C:7D:6A",
    "manufacturer": "Paul Wieland",
    "model": "ratgdo",
    "sw_version": "2.1",
    "configuration_url": "http://10.10.86.40/"
  }
}
QUEUE MSG SENT [ratgdo/garage_west_door_opener/status/availability] 
online
Obstruction status obstructed
QUEUE MSG SENT [ratgdo/garage_west_door_opener/status/obstruction] 
obstructed
QUEUE MSG ARRIVED [ratgdo/garage_west_door_opener/command/set_code_counter] 
583
QUEUE MSG ARRIVED [ratgdo/garage_west_door_opener/command/sync] 
```

PaulWieland, 2023-05-19:
And the end of the log It looks like you set it to 538. Sync should have sent a series of codes. Either the log is cut off or it didn’t sync for some reason. You can reboot it and it will sync on startup. You should see it continue from 538.

Automayted, 2023-05-19:
> And the end of the log It looks like you set it to 538. Sync should have sent a series of codes. Either the log is cut off or it didn’t sync for some reason. You can reboot it and it will sync on startup. You should see it continue from 538.

I sent 583, which is 1 greater than the last code shown on the old device.

You're looking at the whole log. There was no response, and you can see the device is online and subscribing/publishing.

Using the web flasher's logs -> reset simply increments the code number shown at startup by 6.

Automayted, 2023-05-19:
UPDATE: After noticing the codes increment by 6 on each reboot, I manually rebooted the new ESP about 60 times to forcibly update the code to >583. After that, the new ESP is controlling the opener again. 

Something is not working as expected with `ratgdo/garage_west_door_opener/command/set_code_counter` and/or `ratgdo/garage_west_door_opener/command/sync`

Attached the full log for reference. 
[esp-web-tools-logs (2).txt](https://github.com/PaulWieland/ratgdo/files/11512607/esp-web-tools-logs.2.txt)

PaulWieland, 2023-05-19:
I just tested `set_code_counter`, `sync` and `query` and all three work on 2.1.

How are you formatting your mqtt commands?

[image]

[image]

[image]

Automayted, 2023-05-19:
Thanks, Paul!

I'm using MQTT Explorer to publish the messages thus far.

There is a slight difference in the documentation vs what you provided in screenshots: Docs say to use the ~/command/sync topic with no mention of payload, where you show ~/command with sync as a payload. Also, not seeing any reference to the ~/command topic and query payload. I'll retest next week when back home.

PaulWieland, 2023-05-19:
You're right. I'll update the docs.

Automayted, 2023-05-22:
Paul, when publishing a payload of `sync` and `query` to the `~/command` topic in the format you show above, I'm not seeing any counter data in the responses. I'm subscribed to `~/#` and only get the general status messages returned (light, door, etc.). Do I need to be connected over serial to see the current counter, or should that be returned over MQTT as well?

Inject node configs:
[image]

[image]

MQTT In node config:
[image]

Debug output:
[image]

PaulWieland, 2023-05-22:
It doesn't return the current counter (maybe it should during a "query").

PaulWieland, 2023-05-22:
corrected the docs

Automayted, 2023-05-22:
Got it! So in the event of a device replacement or full clearing of the ESP, users need to connect over serial to obtain the current counter.

I'll update this Issue to a feature request to include the current counter during a query. I believe this is beneficial in the event of a complete ESP hardware failure where a user may not be able to get the current counter. If the counter was included with the query response(s) a user could then create a Node-RED flow to log the last counter value for reference.

PS: Just saw you updated the docs - thanks!