PaulWieland/ratgdo #54: just installed new device, controls do not work but obstruction does - v2.1 liftmaster 8500

apaperclip, 2023-05-24:
Hello,

I just received my ratgdo shield, I flashed an esp8266 d1mini, configured and attached it.  I just wired things up and something isn't quite working right. I have a liftmaster 8500 and am using HA. I have other mqtt devices that work correctly.

Here's what I've done and noticed (its a lot, but I wanted to be thorough)

- with the opener power plug pulled, I wired the ratgdo
- I plugged in the opener and waited for the wall control to stop flashing, and beep saying it was ready.
- I applied power to the esp8266 on the ratgdo and the garage door opened.
- no other commands (light on/off, door open/stop/close) or status reporting of those items work from HA. 
- The obstruction sensor did correctly update when I blocked the beam.
- When connected via serial, clicking reset toggles the garage door opened/closed
- I tried to set code value of 500 and sync but that did not work. I tried again with 600 and 700.
- The logs mention 2 doors, I only have 1.
- I see msg's being sent for light status but the value is literally "ratgdo/garage/status/light" instead of on/off (viewed in mqtt explorer and logs below), these are seen by mqtt explorer.
- I do not have an ha sensor for the control panel motion sensor nor see any message sent to a related topic 
- The docs say the set_code_counter is ratgdo/garage/command/set_code_counter, but the device only acknowledges if its ratgdo/garage/set_code_counter.

For a new device, how is the correct code value determined?  I see posts about recording the last code when replacing or erasing, but for a brand new device, does it start at 1? Is the  next valid code unique to the controller (button, car, ratgdo, etc)

Here is a snippet of logs, let me know if you'd like to see something specific or have any suggestions for what I should try next. 

```
UEUE MSG ARRIVED [ratgdo/garage/command/light] 
off
MQTT: turn the light off
rolling code for 646|light : 55010052012014412082415033D927B01F5DA2
Write successful
55010052012014412082415033D927B01F5DA2 | LIGHT:3
Light state ratgdo/garage/status/light
QUEUE MSG SENT [ratgdo/garage/status/light] 
ratgdo/garage/status/light
QUEUE MSG ARRIVED [ratgdo/garage/command/light] 
on
MQTT: turn the light on
rolling code for 647|light : 550100511349A29A4D34934114434C95E1349C
Write successful
550100511349A29A4D34934114434C95E1349C | LIGHT:2
Light state unknown
QUEUE MSG SENT [ratgdo/garage/status/light] 
unknown
QUEUE MSG ARRIVED [ratgdo/garage/command/light] 
off
MQTT: turn the light off
rolling code for 648|light : 55010098249243241201249A2BB5FE6E568347
Write successful
55010098249243241201249A2BB5FE6E568347 | LIGHT:3
Light state ratgdo/garage/status/light
QUEUE MSG SENT [ratgdo/garage/status/light] 
ratgdo/garage/status/light
QUEUE MSG ARRIVED [ratgdo/garage/command/door] 
stop
MQTT: stop the door
The door is not moving.QUEUE MSG ARRIVED [ratgdo/garage/command/door] 
open
MQTT: open the door
rolling code for 649|door1 : 550100961FEDBCDBEDF7F89030C848B13B2DBE
rolling code for 649|door2 : 550100961FEDBCDBEDF7FA9030C848B13B3DBE
Write successful
550100961FEDBCDBED9030C8C9FFFEFEFE80FE
550100961FEDBCDBEDF7F29030C848B13B3DBE
QUEUE MSG ARRIVED [ratgdo/garage/command/door] 
open
MQTT: open the door
rolling code for 650|door1 : 55010041120826104124954022982533736DBE
rolling code for 650|door2 : 55010041120826104124914022982533737DBE
Write successful
5501004112082610954022882433736DFEFE00
QUEUE MSG ARRIVED [ratgdo/garage/command/door] 
stop
MQTT: stop the door
The door is not moving.QUEUE MSG ARRIVED [ratgdo/garage/command/door] 
close
MQTT: close the door
rolling code for 651|door1 : 5501004026D26F26D36DB321104880D589A4B3
rolling code for 651|door2 : 5501004026D26F26D36DB721104880D589E4B3
Write successful
```

and thank you!

PaulWieland, 2023-05-24:
ratgdo rolling code counter starts at 0 for new installs, theres no reason to manually set it unless for some reason you erase the flash chip on the ESP8266 after having used it with your door.

When you open the door with the wall mounted button does the serial console output any data?

Can you please post a picture of your wiring.

wardwolfram, 2023-05-24:
I have a Chamberlain RJ070 which has the same DNA as the Chamberlain 8500. I do not know how to get the MQTT logs such as apaperclip posted.  

The results I am getting are:
1. Obstruction is always detected (does not report not-detected).
2. No actions with garage door 
3. No actions with the light

[image]
[image]
[image]

I'll keep an eye out on this thread. Do you have anything that I can provide to assist?

wardwolfram, 2023-05-24:
> @wardwolfram Your D1 Mini is installed into the shield upside down and it looks like the wires are not connected to the right terminals. Please look at the silk screen labels on the terminals and pins for the D1 Mini and connect accordingly.

Ah, yes,... thank you.... I'll flip it around...

wardwolfram, 2023-05-24:
After swapping the leads, the behavior is the same for me except for one thing, the functionality of the wired Chamberlain garage door control unit (button to open/close garage door, switch with timer settings, button and sensor to turn on/off it's garage light) does not function. When I remove the Ratgdo wires, the button begins to function.

I re-installed the FW used different IP, etc. along with swapping the wires to the correct locations.

Do I need to manually 'add' the MQTT Topic Prefix into HA MQTT settings or any other setting besides the MQTT Topic Prefix setting inside the Ratgdo?  
[image]
[image]

MQTT

apaperclip, 2023-05-24:
> ratgdo rolling code counter starts at 0 for new installs, theres no reason to manually set it unless for some reason you erase the flash chip on the ESP8266 after having used it with your door. [...]

Thank you for taking the time to review and reply
[image]
[image]

I see no logs in the console when triggering the door or light via the wall panel.

wardwolfram, 2023-05-24:
> > After swapping the leads, the behavior is the same for me except for one thing, the functionality of the wired Chamberlain garage door control unit (button to open/close garage door, switch with timer settings, button and sensor to turn on/off it's garage light) does not function. When I remove the Ratgdo wires, the button begins to function.
> > I re-installed the FW used different IP, etc. along with swapping the wires to the correct locations.
> 
> I think your esp is still upside down. The usb plug should face the direction of the two holes in the ratgdo. https://user-images.githubusercontent.com/4663918/235453980-04a642fa-a181-4297-b4f3-06e1315e02fa.png

Here is a picture with the shield facing forward.  I mapped according to this. Or, am I still missing something?  
[image]

wardwolfram, 2023-05-24:
> I think your esp is still upside down. The usb plug should face the direction of the two holes in the ratgdo. [...]

Ah... I see.  Gosh, not sure how the USB was rotated a 180?... I must have not been paying attention. I'll give this a try.. Thanks for the heads up!

wardwolfram, 2023-05-24:
> I think your esp is still upside down. The usb plug should face the direction of the two holes in the Ratgdo. [...]

I think that did the trick. Thanks for pointing that out. The garage door sensors looks to be working. The Ratdgo did show unavailable a couple times and will look into this over the next day.... but looks stable at the moment. Thanks!

PaulWieland, 2023-05-25:
> I see no logs in the console when triggering the door or light via the wall panel.

So the door isn't responding to commands and ratgdo isn't picking up traffic. The obstruction line is being read which means the ground wire connection is okay.

The connection of the red signal wire is suspect. If you have a multi meter, take a reading of the voltage on the screw terminal of the ratgdo shield and one of the ground terminals on the shield. It should be around 11-12v.

apaperclip, 2023-05-26:
I'm seeing 5.5v both on the ratgdo and at the terminals on the opener.  I think this may may be why:  Apparently I have a first gen 8500 and it may not do secure+ 2.0 out of the box.    Check out this [link  ](https://advancedsecurityllc.com/wp-content/uploads/2021/02/Garage-Door-Compatibility-List.pdf )and note the l (828LM + 888LM) combo package. This is what my installer installed ~9 years ago.   Also this [link ](https://www.windsordoor.com/assets/uploads/modules/accessory-compatibility-chart-91817-82926.pdf)seems to have a caveat for 8500 support.  Finally this [product ](https://www.garadget.com/product/security-2-0-dry-contact-adapter/)wont work with 8500, maybe this is all why.  Maybe the 888LM wall control has the extra logic for secure+ 2.0?

I took the reading while the esp was not powered but happy to power it and take another reading if you think it would matter.

PaulWieland, 2023-05-26:
Try shorting the red and white terminals together and see if opens the door.

PaulWieland, 2023-05-26:
Ok its definitely not a security+ 2.0 opener. I want to support these older openers with a future release, but I don't currently know if the door status is reported on the signal line. Do you know what those green terminals are for? Some Genie door openers have terminals exposed which give the door state - might be the same here. If not, you have to resort to using an external reed switch to detect if the door is open or not and you can't easily get discrete opening/closing statuses.

If that signal line is 5v instead of 12v the current hardware will not be able to read or detect a signal, so I need someone with one of these openers who also has (or wants to buy) a cheap $15 signal analyzer to do some investigation. I'm also on the lookout for a used unit that I can test with.

apaperclip, 2023-05-26:
Yeah it appears to be 2.0 _ish_ with the 888LM control.  Via the wifi gateway I do get opening/closing in app so it somehow knows the transit states from the wireless association to the wall control.   I think the green terminals run to the cable tension monitor page 10 [here](https://www.devancocanada.com/files/manuals/8500C:Installation-manual.pdf)

the huge appeal for this was to provide local replacement control of the wifi bridge with native integration and full support of lights/obstruction/motion/transit states.  If that may still be in the cards share a link to the signal analyzer, I think i'd be up for the time and effort to help.

apaperclip, 2023-05-26:
Ordered!  This will be something new for me so it should be fun.  I'll be using it on either an android phone or chromebook, based on the docs that should be possible.  Do you want to keep our back and forth conversation while we work through this here or a new issue, or somewhere else?

PaulWieland, 2023-05-26:
I think a chat over [Signal](https://signal.org/download/) would be more efficient.

apaperclip, 2023-05-26:
ok, before i do that any chance we are on mutual discord servers (esphome or home assistant) if that would work.

I'm not seeing a clearly labeled order# on the receipt for the purchase (done through gpay)

PaulWieland, 2023-05-26:
Sorry I don't use discord. check your spam for an email from square / ratgdo. the order number is in the top right along with the order date.

apaperclip, 2023-05-26:
ok,  no problem. I'm on signal now.    also, rookie mistake on my part for the order #, found it on the shipping notification.  - 1414831689 .

agoode, 2023-06-16:
I have an old opener with the 888LM and would be interested in working this out. I just ordered a signal analyzer and will take a look.

PaulWieland, 2023-06-16:
@agoode ill let you know when I have an alpha test firmware ready. I think I have it cracked just need to implement it. 

I also am trying to add an upload firmware option to the web ui at the same time for wireless updating.

PaulWieland, 2023-06-23:
I implemented decoding for Secplus 1, but I'm worried that these openers use a different voltage range. If anyone has a scope or a volt meter capable of capturing min/max voltage please let me know.

agoode, 2023-06-23:
I can measure the voltage, what do you want to measure across?

PaulWieland, 2023-06-23:
@agoode need to capture the minimum voltage when the garage door is broadcasting its status (which I think it does continuously). Measure between the GDO red and white terminals. I'm hoping that the high is around 12v and the low is around 1.5v.

agoode, 2023-06-23:
I measured at the wall unit, the voltage goes between 4.4v and 5.4v.

PaulWieland, 2023-06-23:
does your meter log mininum voltage (some fancy fluke meters do)?

We're trying to measure the low and high signal voltage of a serial data line. The low voltages only last for tiny fractions of a second.

agoode, 2023-06-23:
Yes, I did use max/min on a fluke 115. Let me dust off my scope.

PaulWieland, 2023-06-23:
Okay, I would expect there to be a much bigger voltage differential between min/max.

agoode, 2023-06-23:
If I am reading my oscilloscope correctly, it goes between 0 and 5.5V.

[image]

PaulWieland, 2023-06-23:
Ok, that voltage should work with the ratgdo shield. Now to determine the correct baud rate... I thought it was 1200 baud, 8 bit, even parity based on the data capture I was sent. 

Can you measure the width of the smallest pulse? Is it around 833µs?

agoode, 2023-06-23:
Looks like 1200 8-E-1 to me. It claims 840µs.

[image]

PaulWieland, 2023-06-25:
I just ordered a logic board with a purple learn button off of ebay to test with. It should have the same control logic as the 8500, assuming [this chart](https://www.windsordoor.com/assets/uploads/modules/accessory-compatibility-chart-91817-82926.pdf) is correct.

cerickard2, 2023-08-16:
Hi Paul!  How goes your testing with the half-breed logic board?

PaulWieland, 2023-10-06:
I just pushed version 2.23a.

**What should work:**

- Controlling both the door and the light
- Reading the light state (on, off)
- Reading the door state (open, closing, closed, opening)
- Obstruction state
- Uploading firmware OTA using the Web UI

**What isn't fixed yet:**

- the door will be triggered if you flash the firmware or reboot the ESP with it plugged into the shield
- purple button opener door status interpretation. If anyone has one of these openers, 2.23a will output debug information into the serial console which should tell us which values correspond to which door states

bravehand, 2023-10-06:
Nice work!  I can confirm that "Controlling both the door and the light" & "reading the door state (open, closing, closed, opening)" are working for me.  I don't have a light or a purple button, but this is great thank you!

apaperclip, 2023-10-07:
This is great! Thank you so much for working through this and providing it!

I confirm the control items in your list above worked!  I flashed fresh via serial (did not erase) so I did not test OTA.
I did some basic failure testing (power to ratgdo, power to gdo) and things responded like we know and expect.

One thing I did notice is that when I pressed stop while opening, I could not resume opening.  I could only go back down.

It's cool to see the esphome port repo moved under your project.  Thinking ahead - once you have this version where you want it do you think it can have an esphome port as well?  I see some neat features over there like setting the % open, # of opens, motor state, etc. I don't know if our secplus 1.0 supports all that though.

PaulWieland, 2023-10-07:
> I could not resume opening. I could only go back down.

Thats how all their openers work afaik. If you interrupt the door while it is opening it stops. The next door command causes it to reverse and close.

llamafilm, 2023-10-08:
Thank you @PaulWieland and everyone else who worked on this.  I just installed my new board with a Liftmaster 8500, thinking it was Security 2.0 and then I found this thread.  With version 2.23a it mostly works as described, but sometimes I have to press the open/close button twice.  Here's an example log where I tried to close the door and nothing happened.  After waiting 10 seconds I pressed _close_ again and then it worked.

```
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C QUEUE MSG ARRIVED [garage/ratgdo_48876/command/door] 
close
MQTT: close the door
393F 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C 3852  | door state: 52 1010010
3A5C 3900 3A5C QUEUE MSG ARRIVED [garage/ratgdo_48876/command/door] 
close
MQTT: close the door
3852  | door state: 52 1010010
303A 3900 3A5C 3844  | door state: 44 1000100
Door state unknown
QUEUE MSG SENT [garage/ratgdo_48876/status/door] 
unknown
3A5C 3900 3A5C 3844  | door state: 44 1000100
3A5C 3900 3A5C 3844  | door state: 44 1000100
3A5C 3900 3A5C 3844  | door state: 44 1000100
```

Opening after a power loss/restoration is a major security risk.  Is it possible to fix that?

If the stop button serves no purpose and open & close do the same thing, does anyone know if there's a way to change the UI to make it just one button instead of 3?

apaperclip, 2023-10-09:
@llamafilm  
Do you mean the open/close button in HA or the physical button with your GDO?
Paul is working on the open/close action after power recovery
I'm pretty sure the stop button in HA works for me, but I will retest

PaulWieland, 2023-10-09:
@llamafilm The stop button stops the door if it's moving. This only works if the correct status is being read from the GDO and you can see in your log that state 44 is unknown.

I think these states are depending on the type of wall control panel being used, but I don't have it 100% sorted yet. If you don't mind, do a full cycle of your door (closed -> wait 20s -> open -> wait 20s -> close)  and post the serial log.

Also update to 2.24a

llamafilm, 2023-10-10:
Ok yes the stop button does work.  I was just confusing myself, thinking about the physical wall button which has only one button for up/down/stop.  I think the currently implementation with 3 discrete buttons is actually better.

With 2.24a the behavior is better, I no longer need to press twice, and I haven't seen any unknown states yet.

I also noticed that downloading the serial log causes the device to reboot, which then causes the door to either open or close.  I attached a log showing the full cycle as you said.

[esp-web-tools-logs (5).txt](https://github.com/PaulWieland/ratgdo/files/12852776/esp-web-tools-logs.5.txt)

llamafilm, 2023-10-10:
By the way, where is the actual source code for this?  I noticed the CPP files under `src/` haven't changed recently while you are adding new releases.  I'm not a C programmer so probably wouldn't be much help here, but I was curious to have a look at the code to see if I can understand what causes the open on power restore.

apaperclip, 2023-10-10:
> With 2.24a the behavior is better, I no longer need to press twice, and I haven't seen any unknown states yet.

I was seeing unknown states as well, it seemed to be related to when an open or close was executed via the ratgdo. Since upgrading to 2.24a I have not had any unknowns but only limited testing so far.

 Paul, can you elaborate what you mean by wall panel emulation in the latest commit? Are you doing anything different or just giving a name to what you've been working on.   Just genuinely curious.

northrup, 2023-10-11:
> [...] where is the actual source code for this?

@llamafilm it looks like @PaulWieland is doing this work in the `2.0-secplus1.0` branch. You can see it here: https://github.com/PaulWieland/ratgdo/tree/2.0-secplus1.0

apaperclip, 2023-10-11:
Quick update on my unavailable state log entries.  On 2.24a I'm still getting them and they are not associated with a garage door up/down action.  I've had 2 since flashing 2.24a ~15 hours ago.  As a first step I will pull the esp off and plug it into a pc and let it run while gathering logs.

i've already tried a different usb power supply and different usb cable to rule those out.

apaperclip, 2023-10-12:
New observation:
When I trigger open/close from HA I do not see the opening or closing states in HA, it just does the final state of closed or opened.  However, when I use the wall panel, HA does show opening/closing.

I will grab some serial console logs soon.

apaperclip, 2023-10-15:
> New observation: When I trigger open/close from HA I do not see the opening or closing states in HA, it just does the final state of closed or opened. However, when I use the wall panel, HA does show opening/closing.
> 
> I will grab some serial console logs soon.

This may be an oversight on my part. I think the "ings" are part of the trigger message like `garage is closing triggered by service Cover: Close`

PaulWieland, 2023-10-24:
This issue is being addressed by v2.5 hardware and firmware. If you received a v2.0 board and you own a wall mounted jackshaft opener, please email ratgdo@gmail.com

closed 2023-10-24 by PaulWieland