PaulWieland/ratgdo #55: Feature Request: Add OTA Firmware Update Functionality to the Web-UI

Automayted, 2023-05-25:
As discussed [here](https://github.com/PaulWieland/ratgdo/issues/48), currently the only way to update device firmware OTA is via [espota.py](https://github.com/esp8266/Arduino/blob/master/tools/espota.py). It would be beneficial to all users if ratgdo could incorporate an OTA update feature into the web-UI, similar to how Tasmota, WLED, and many other ESP-based projects have done.

For reference:
[Tasmota](https://github.com/arendst/Tasmota)
[image]

[WLED](https://github.com/Aircoookie/WLED) 
[image]

PaulWieland, 2023-05-25:
the Arduino OTA that is supported by ratgdo via arduino bootstrapper requires a client to authenticate and push the firmware to the ESP. I don't think thats possible to do from a browser application.

It would be possible to have the ESP8266 download the firmware from the internet and update itself after receiving a specific mqtt command or web UI button click.

Automayted, 2023-06-11:
https://github.com/ESPHome-RATGDO/esphome-ratgdo added this functionality. Closing.

closed 2023-06-11 by Automayted