PaulWieland/ratgdo #56: New Install Several Issues?

efaden, 2023-05-26:
1) The USB port on the D1s that came with don't seem to work?  No serial USB shows up, despite drivers, etc.

2) When I connect the red/white/black wires as shown the wall control doesn't work and goes dark?

3) I can see ESP_XXXX in my wireless networks.  I can connect, but I can't get to 192.168.4.1 or any interface?

Thanks

PaulWieland, 2023-05-26:
step 2 and 3 aren't relevant until you flash the firmware. Do you have a USBC or USBA version of the D1 Mini?
What USB cable are you using?

efaden, 2023-05-26:
It's the USB c.... But apparently it doesn't work with USB c to c.  Switched to USB a to c and it works. 

Got everything installed and all good.

One question though.  It's the position reliable and could it be reported to the mqtt?  Eg could I tell "how open the door is" reliably? Cracked vs fully open

PaulWieland, 2023-05-26:
Unless the last message is `closed` then the door is some state of open.

efaden, 2023-05-27:
But it's there a way to tell the actual position?  Or could the actual
position be reported?

PaulWieland, 2023-05-27:
No, the logic board doesn’t send the position, only the moving state.

efaden, 2023-05-27:
Last question, I assume it's no and only accessible through myq, but if you choose the door remotely it beeps and blinks the lights.  I'd like to be able to trigger that mode also.... Is that possible?  I assume not

PaulWieland, 2023-05-27:
It might be possible but in order to find out you need to capture the packets being sent to the door when you close it remotely. 

ratgdo outputs all traffic it reads to the serial console, so what you can do is open the console and then remotely trigger the door with myq. Copy and paste all of the logged traffic here and I'll see if its using some sort of special command to trigger a mode I don't know about.

efaden, 2023-05-27:
Can I open the serial console remotely?  Otherwise I can grab my laptop.

But yeah I can grab the packets of it sends them

PaulWieland, 2023-05-27:
No, not at this time as I haven't incorporated web serial. Trying to keep the project from getting bloated.

efaden, 2023-05-27:
I'll grab my laptop and see what I can do and post back here

efaden, 2023-05-27:
```
550100801E2A67B19B6DB6143CB7D1AB969EBE | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo/singleratgdo_single/status/motion] 
detected
550100801E2A67B19B6DB6143CB7D1AB969EBE | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo/singleratgdo_single/status/motion] 
detected
550100082B9AFB4796CD04AA29ECB87D565863 | STATUS: door:5 light:1 lock:0 obs:1
Door state closing
QUEUE MSG SENT [ratgdo/singleratgdo_single/status/door] 
closing
550100082B9AF347924D24AA29ECB87D4459E5
550100801E2A67B19B6DB6143CB7D1AB969EBE | MOTION:1
Motion state detected
QUEUE MSG SENT [ratgdo/singleratgdo_single/status/motion] 
detected
550100082B9AF347924D24AA29ECB87D4459E5
550100082B9AF347924D24AA29ECB87D4459E5
55010006166D04B869366BA032851CA729AD94 | STATUS: door:2 light:1 lock:0 obs:1
Door state closed
QUEUE MSG SENT [ratgdo/singleratgdo_single/status/door] 
closed
55010061214A083852249E4A2E620F0D1ECF39 | STATUS: door:4 light:1 lock:0 obs:1
Door state opening
QUEUE MSG SENT [ratgdo/singleratgdo_single/status/door] 
opening
55010061214A0A385B24964A2E620F0D1A5E7D
55010061214A0A385B24964A2E620F0D1A5E7D
55010061214A0A385B24964A2E620F0D1A5E7D
550100602558411CC06DBE40274F9DD4776DB3 | STATUS: door:1 light:1 lock:0 obs:1
Door state open
QUEUE MSG SENT [ratgdo/singleratgdo_single/status/door] 
open
```

Thats a close and open through myq

PaulWieland, 2023-05-28:
Seems like in this case it closed the door with a wireless command because there is no close message in the log. It immediately jumps to the status of "closing".

efaden, 2023-05-28:
Yeah.  I thought that was interesting.  Tried a few times.  Just seems to
close without any message and then it is just closed.  So I guess myq
doesn't send anything

closed 2023-06-23 by PaulWieland