PaulWieland/ratgdo #57: No state changes but actions work

ColColonCleaner, 2023-05-27:
Installed the ratgdo 2.0 on a new chamberlain security 2.0 opener.
Can control the door and the light over mqtt but the state of both the light and the door never changes, it's always in 'unknown' state.
I see in mqtt that there are no messages flowing for status, but telling it to actuate over mqtt works.

I can't get logs from the terminal because it's out in the garage, no computer out there, and I have no figured out how to see esphome logs over WiFi. What should I do here?

ColColonCleaner, 2023-05-27:
Oh i forgot to mention the obstruction sensor works, that state is flowing fine.

ColColonCleaner, 2023-05-28:
Sometimes the commands to actuate the door don't work, i see the LED on the esp board flicker when i send the command so I know the command was received but the opener does not actuate. The weirdest part is that if I wait 10-20 minutes then the commands work again. And also if i power cycle the opener while leaving the esp on then it works for some time before stopping again. Very confusing. Maybe related to statuses not updating?

PaulWieland, 2023-05-28:
It's difficult to know exactly whats going on without a serial log. It does make sense that the controls don't behave correctly without status feedback because ratgdo wont open a door thats already open and vice versa.

It's possible you have a defective board thats unable to read the serial line. You can ship the board back to me and will test it and send you a replacement.

ColColonCleaner, 2023-05-28:
Ah you think its hardware? I have a few other esps that I can flash with this firmware to see if that solves it. I'll let you know. In the meantime is there no way to get a serial over Ethernet open to read the logs on an esp?

PaulWieland, 2023-05-28:
> is there no way to get a serial over Ethernet 

Unfortunately not at this time as I haven't implemented web serial. I may in the future. 

And yeah if you have a spare D1 Mini you can try that in case it has a bad digital I/O pin. If that doesn't fix it then it could be a bad component on the shield.

ColColonCleaner, 2023-05-28:
Ayy! Swapped out the d1mini with a different one and everything works now. Sweet!

PaulWieland, 2023-05-28:
glad to hear!

ColColonCleaner, 2023-06-09:
@PaulWieland running into this problem again. I can run actions but the states don't update. What do you need from me to debug this? I can probably take a laptop out there and do the flash to see the logs.

I run my actual garage cover state in HA off of reed switches at the bottom and top of the door, so the state from ratgdo doesn't actually matter to me, but what it thinks the door state is in causes a problem. When the door is physically open, and HA tells the ratgdo to close, if the ratgdo doesn't think it's open then nothing happens. So if i have an automation that should have closed the door, then the door remains open.

Can there be a way to disable the door state check if we can't figure this out? I need the action to work when people press it.

rogerquake, 2023-06-28:
I'm seeing something similar related to the status not showing. I see this in HA and i'm not seeing any MQTT messages that mention status. I'm assuming that I should see the opening sttaus for example when I also manually open the garage with the physical button or remote?

The good news is that control via HA/MQTT works great.

[image]

Any ideas? My end goal is for ratgdo to provide opening status to a relay that will lock/unlock a dead bolt on the garage.
I'm

joshkephart, 2023-07-05:
I am also experiencing no state changes on v2.1. My ratgdo hardware is a V1 board. I installed 2.1 after running 1.7 since it was released. 1.7 was fully functional but 2.1 does not report any state changes for the cover entity over mqtt (obstruction works fine). Upon a fresh bootup, checking the mqtt messages in Home Assistant mqtt info for the device, I get fresh online status messages for the obstruction and light button, but not for the cover or the new light entity with separate on/off commands.

PaulWieland, 2023-07-05:
@joshkephart you can’t run 2.x firmware on v1 boards unless you solder a transistor to the shield.

joshkephart, 2023-07-05:
> @joshkephart you can’t run 2.x firmware on v1 boards unless you solder a transistor to the shield.

Welllllll that explains that then, I almost asked that question first.... 

Is that mentioned anywhere in documentation yet? I just blindly updated I definitely could have overlooked it... 

I assume I can revert back to 1.7 easily enough without any negative consequences?

PaulWieland, 2023-07-06:
Says so on the ESP Web Tools flasher.
[image]

closed 2023-10-06 by PaulWieland

ColColonCleaner, 2023-11-01:
Ran into a problem last night where the ratgo state got stuck in closing status for hours even though the door was fully closed.

Because of all the state mismatch issues I have been using a script to send commands to the ratgo that would make it happy. E.G. if the door is physically open, but the ratgo thinks it is closed, i send the open command to make it move the door, or vise versa. I didn't account for it possibly getting stuck in 'closing' status so my script failed and we had to use the remotes to move the door this morning until it got the status synced again.

@PaulWieland Could we please have an option to give the door commands regardless of what the ratgo thinks the state of the door is? Or even better, also have a way to force update the state of the ratgo to match what the door actually is if we notice the two don't match for a while?

ColColonCleaner, 2023-11-01:
[image]

PaulWieland, 2023-11-01:
What version of the control board do you have?