PaulWieland/ratgdo #58: Not seeing device created in Home Assistant

underqualifieditguy, 2023-05-31:
Hi Paul,

I received the ratgdo this afternoon and I flashed it/wired it up per the instructions. I can see the following MQTT messages in Home Assistant but I am not seeing an actual device get created. Is there something I am doing wrong?

Message 497 received on /home/garage/Garage Door/status/lock at 10:45 PM:
unlocked
QoS: 0 - Retain: true
Message 496 received on /home/garage/Garage Door/status/light at 10:45 PM:
off
QoS: 0 - Retain: true
Message 495 received on /home/garage/Garage Door/status/door at 10:45 PM:
closed
QoS: 0 - Retain: true
Message 494 received on /home/garage/Garage Door/status/motion at 10:45 PM:
detected
QoS: 0 - Retain: true
Message 493 received on /home/garage/Garage Door/status/obstruction at 10:45 PM:
clear
QoS: 0 - Retain: true
Message 492 received on /home/garage/Garage Door/status/availability at 10:45 PM:
online
QoS: 0 - Retain: true
Message 491 received on homeassistant/light/Garage Door/config at 10:45 PM:
{"~":"/home/garage/Garage Door","name":"Garage Door Light","unique_id":"Garage Door_84:F3:EB:5F:00:8B_light","availability_topic":"~/status/availability","command_topic":"~/command/light","payload_on":"on","payload_off":"off","state_topic":"~/status/light","device":{"name":"Garage Door","identifiers":"Garage Door_84:F3:EB:5F:00:8B","manufacturer":"Paul Wieland","model":"ratgdo","sw_version":"2.1","configuration_url":"http://192.168.1.134/"}}
QoS: 0 - Retain: true
Message 490 received on homeassistant/cover/Garage Door/config at 10:45 PM:
{"~":"/home/garage/Garage Door","name":"Garage Door","unique_id":"Garage Door_84:F3:EB:5F:00:8B","availability_topic":"~/status/availability","device_class":"garage","command_topic":"~/command/door","payload_open":"open","payload_close":"close","payload_stop":"stop","state_topic":"~/status/door","device":{"name":"Garage Door","identifiers":"Garage Door_84:F3:EB:5F:00:8B","manufacturer":"Paul Wieland","model":"ratgdo","sw_version":"2.1","configuration_url":"http://192.168.1.134/"}}
QoS: 0 - Retain: true

PaulWieland, 2023-06-01:
Make sure you haven't disabled mqtt auto discovery in home assistant and that you haven't changed the auto discovery prefix.

PaulWieland, 2023-06-01:
Good to know. I will fix the example and also note that spaces aren’t allowed.

closed 2023-06-01 by PaulWieland