PaulWieland/ratgdo #60: Communications Issue Using RatGDO V2.1 and Chamberlain RJO20 Opener

cerickard2, 2023-06-04:
I have a the board wired to the GDO with the three wires.  No switches or other connections are used.  When I power on the RatGDO board, the garage door immediately opens or closes.  After that it does not respond to any commands from HomeAssistant.  I checked the console and it is registering with MQTT and Home Assistant buttons send commands that are received by the board.  When I step in front of the sensors, it registers an obstruction and clears it after I move away.  It just appears to be unable to communicate with the GDO controller.

The RJO20 says it is Security+ 2.0 and it has a yellow button.  The unit has a control panel elsewhere in the garage with extra programming buttons and a motion detector.

I have attached several log files from various attempts to aid with troubleshooting.

Thank you!
[esp-web-tools-logs.zip](https://github.com/PaulWieland/ratgdo/files/11647425/esp-web-tools-logs.zip)

cerickard2, 2023-06-04:
One other thing...  Whenever I am attached to the console and select "Download Logs", the RatGDO unit reboots and open/closes the garage door.  The logs download successfully though.

cerickard2, 2023-06-04:
I forgot that you asked for me to press to the big button on the MyQ control panel.  Here is a log of when I did that.  However, I didn't see anything written when I pressed it each time.  When I did go to the panel, the yellow LED was blinking.  The button still worked though.
[esp-web-tools-logs (6).txt](https://github.com/PaulWieland/ratgdo/files/11647448/esp-web-tools-logs.6.txt)

PaulWieland, 2023-06-06:
This is perplexing. The last log shows MQTT traffic and valid codes being sent. There are also other codes present in the log which ratgdo did not generate but they are also invalid. There should also be no way that ratgdo would toggle the door upon reboot.

Its possible that the voltage on the control line is out of my expected design range. I assumed between 0 and 2v for low pulse and up to 12v for high pulses. If the low pulses are above 2v for some reason, then ratgdo can't hear it. It's of course also possible theres a defective transistor on the ratgdo board.

If the only issue were that ratgdo was unable to read the messages then I would expect it to be a faulty board. But the fact that your door also won't respond to commands has me thinking something else is wrong.

Can you send a `sync` command over MQTT and see if that triggers it to open?
Can you post a photo of your wiring?

cerickard2, 2023-06-11:
I see with issue #62 that you have determined these side mounted openers are different.  I have the info you requested above but I doubt you will get anything from it that you didn't get from the other ticket.  I'll attach it anyway.

I'm assuming from what you said, that a standard relay will work with our openers.  I have a Zooz Z-Wave relay that I will try with it.

Thanks!

[esp-web-tools-logs-2.zip](https://github.com/PaulWieland/ratgdo/files/11715270/esp-web-tools-logs-2.zip)
[image]
[image]
[image]

PaulWieland, 2023-06-11:
I can easily make a version of the firmware that can open/close the door and report obstructions. What I need to invest time in is figuring out if the door status is being reported and if so, how to read it. In the short term I could add support for a Reed switch for everyone who has one of these sort of security+ 2.0 openers (maybe I’ll call them security plus 1.5).

cerickard2, 2023-06-11:
That sounds great!  Is there anything I can help with to get you information?

cerickard2, 2023-06-14:
@PaulWieland Would it be helpful to hook up a Pi and run argilo's script to capture some logs?  I haven't done anything like that but would be up for it.

PaulWieland, 2023-06-15:
@khaimong has acquired a bunch of data captures using a logic analyzer and sent them to me. I'm working on reverse engineering them now. It looks like it will be quite doable to make ratgdo support Security + 1.0 openers.

khaimong, 2023-06-15:
> @khaimong has acquired a bunch of data captures using a logic analyzer and sent them to me. I'm working on reverse engineering them now. It looks like it will be quite doable to make ratgdo support Security + 1.0 openers.

I think you are crediting the wrong person about data captures.   It wasn't me that got the captures. But I thank you for analyzing it trying to make it work.  I'll gladly test it for you.

nvllsvm, 2023-09-08:
The 2.22a build listed in online flasher did not work with my RJO20. I'm happy to help test any future betas as well.

PaulWieland, 2023-10-06:
Please see my last comment in #54

nvllsvm, 2023-10-24:
v2.26a works flawlessly for me. Thanks!

PaulWieland, 2023-11-01:
Support for security + 1.0 added in v2.5

nvllsvm, 2023-11-01:
@PaulWieland - I'm not seeing the source code for v2.5. Looks like the src directory was removed with the most recent commit. https://github.com/PaulWieland/ratgdo/commit/99f614d8e8e7d7874241f54a502cb35d06507bb1

Is ratgdo closed source now?