PaulWieland/ratgdo #61: Serial Port intermitant dropping

GlennGoddard, 2023-06-09:
I tried on a few computers and found that the serial port was constantly cycling there or not.
After hours of attempts, finally determined that 3v and 5v (Vin) pins had to be shorted on the supplied D1 mini in order to establish comms and to program.
Programmed and setup wifi and MQTT, all working as long as jumper is installed.
Now the however, I am unable to easily short these pins once the ratgdo board in attached.
This is a two fold post, one as the issue and two as assistance to anyone else that might not be able to connect.
I am looking for any potential things I may have missed before I make any permanent modification.

Would there be any issues (I have not fount the schematic in detail yet) of me connecting a jumper between the 3v and Vin on the back of the ratgdo board?  I am a little hesitant since I don't know where all the wire traces go internally and I don't really want to melt any of them off.  I have a screw post for 3V but not for 5V.

I might try additional standoffs and then using all jumper from the D1 to the ratgdo board and just 'y' the 3.3 volt.

PaulWieland, 2023-06-09:
Shorting the 3.3v and 5v pins will burn something out. The ESP8266 uses 3.3v logic and is only tolerant within .5v. There is a 3.3v voltage regulator on the board so that it can be powered from USB which is 5v. By shorting the, you are supplying 5v to the ESP. If it hasn't died already, it soon will.

GlennGoddard, 2023-06-09:
I don't disagree, but the d1 was already not working.  I at least will get some use out of it.  Based on arduous Google searches, it seems like the internal 3 volt buck from 5v is a common issue.  Has anyone else reported a similar issue?

closed 2023-06-23 by PaulWieland