PaulWieland/ratgdo #62: Ratgdo V2.1 does not work with LiftMaster 8500

khaimong, 2023-06-09:
I have a Liftmaster 8500 garage door, which is for a tall RV garage door, side mounted.  I got my Ratgdo V2.1, flashed it, but it does not work correctly. with this door opener.   

1. As soon as I plug in the flashed ESP, the door will move. "Plug in" meaning supply power to the ESP with a USB adapter. If open it will start closing, and if closed, it will start opening.   Sometimes, it will open/close all the way.  Other times, it will stop midway. 
2. After the first movement after plugging in, the controls from HA has no effects. 
4. The door open/close status is not reported.
5. The light on/off does not work.
6. It detects the obstructions correctly.
7. The manual control buttons on the wall device works correctly to open/close door and light. 
8. This garage door has a cable tension sensor as well as a solenoid lock, but not sure if this is relevant. 

I have a second garage door which is a Liftmaster 45ACT.  Ratgdo V2.1 works correctly on this door.  I take the working ESP from this door and put it on the Liftmaster 8500, and it shows the same symptoms as stated above.

PaulWieland, 2023-06-09:
Sounds similar to #60 

I may have to acquire a logic board from one of these openers in order to figure out what’s going on. It’s extremely odd that the door opens when ratgdo is powered on because it is only sending the sync/learn commands. 

What happens if you send a sync command to ratgdo? Does the door move?

khaimong, 2023-06-09:
I have mqtt_explorer.  How do I send a sync command?

PaulWieland, 2023-06-09:
`mqtt.topic = home/garage/MyGarageDoor/command`
`mqtt.payload = sync`

khaimong, 2023-06-10:
The sync command does not make it move, or make a difference. 

I don't know whether this is useful information:  The wall mounted controllers are different for the Liftmaster 45ACT (for which Ratgdo works) and the Liftmaster 8500 (for which Ratgdo doesn't work).   Attached is a pic, top one is for the Liftmaster 8500.  I don't have myQ, though the wall-mount has the myQ symbol. 

[image]

PaulWieland, 2023-06-10:
What happens if you short the control and ground pins?

khaimong, 2023-06-10:
When I short the control and ground pin at the RATGDO terminals, the garage door closes and opens, all the way.   The status of the garage door is not reflected in the HA or MQTT.

PaulWieland, 2023-06-10:
That explains what’s going on the.  This is not a security + 2.0 door.

khaimong, 2023-06-10:
Thank you.  Hmmmm.   Though the literature says it is a security +2.0 opener.  When I switch the wall mount controllers, on the two garage openers that I have, I find that they are not compatible, i.e. The controller that works for the 8500 does not work on the 45 ACT, and vice versa.

PaulWieland, 2023-06-10:
Yeah that makes sense and is the same reason ratgdo isn’t working with it. 

With a security + 2.0 opener, it will not allow you to control it with a simple relay/button which is the way all openers worked for decades. That’s what drove me to build ratgdo.

kpfleming, 2023-06-18:
I have a LiftMaster 8500 as well, and it definitely supports Security+ 2.0 for wireless connections, because the HomeLink buttons in my car are able to control it. It's unfortunate that they didn't use the same protocol to talk to the wall control.

PaulWieland, 2023-06-18:
> I have a LiftMaster 8500 as well, and it definitely supports Security+ 2.0 for wireless connections, because the HomeLink buttons in my car are able to control it. It's unfortunate that they didn't use the same protocol to talk to the wall control.

It's a half breed. The wall mounted control pad that it requires works with Security+ 1.0 openers dating back to 1996. It doesn't support any of the security + 2.0 wall mounted control pads from 2011+. [See Here](https://www.windsordoor.com/assets/uploads/modules/accessory-compatibility-chart-91817-82926.pdf)

bravehand, 2023-10-03:
I've got the same opener and the same issue.  Switching to the Security+ 1.0 beta firmware seems to stop the door from opening/closing on boot but only the obstruction detection is working.   Kinda funny, I bought the ratgdo to solve a problem I didn't really have (but thought I did based on the 8500's manual).

PaulWieland, 2023-10-06:
I just pushed a new alpha firmware out that will hopefully work with the 8500s. [See here.](https://github.com/PaulWieland/ratgdo/issues/54#issuecomment-1751410243)

bravehand, 2023-10-06:
Nice work! I can confirm that "Controlling the door" & "reading the door state (open, closing, closed, opening)" are working for me. I don't have a light or a purple button, but this is great thank you!

PaulWieland, 2023-11-01:
Support for security + 1.0 added in v2.5

closed 2023-11-01 by PaulWieland