PaulWieland/ratgdo #66: HomeKit Integration doesn't seem to work/sync with garage door state

jprasm, 2023-06-22:
The device works just fine within HA. 
However, when actuating open/close via HA's HomeKit integration (for example on the Home app in iOS) the door does not open....although it appears that the command is given via MQTT (using MQTT explorer the command/door=open or command/door=closed is seen to change).

PaulWieland, 2023-06-22:
If the door opens and closes with the respective mqtt commands then I’m not sure how this could be a ratgdo issue… unless the wrong commands are mapped in the auto discovery, but then HA would fail too. 

Can you open the serial monitor while sending commands to compare what ratgdo is seeing?

jprasm, 2023-06-22:
Happy to, Paul....err....can you tell me how to do that? (I'm sorry....I'm new!)

jprasm, 2023-06-23:
Paul, I know you don’t have time to coach everyone here with every troubleshooting issue! If you, or anyone else reading this, can point me to a website or something that explains how to see the serial output of your device, I’ll try and figure out the rest and report back.

PaulWieland, 2023-06-23:
Hi @jprasm connect the usb cable to your computer, open the web flasher to connect to ratgdo. There is a button there to open the serial monitor/console.

jprasm, 2023-06-26:
I want to apologize, I may have figured this out. HomeKit was referencing a template cover that had a condition in it that HomeKit didn't like. 

I've removed the condition, and it seems to be working. Will report back if it is still an issue.

jprasm, 2023-06-26:
I may have spoken too soon, Paul. Still having a few issues. I still haven't managed to get the serial output, because I haven't had time to dismantle it and hook it back up to the computer.

Your idea is a good one, and once it's totally solved, I will update this for sure. For anyone stumbling across this thread here is the description of the issue + my use case:

The Liftmaster garage door I have has an audible warning (beeping) and a delay when it is closed remotely via the MyQ integration. I actually appreciate this safety feature, so I wanted to replicate this behaviour with ratgdo. I also wanted to have it work within HomeKit, so that I can open/close the garage using Carplay from the car when I pull up to the house.

1) The ratgdo autodiscovery creates an entity in HA (i.e. cover.liftmaster) . This entity works perfectly well as is at this point, but the closing lacks the audible warning and safety delay that I want. 

2) So I created a template cover that has a delay script built into it (i.e. cover.liftmaster_w_delay). It steals its state from the auto-discovered entity (entered into the value_template below)

3) Now trying to integrate the 'templated' cover into HomeKit (i.e. cover.liftmaster_w_delay).

Here's the template cover yaml:
```
cover:
  - platform: template
    covers:
      liftmaster_w_delay:
        device_class: garage
        friendly_name: "Garage Door | w/ Safety Delay"
        unique_id: cover001
        value_template: "{{ states('cover.liftmaster') }}"
        open_cover:
          - service: cover.open_cover
            target:
              entity_id: cover.liftmaster
        close_cover:
          - service: script.garage_safety_delay
```

Here's the relevant portion of the homekit.yaml file:
```

homekit:
  - filter:
      include_entities: 
        - cover.liftmaster_w_delay
        - light.garage_lights
```
Anywho....

The device appears in the home app. When I call it to open it within the home app, the garage will open 'sometimes' and sometimes there appears to be a lengthy delay (to the point where I'm not sure the command has gone through), so I toggle it off and on again. In Apple's Home app, the state freezes on "opening", but the garage has yet to open. Sometimes, a subsequent toggle of the device within the home app gets it to work. 

The whole thing works flawlessly within HA, via the dashboard, etc.

jprasm, 2023-07-01:
Paul, this device requires _two flicks_ of the switch in the iOS Home app to open the garage door (for me at least). 

I have a Liftmaster deadbolt 841LM on my door. One flick of the switch in iOS unlocks the deadbolt, then nothing. 
Another flick _OFF_ then _ON_ opens the garage door (closing the door seems fine).

Within HA, it unlocks the deadbolt and continues with opening the door all in one tap. As it should.

I still haven't gotten you the serial output. I'm sorry. Between work and life I haven't found time to dismantle the thing and reconnect it to the computer.

You can see in the template above that it is being triggered simply by invoking the **cover.open_cover** service within HA and also in HomeKit.

jprasm, 2023-07-17:
Update: This situation has resolved itself without any intervention by me. If I ever find out why, I'll report back.

closed 2023-10-06 by PaulWieland