PaulWieland/ratgdo #67: LiftMaster 8160W - No Actions/No Responses

WinterPhoenix, 2023-06-24:
Hi everyone,

I'll just do a list of my setup since it's a more readable format:
- Wemos D1 Mini
- Ratgdo Shield v2.0
- Ratgdo Firmware v2.1
- MQTT Broker: Mosquito inside of Home Assistant
- Garage Door Opener: LiftMaster 8160W

ratgdo shows up in Home Assistant, but always reports:
- Obstruction: Detected
- Light: Unknown
- Cover: Unknown

Furthermore, trying to do any actions just doesn't work. I tried `set_code_counter` followed by `sync`, with a value of `500` then a value of `1000`, but no dice.

I've looked through several issues now trying to figure this out, including #54, which seemed promising until I tried shorting Ctrl and Ground and it did absolutely nothing. Also, hitting the wall door opener does open the door, but there's nothing reported in ratgdo's logs.

- Voltage across Ctrl/Gnd at the shield: ~10.95V
- Voltage across Obst/Gnd at the shield: ~5.90V

Here are some pics:
[image]
[image]

Here are the logs:
[Forthcoming]

Any help is appreciated; thanks for the great project!

PaulWieland, 2023-06-24:
It’s not wired right. Theres a silkscreen label for all of the terminals.

WinterPhoenix, 2023-06-24:
PFFFT. I had it flipped, nice. Rewired it, everything works now. Sorry about that!

closed 2023-06-24 by WinterPhoenix