PaulWieland/ratgdo #69: Trigger Other Relay w/ Deadbolt From Ratgdo Using Wire

rogerquake, 2023-07-07:
I'm looking for some advice on using the ratgdo to trigger another relay that I have setup for a garage deadbolt lock. I'm trying to get the deadbolt to unlock when the garage door opens.

Below is an image of the relay I have setup for the deadbolt. I'm using this relay and works great with HomeAssistant. I have a 5V 1.5A adapter attached to the NO/NC terminals on the relay to power the deadbolt. The deadbolt itself is connected to the Common terminals in the middle on each relay. (They aren't connected in the picture)
terminals.

https://www.amazon.com/gp/product/B08YY7XFS3/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

[image]

I first tried creating an automation in HomeAssistant, however the timing is a bit too slow. There is a 1-2 sec delay before the deadbolt unlocks after the "opening" status returns from ratgdo so the garage door wheels hit the deadboot.

Since the automation method is too slow, I'm thinking physically wiring the ratgdo to the relay, but unsure exactly how. I'm assuming it would be the "pull to ground when door opens" terminal. If that is correct, where would this wire connect to on the deadbolt relay side? The relay has NormallyOpen/NormallyClosed and Common terminals.

PaulWieland, 2023-07-07:
Do you have a V1 (blue) or V2 (white) ratgdo shield?

Right now I don't think the status output pins can help you because the state only changes when the door is fully open or fully closed. On top of that the door status that is sent by ratgdo is read from the door opener as it broadcasts status on the serial line. This is probably not be fast enough to open a deadbolt before the door hits it.

I'm assuming you want the deadbolt to keep someone from manually opening the door if the door opener trolley is unhooked? In that case you could put a reed switch on the trolly and have it control your relay. That way every time the trolly moves from its closed position it will trip the deadbolt relay instantly, but if someone tries to manually open the door (by releasing the trolley and then lifting the door) the dead bolt will keep it locked.

rogerquake, 2023-07-07:
It's a V2 shield. Got ya, good to know. I have a zigbee sensor on there now that reads changes in its axis to determine if it's open or closed, but was hoping the output from the ratgdo might be faster to report the garage door is opening.

My idea if that wasn't possible is to replace the physical garage door button with a switch that triggers HomeAssistant to first unlock the deadbolt and then open the garage door. For access outside the garage, I was thinking just to use geolocation and some away/home logic to unlock the deadlock when i'm in the vicinity after being away for a certain period of time.

Appreciate the quick response!

closed 2023-10-06 by PaulWieland