PaulWieland/ratgdo #70: Shield & ESP8266 Defective Devices

rogerquake, 2023-07-07:
Hi Paul, I was having trouble with my original order of a shield and ESP8266 where the garage door controls from the device worked, but I wasn't receiving any status or seeing any logs when opening the door with the physical button or remote.

I ordered another ESP8266 and shield and now my everything works!

However, the way I got it working was using the new shield and the original ESP8266 I ordered. For some reason, the new shield and ESP8266 together allows me to capture the garage door status however open/close doesn't work. 

So with the new order, the ESP8266 seems defective and with the original order the shield is defective.

I don't want a refund, but i'd would like I get replacements for the parts that aren't working. I can definitely use them for another family member. Please let me know the best way to go about this.

PaulWieland, 2023-07-07:
@rogerquake I'm sorry to hear that! 
If you ship back the defective unit I will ship you a replacement.

rogerquake, 2023-07-07:
Will do! Thanks for all your hard work on this project.

closed 2023-07-07 by rogerquake