PaulWieland/ratgdo #71: Not working on chamberlain 05DCTWF

mkaatman, 2023-07-08:
Hopefully this is simple user error on my part and someone can guide me.

I used three wago connectors. Red for red, yellow for white, black for black.

The motion detection is working. When I hook up the red wire from the ratgdo to the wago it cuts power to the wall mount control.

[image]
[image]

Thank you for any assistance.

mkaatman, 2023-07-08:
A friend has informed me that splicing it like this won't work. I'll remove the wagos tomorrow and see what happens

PaulWieland, 2023-07-08:
Wiring like that should be fine. Have you flashed the firmware?

PaulWieland, 2023-07-08:
Also does it still short to ground (cut power) if you remove the d1 mini from the shield?

mkaatman, 2023-07-08:
> Wiring like that should be fine. Have you flashed the firmware?

I have. It's running 2.1

> Also does it still short to ground (cut power) if you remove the d1 mini from the shield?

I'll try that today and report back.

mkaatman, 2023-07-09:
> Also does it still short to ground (cut power) if you remove the d1 mini from the shield?

Yes, with the d1 removed, as soon as I power the shield the wall switch loses power.

PaulWieland, 2023-07-09:
And can ratgdo control the door at all?

mkaatman, 2023-07-09:
No it can't. The led blinks when I send commands.

PaulWieland, 2023-07-09:
What was your order number?

mkaatman, 2023-07-09:
877677197

mkaatman, 2023-07-09:
Closing as id10t error. I pulled the wagos out and restriped the garage wires and everything is working now.

closed 2023-07-09 by mkaatman