PaulWieland/ratgdo #72: Is Chamberlain PD210D supported?

cliff85, 2023-07-09:
Hello

Is Chamberlain PD210D SN 41ac050-1m supported? I ordered the ratgdo before looking at compatibility and I am getting similar results to some of the other open issues (like https://github.com/PaulWieland/ratgdo/issues/54). When I power on ratgdo the garage door will open or close depending on its status and I can see it in HA however no commands from HA work.

PaulWieland, 2023-07-09:
Hi there, the PD210D is a Security Plus 1.0 opener so it is not currently supported.

I am going to add support for 1.0 openers as soon as I can, but it might require a very slight hardware revision to prevent ratgdo from toggling the door when it reboots. I'm still working on finding hardware to test with.

These older style door openers can be controlled with a simple relay, which is good - but in the case of ratgdo the signal line is pulled low by the D1 mini when it reboots which causes the door to open or close.

cliff85, 2023-07-09:
Got it thanks. What should I monitor to track status?

PaulWieland, 2023-07-09:
The old fashioned way of doing this was to use a reed switch attached to the track/trolly to detect the door state. Your model opener has a bunch of limit switches built in, which if you know how to use a multi meter you could figure out which switch correlates with which door position - but to use them you would need a circuit and a custom firmware for a microcontroller. 

That being said, ratgdo will be able to read the door state directly from the serial line without any soldering on these 1.0 openers. I just need some more time to get the specifics sorted out.

cliff85, 2023-07-10:
Sorry I should have been more clear with my question. What is the best medium to track status on ratgdo Security Plus 1.0 support? Is there a open github issue or this one? Also if there is something I can donate to make it easier like a board from ebay let me know!

closed 2023-07-21 by cliff85