PaulWieland/ratgdo #74: Question

craigwitter, 2023-07-14:
Good morning,
  While the garage door hardware works great to control the door, is it expected that the remote on the wall (also a security enable device), would take multiple pushes to activate/close the door after installing it? I presume it is having to cycle through codes as the last code it knew about changed due to the ratgdo hardware? 

Thanks in advance,
Craig

PaulWieland, 2023-07-14:
If you flash ratgdo or reboot it while connected to the door opener, it pulls the 12v line low which causes the wall control panel to lose power and then restart. When that happens you have to click the button three times to resync it with the door opener. But otherwise ratgdo and the factory wall mounted button do not interfere with each other.

zhudhjen, 2023-09-09:
Thanks Paul for developing this and answering this question. By saying they "do not interfere with each other", are the door status synced between both when the operation happens on the other end? 

1. When the door is closed by factory wall panel, is the status in ratgdo updated?
2. When light is turned on/off by factory wall panel, is the status in ratgdo updated?
3. When light is turned on by ratgdo, will motion sensor on factory wall panel override the light status? E.g. turn the light off after motion expire?

Thanks in advance!

PaulWieland, 2023-09-11:
> Thanks Paul for developing this and answering this question. By saying they "do not interfere with each other", are the door status synced between both when the operation happens on the other end?
> 
> 1. When the door is closed by factory wall panel, is the status in ratgdo updated?
yes

> 2. When light is turned on/off by factory wall panel, is the status in ratgdo updated?
yes

> 3. When light is turned on by ratgdo, will motion sensor on factory wall panel override the light status? E.g. turn the light off after motion expire?
yes

> 
> Thanks in advance!