PaulWieland/ratgdo #76: What are all the magic numbers in getRollingCode() doing ?

zinahe, 2023-07-15:
Hi,

Now that ratgdo is working for me; I'm going through the source trying to understand what happening. Can somebody kindly explain:

1. the reasoning behind the 6 reboot code sequences that are transmitted at every reset ? Why 6?
2. while I understand the rolling code algorithm needs a combination of fixed and varying data input parameters; where did all these magic (fixed+data) numbers for the 6 reset coming from?
3. why is this another magic number 64-bit **id** being OR'd to the fixed value?

```
void getRollingCode(const char *command){
	. . .
	uint64_t id = 0x539;

	if(strcmp(command,"reboot?") == 0){
		fixed = 0x???;
		data = 0x???;
	
	}else if(strcmp(command,"door?") == 0){
		fixed = 0x???;
		data = 0x???;

	}else if(strcmp(command,"light") == 0){
		fixed = 0x???;
		data = 0x???;
	}

	. . . 

	fixed = fixed | id;

	. . . 
}
```

zinahe, 2023-07-21:
Anyone ?

bdraco, 2023-08-27:
https://github.com/ESPHome-RATGDO/esphome-ratgdo/blob/fd8e60245ee188f78933152233e1c69884d96544/components/ratgdo/ratgdo.h#L56

closed 2023-10-03 by PaulWieland