PaulWieland/ratgdo #77: Zooz ZEN17 Wiring

brandontrue, 2023-07-16:
I am trying to have a momentary relay switch (Zooz ZEN17) connected to the 2.0 board. How should I wire this to the status/trigger side of the board?

https://www.support.getzooz.com/kb/article/1416-zen17-universal-relay-faqs/

zinahe, 2023-07-16:
Someone can correct me if I'm wrong, but at the moment **ratgdo** requires two separate monetary push buttons for opening and closing respectively. I am guessing this is done to mitigate the high level of uncertainty that comes with denouncing a mechanical switch. 

Just wondering if you have any particular reason why you would need to use the "Zooz" device to top of ratgdo. The Wemo D1 Mini (esp8266) is a very flexible WiFi capable device which allows you to add any kind of custom automation.

Regards,

brandontrue, 2023-07-16:
The wifi connection is very unstable in my garage (even with an extender with is another issue). Looking to use the Zooz to bridge to my Zwave network into Home Assistant

PaulWieland, 2023-10-06:
There are two trigger pins, one to open and one to close the door. Connect either pin to ground momentarily and it will command the door to open or close. The contact has to be closed for 500ms. Anything shorter than that will be ignored (this is a safety to avoid some sort of electrical noise causing a false trigger).