PaulWieland/ratgdo #78: Device not responding / Can't open web ui

underqualifieditguy, 2023-07-26:
About three days ago, my garage door device went unreachable in Home Assistant / HomeKit. It's on a static IP and responding to ping, confirmed it's the right device by unplugging the power and the ping stops (I didn't see a MAC address printed on it). The web UI does not load, it definitely is still enabled because I've been able to manage it since I flashed it and wired it up. Any thoughts? Should I try reflashing it?

PaulWieland, 2023-07-26:
That’s really odd. Try reflashing. Don’t erase it or you will lose the stored code counter.

underqualifieditguy, 2023-07-26:
> That’s really odd. Try reflashing. Don’t erase it or you will lose the stored code counter. 

I'll give this a shot this evening and report back.

underqualifieditguy, 2023-07-27:
I re-flashed it, still not working. All I see in the logs are MQTT attempts repeating over and over. I know that MQTT is working because I have lots of other devices communicating on it and they aren't facing any issues.

underqualifieditguy, 2023-07-27:
OK let's hold up a sec maybe I do have issues with MQTT causing this. I'll get back to you.

underqualifieditguy, 2023-07-27:
This was totally of my own doing, I accidentally overwrote my MQTT auth file, sorry to disturb!

closed 2023-07-27 by underqualifieditguy