PaulWieland/ratgdo #80: MQTT Warning with Home Assistant 2023.8.0

sgruby, 2023-08-02:
Just upgraded to Home Assistant 2023.8.0 and received a warning message.

Discovered entities with a name that starts with the device name

This stops working in version 2024.2.0. Please address before upgrading.

Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

light.garagedoor_light
binary_sensor.garagedoor_obstruction

kirsch33, 2023-08-10:
Same issue here

God-TM, 2023-08-14:
I'm seeing this as well:

binary_sensor.garage_door_obstruction
light.garage_door_light

Timeteo, 2023-08-22:
+1
I see that for people using the zigbee2mqtt integration, the maintainers will be fixing it in an upcoming version. Will that be the same for ratgdo? Or should I manually update the entity names?

laptopfreek0, 2023-10-03:
+1
This is showing up as well. It might be possible to manually remap this with the MQTT configuration.yaml but this shouldn't be required since the change should be simple enough within the firmware.

PaulWieland, 2023-11-01:
Corrected with v2.2 / v2.5

closed 2023-11-01 by PaulWieland