PaulWieland/ratgdo #82: New LiftMaster 98022 compatibility?

NWSAlpine, 2023-08-06:
I have an older chamberlain with a 41AC050-2M control and doesn't appear to work with ratdgo. I've wanted to upgrade to a jackshaft opener for noise reduction so I ordered the brand new LiftMaster 98022. Anyone hook one of these up yet with ratdgo 2.0 boards?

PaulWieland, 2023-08-08:
The installation manual says that it comes with a 880LMW wall control panel.
According to [this chart](https://www.windsordoor.com/assets/uploads/modules/accessory-compatibility-chart-91817-82926.pdf) it is secplus 2.0 and will work with ratgdo.

Another way to check compatibility is to short the wall control panel wires together. If the door opens, then its security plus 1.0 or earlier. If it doesn't open then its security plus 2.0 and will work with ratgdo.

Eventually ratgdo will support 1.0 as well, but it's not ready yet.

tfcarlin, 2023-08-22:
@PaulWieland Will the v2 hardware support security plus 1.0?

PaulWieland, 2023-10-03:
@tfcarlin  yes, but in a (near) future release.

closed 2023-10-03 by PaulWieland