PaulWieland/ratgdo #83: Finding 3.3V on Chamberlain 50dcbl5lk

rohankapoorcom, 2023-08-11:
I got my ratgdo setup with the Chamberlain logic board model number 50dcbl5lk using the USB connection for power. I'd like to power it internally so that it is powered by the garage door opener's battery for backup.

I tried to find a source for 3.3v on the board but couldn't find one with some multimeter probing. Any suggestions on where to look?

Thanks!

xTerminatioNx, 2023-09-05:
I am far from an expert but when I was looking for similar info I found this comment: https://github.com/PaulWieland/ratgdo/issues/17#issuecomment-1292570895

Specifically this part:

> ...
>You can get enough 3.3V to power the ESP8266 from the center pin of the LM317 voltage regulator (circled in the last picture below)
> ...
>[image]

My board does not have the same model number as theirs (or yours) or look exactly the same but I was able to find an LM317 component fairly easily on my board. By soldering to the center pin I was able to get 3.3V reliably and the ratgdo works great using it.

There could be better options but this is the best answer I could find at the time and it works so I'll keep using it unless I find a reason not to.

Hope that helps. Good luck.

rohankapoorcom, 2023-09-08:
@xTerminatioNx You're amazing, thank you! There were in fact 2 LM317D2s on the board and one of them had 3.3V on the center pin (the center pin on the other had 0V to ground). Soldering to that worked perfectly!!

Thank you again!

rohankapoorcom, 2023-09-08:
Actually I think I spoke too soon. It works perfectly when there is AC power but on battery I get `Motor Control Error Code 1-5 See User Manual` displayed on the screen and nothing runs. Disconnecting the 3.3V connection from ratgdo immediately makes the door work again.

I'm wondering if that LM317D2 doesnt have sufficient power when on battery.

rohankapoorcom, 2023-09-09:
Yeah, I live in California, so the battery usage is mandated in the building codes.

I can confirm that tapping the battery directly with the regulator works great!

closed 2023-09-09 by rohankapoorcom