PaulWieland/ratgdo #84: OTA or API command to reboot remotely

kirsch33, 2023-08-12:
ever since https://github.com/PaulWieland/ratgdo/issues/37 was resolved I have been having ALMOST zero issues except that every 2-3 weeks I find that the ratgdo will have the wrong (opposite, really) door status and seemingly the only way to resolve is simply to login to the web UI and initiate a reboot.

is there any way to set a periodic reboot to a cronjob on the ratgdo or initiate one with a home assistant automation?

PaulWieland, 2023-08-12:
Are you using V2.0?

kirsch33, 2023-08-12:
V1.7x as I do not have a V2.x board

PaulWieland, 2023-08-12:
Okay, this isn't an issue with v2 since ratgdo continuously gets status updates from the GDO. You can update your v1 board to v2 if you can do some basic soldering.

kirsch33, 2023-08-12:
well i own a soldering iron but each time i used it, it comes out in a barely functional glob. not sure I have the skill to do it on a small board like the GDO...

but maybe this is a good opportunity to learn, if youre saying that its not possible to solve this issue on a v1 board?

PaulWieland, 2023-08-12:
For the mod, the soldering happens on the large pins one the back side of the board so it’s quite approachable. The main thing for soldering is to use flux and leaded solder. Solder that globs is usually lead free (what you usually find in the hardware store is for plumbing which must be lead free). 

Anyway v2.x works much better than v1.x and will fix your problem.

kirsch33, 2023-08-12:
wow, TIL.. thanks for the info and I'll plan to update to v2.X

kirsch33, 2023-08-12:
@PaulWieland follow up Q...best online source for small quantities of electronic parts? the mosfet needed for the conversion I mean. link to a recommended product would be even better if you have one!

closed 2023-08-12 by kirsch33

PaulWieland, 2023-08-12:
I’ll mail one to you. What’s your order number so I can look up your address?

kirsch33, 2023-08-12:
awesome. order number is 1409669259

kirsch33, 2023-08-27:
@PaulWieland i received the mosfet and was able to install it without issue! I appreciate your advice on the soldering and also for sending the part free of charge. v2.1 is now flashed and working well.

didnt get a good pic of the mosfet itself before reinstalling but heres my soldering work on the back side of the board. flux + leaded solder definitely made it way easier. i used some electrical tape to hold to mosfet in place below the ESP, and had to bend the 2 outer legs a bit to give some more room
[image]