PaulWieland/ratgdo #90: Can it be used with a LiftMaster Gate and Trip Sensor?

Yediyd, 2023-09-08:
Any reason you think this wouldn't work with a liftmaster gate instead of garage door?

PaulWieland, 2023-09-08:
It’s quite possible it would work. Do you know which control panels are compatible with the gate?

Yediyd, 2023-09-11:
It's the LiftMaster LA400UL: https://www.liftmaster.com/24vdc-residential-linear-actuator/p/LA400UL

Control board is this one: https://www.gateopenersunlimited.com/liftmaster-001d9040-1cc-main-control-board-k1d9040-1cc-la400ul-la412ul-la500ul.html

PaulWieland, 2023-10-03:
For everyone running these gate actuators, do you have a hardwired button somewhere near the gate that allows you to open it?

Yediyd, 2023-10-03:
Not exactly, usually use remotes but the board has these control ports and buttons on them:

https://cdn11.bigcommerce.com/s-fdyqgtwe91/images/stencil/1280x1280/products/4655/16376/liftmaster-k1d8389-1cc-main-board-with-heat-sink-32__41392.1675254987.jpg?c=1

Pauldy, 2023-10-04:
So tonight I tried using the SBC port and it just opened the gate at startup, so thats likely literally just a momentary contact port.  Then I tried the expansion board port black was gnd, red 12v, yellow and blue 3.3v.  I tried white on gnd and red on yellow, blue, and red without success.  Security+ 2.0 is referenced all over the manual but I can't find anything but wireless interfacing.

PaulWieland, 2023-10-04:
The question is whether that 12v port is a serial port, and if it is, is it sec plus 1 or 2. What would be ideal is to get the gate position directly from the logic board, otherwise it would have to work off of limit switches. Hard to figure out without the right equipment...

YBonline, 2023-10-04:
the other possibility is the 2 pin green plug above the lock... on different versions of the board, that is labeled for communication to another matching control board for dual gate operations ("link")... they allow either communication over that 2 wire plug, or wireless with security+ 2.0

Pauldy, 2023-10-04:
Interesting that port is unlabeled on all my documentation.  I'll take a look at that one today.

Pauldy, 2023-10-04:
[image]

Promising with com link silk screened there.  Looks like mine might need a little help though as there is no terminal block populated.

YBonline, 2023-10-04:
> Interesting that port is unlabeled on all my documentation. I'll take a look at that one today.

Yes, the connected control boards is, from my understanding, technically not supported on that model... but I have no idea if   they disable the communication port entirely, or just removed the terminal block connector.

I'm awaiting my ratgdo, but I have a few boards that do have the terminal block on there for my gates

PaulWieland, 2023-10-04:
Are any of the other terminals breakouts for the limit switches? If theres no serial communication then ratgdo can still reliably determine door/gate status (closed, opening, open, closing) if there is a limit switch for fully open and another for fully closed.

Pauldy, 2023-10-04:
I'm pretty sure the device knows the gate position because you manually set each arm to open to point x and it does a very good job of hitting it even after multiple start/stops.  But it may also be handled by the swing arm itself and there appears to be communications there to the arm.   In myq it knows if it is open or closed even when using the remote and it gets all its data via a wireless gateway device.

YBonline, 2023-10-04:
> Are any of the other terminals breakouts for the limit switches? If theres no serial communication then ratgdo can still reliably determine door/gate status (closed, opening, open, closing) if there is a limit switch for fully open and another for fully closed.

If you get the auxiliary board, the answer is yes. You can configure dip switches for AUX relay 1 to OFF OFF ON and aux relay 2 to OFF ON ON, then you have the aux relay 1 terminal as open, aux relay 2 is moving (opening/closing), and lock terminal (on main board) would indicate closed. Without the auxiliary board, there is only the closed output.

Pauldy, 2023-10-04:
I poked around on the com link and was unable to find continuity between either of the pins and the common ground and there are quite a few ics there that aren't populated.  So I guess it's hurry up and wait till you can try on one with the terminal block.

Yediyd, 2023-10-04:
Maybe I missed it, what auxiliary board is this?

YBonline, 2023-10-04:
Yes, thats the expansion board, I guess I called it by the wrong name (auxiliary board)... sounds like it could be a month before I get my ratgdo, I just ordered it the other day after the MyQ outages... and I did only order 1 and if its one per door/gate, I'm gonna need like 10+ of them. Thankfully MyQ started working again for me so hopefully they be nice

TopBanana123, 2023-11-01:
Is it possible to wire up ratdgo to a gate wireless control? I guess the problem there is you wouldn't get the status of the gate

Another question/idea. My LA400 is actually wirelessly linked to my MyQ Garage door opener via the learn button. Then, my Garage Door Opener in MyQ shows both the garage door and the gate and provides all of the details and control, including stopped, obstruction, etc. So could there be a way to intercept that wireless signal? Or, could ratdgo somehow interface with the garage door opener to control the gate?

ratsputin, 2023-11-01:
> Yes, thats the expansion board, I guess I called it by the wrong name (auxiliary board)... sounds like it could be a month before I get my ratgdo, I just ordered it the other day after the MyQ [...]

I just hooked this expansion board up to my gate controller.  It, along with two Shelly Plus 1's will give you everything you need.

Here's a [post ](https://www.reddit.com/r/homeassistant/comments/17hosxt/finally_was_able_to_eliminate_myq_with_my/) I did on Reddit.  If you look at the comment string, you'll see how I did it.

Yediyd, 2023-11-08:
@PaulWieland ,
Mind taking a look at this and helping me with wiring?

I have the expansion board for the gate and I think I have an idea of how to wire it up but need some help.

Here is a close up of the parts on the board that matter:
[image]

From the ratgdo, on the right side with QR code facing up, the Close and Open terminals from the trigger section should go to the OPN and CLS terminals on the right side of the expansion board to operate the gate. Additionally, I have photo eyes on the gate and I'm assuming I can wire the Obst from the ratgdo to the port labeled EYE ONLY on the right side of the expansion board, but I'm not certain is that is a status terminal (are the eyes breached) or a control terminal for inputting additional sensors.

The main question I have is the status outputs for reading open closed and moving. The two relays 1 and 2 in the middle of the expansion board can be toggled for different statuses; from left to right Open, Closed, Moving. I'm thinking I can use the first relay to tell Ratgdo the gate is open when positive and closed when negative (NO vs NC), but not sure if I'm understanding that correctly. And then the second relay could be used for the Moving status for when the gate is opening. Any thoughts on how to wire the statuses into the ratgdo with that in mind?

Yediyd, 2023-11-08:
Correction: the relay toggles are not simply open, closed, moving. The chart below shows what they can be set to:

[image]

ratsputin, 2023-11-08:
> Correction: the relay toggles are not simply open, closed, moving. The chart below shows what they can be set to:
> 
> [image]

Take a look at this thread in my post about using Shelly Plus 1s to do the same thing: https://www.reddit.com/r/homeassistant/comments/17hosxt/comment/k6qh2fc/?utm_source=reddit&utm_medium=web2x&context=3

I talk about this specific issue.

Yediyd, 2023-11-08:
@ratsputin ,
I saw your post, thanks for sending the manual but I’m trying to figure out how the ratgdo should be wired up. I’m guessing the shellys only get open or closed and not moving?

ratsputin, 2023-11-08:
> @ratsputin , I saw your post, thanks for sending the manual but I’m trying to figure out how the ratgdo should be wired up. I’m guessing the shellys only get open or closed and not moving?

The problem you're going to run into is that the accessory board does not provide a "moving" state.  You can only derive a moving state from the fact that both the "gate closed" and "gate open" switches are both off (as I referenced in the truth table).

Yediyd, 2023-11-08:
What about row 4? off on on: energizes when motor is on

ratsputin, 2023-11-08:
> What about row 4? off on on: energizes when motor is on

As you only have two relays, you'd lose either the "closed" or "open" state if you used one of the for motion.

YBonline, 2023-11-08:
> > @ratsputin , I saw your post, thanks for sending the manual but I’m trying to figure out how the ratgdo should be wired up. I’m guessing the shellys only get open or closed and not moving?
> 
> The problem you're going to run into is that the accessory board does not provide a "moving" state. You can only derive a moving state from the fact that both the "gate closed" and "gate open" switches are both off (as I referenced in the truth table).
> 
> Edit: Well, it does but you only have two relays. You can't get all three states.

So assuming your control board has a "Lock" relay (I think all the boards do, but I'm not sure, every one I own has it), you can get all three states. The "Lock" relay is energized any time the gate is closed, so you can use the expansion board relays for Motion and Open. We also do have dedicated terminals to trigger Open, Closed, and Stop, so most of the functionality can be emulated.

As far as using a ratgdo, I don't think there are enough terminals, but using the esp board without the ratgdo device should be fine if we aren't using sec+ 2.0. I get my ratgdo tomorrow and my board has the com link board populated, so I'm hoping to do it with just 2 wires (I'm already using the aux relay to tell me if the breaker blew for AC power, so there isn't enough relays for me)

drharis, 2023-11-09:
Hello,
Been following this thread. I also have a liftmaster gate opener (LYN400) and I am in UK. The control board is CB124EV. I am also really keen to know if ratgdo will work with this. My control board looks completely different the ones which is shown here. I will post the pictures below. 
[image]
[image]

PaulWieland, 2023-11-09:
> The problem you're going to run into is that the accessory board does not provide a "moving" state. You can only derive a moving state from the fact that both the "gate closed" and "gate open" [...]

ratgdo will report all four states (open, closing, closed, opening) by wiring two limit switches. It knows when the door/gate is opening or closing based on the last state of each switch.

The dry contact section of the wiring instructions explain how to connect it:
https://paulwieland.github.io/ratgdo/03_wiring.html

Yediyd, 2023-11-14:
Anyone got this wired up yet to a gate with ratgdo? I'm still unsure what wires need to go where. The guide is for garage doors and I'm not sure how to translate it for the image of the control board I posted above.

Yediyd, 2023-11-15:
So as far as I can tell, ratgdo does not support gates. We need control for both an open button and a close button, but I'm not seeing a way in the documentation to do that from the left side of the ratgdo board.

ratsputin, 2023-11-15:
> So as far as I can tell, ratgdo does not support gates. We need control for both an open button and a close button, but I'm not seeing a way in the documentation to do that from the left side of the ratgdo board.

If you look on the main control board, there is an "SBC" connection--Single Button Control.  It works like you would expect a single-button garage door opener to work.

gpanders, 2023-11-27:
Here is a copy of the wiring diagram for a CSL24UL (which is what I have). It uses the same control board as the one posted upthread and includes the expansion board, as well as the relay state table posted earlier. Maybe this will be useful to somebody.

[image]

YBonline, 2023-11-27:
Now that it finally arrived, hooking a ratgdo up to the link port did not seem to do anything on the CSL24UL :-(

gpanders, 2023-11-27:
>ratgdo will report all four states (open, closing, closed, opening) by wiring two limit switches. If those relays can be configured so that one closes its contacts when the gate is fully open and the other closes when fully closed then you have what you need.

This should be possible with the expansion board. Set Relay 1 to be energized when at open limit (OFF OFF ON) and Relay 2 to be energized when not at the close limit (OFF ON OFF).

The "Open" and "Close" pads in the "trigger" section of the ratgdo board would then connect to the NO terminal on relay 1 and the NC terminal on relay 2, respectively. Then I assume that the SBC terminal on the expansion board connects to `Red Ctrl (to gdo)` on the ratgdo board? (this one I'm less sure about)

The expansion board has terminals for obstruction detection too. My gate uses only the eye sensors so I assume I might be able to connect that terminal to the obstruction sensor pad on the ratgdo board.

I don't have my ratgdo yet or I'd try all this myself.