PaulWieland/ratgdo #91: MQTT commands produce no result

cjcole8, 2023-09-11:
I just got my device, installed it, and it doesn't seem to work.
I flashed it with ratgdov2.1
I connected it to HomeAssistant MQTT, and it automatically showed up just as it should.
However the up/down/stop/light commands don't seem to do anything.

The control panel on the wall is 882LMW.
The actual unit connected to the ceiling is LiftMaster 7675.

Logs attached (is there any way to get the logs over wifi like ESPHome rather than climb a ladder and plugin my computer up by the ceiling?)
[esp-web-tools-logs.txt](https://github.com/PaulWieland/ratgdo/files/12571850/esp-web-tools-logs.txt)

PaulWieland, 2023-09-11:
Can you post a picture of your wiring?

The logs show that the rolling codes are being sent, which means ratgdo is receiving the commands from home assistant via MQTT, so the issue is with the door opener receiving the commands.

Can you press the button on your wall panel and fully cycle the door once, then check and see if ratgdo is receiving the status? Also test and see if obstructions are detected.

There is an [ESPHome port ](https://github.com/ESPHome-RATGDO/esphome-ratgdo)of ratgdo you can use, which gives you a remote console, but I didn't write it so I'm not 100% sure yet if they are logging everything necessary to diagnose whats going on here.

cjcole8, 2023-09-11:
hi - I was looking through old tickets, and found a similar one where the wiring was wrong (#67 ). Sure enough, I did the same thing. It all works now!

FYI - the confusing diagram is here: https://user-images.githubusercontent.com/4663918/235453980-04a642fa-a181-4297-b4f3-06e1315e02fa.png

I see it was my mistake - the board depicted here is flipped over, and I had assumed it was right side up. But if you get more of these, maybe make a note on the picture.

Feel free to close this out, and thanks for all the work you did to make these work!

closed 2023-09-11 by cjcole8