PaulWieland/ratgdo #94: How to setup?

llamafilm, 2023-09-24:
I've never used ESP chips before so perhaps this may be obvious to others, but I'm stumped how to set this up.  When I try to flash the firmware in Chrome, the console shows errors:
```
Received invalid checksum 10. Expected 202
Improv initialization failed. Error: Improv Wi-Fi Serial not detected
```

I purchased the "ratgdo v2.0 shield with ESP8266 D1 Clone".  
I'm using Chrome 116 on macOS Ventura.

I'm trying to use version "ratgdo v2.1" and the serial device looks like this:

[image]

PaulWieland, 2023-09-24:
I haven't encountered that one before, it could be a bad D1 Mini, but it could also be a bad USB cable/adapter. Can you try another cable? Do you have a micro or a usb C version of the D1 Mini?

llamafilm, 2023-09-25:
This one has USB micro connector.  I've tried 2 different cables and 2 different A-to-C adapters (one from Apple and one from Anker), and always get the same result.

PaulWieland, 2023-09-25:
What’s your order number?

llamafilm, 2023-09-25:
000090

PaulWieland, 2023-09-25:
Thats not a valid order number, how about your email?

PaulWieland, 2023-09-29:
New d1 mini sent.